stronnag/mwptools #5: mwp, compile error listen_all does not exist

rudi48, 2015-03-22:
Hello Jonathan,
Thank you for the interesting program mwp.
Unfortunately I could not compile mwp. I get the error:
mapsources.vala:278.21-278.33: error: The name `listen_all' does not exist in the context of `SoupProxy'
I use Ubuntu 14.04.

Is it possible to help me please?
Regards, Rudolf

p.s.: It would be nice to mention in your README.md that the "*-dev" version of the support packages must be installed, e.g. libchamplain-0.12-dev.

stronnag, 2015-03-22:
Rudi,

Do you have the latest from git? It builds for me using the top level ./makeall.sh script. This script sets specific environment variables to cope with the old version of libsoup shipped with all versions of Ubuntu prior to 15.04.

If you make from the mwp directory, on Ubuntu you would need.
```
$ make SPOPTS="-D BADSOUP"
```
It is unreasonable (IMHO) to expect me to specially support a single Linux disto, in particular one I don't actually use, however you will find that the Ubuntu dependencies are listed in  docs/ubuntu-deps.txt, and the documentation states that you need -dev variants. It also needs to explain to use the top level ./makeall.sh script, an omission that will be corrected today.

stronnag, 2015-03-22:
Documentation updated

rudi48, 2015-03-22:
Hello Jonathan,

Thank you for the hint, now it compiles, and "mwp" starts.
I am sorry, that I did not read the well prepared document ubuntu-deps.txt.
Best regards, Rudolf

closed 2015-03-22 by rudi48