stronnag/mwptools #6: Installation is unnecessarily difficult / broken

stronnag, 2016-03-04:
This is currently being worked. As of 2016-03-04
````
make
sudo make sysinstall 
````
will install in system locations (/usr) and requires no environment variables and
````
make
make install 
````
will install locally, requiring XDG_DATA_DIRS and ~/bin on the PATH.

This will be further simplified (and documented) and the issue closed once the process is complete.

stronnag, 2016-03-04:
As of add9bc7, in theory this is fixed. In practice, it possibly is not. Let's keep the issue open a while yet.

You are advised to read INSTALL before installing and running both:
````
sudo samples/clean-system.sh
````
and
````
samples/clean-local.sh
````
is probably a good idea before reinstalling.

wirgi, 2016-03-06:
Ciao! 
Replay_bbox_ltm  dependencies
we must also install this? 
The Ruby 'rubyserial' gem
and where is it?

stronnag, 2016-03-06:
What happens when you run the commands in the installation
instructions (mwptools/bbox-replay/replay_bbox_ltm.pdf) ? 

Is there some error message?

-jonathan

wirgi, 2016-03-07:
thanks. then..
test installation mwptools : Virtualbox - ubuntu-15.10-desktop-amd64.iso 
**something here....**
`wirgi@wirgi-VB:~$ sudo apt-get install valac-0.20
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package valac-0.20
E: Couldn't find any package by regex 'valac-0.20'
wirgi@wirgi-VB:~$ sudo apt-get install valac-0.20-vapi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package valac-0.20-vapi
E: Couldn't find any package by regex 'valac-0.20-vapi'
wirgi@wirgi-VB:~$ sudo apt-get install libvala-0.20-0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libvala-0.20-0
E: Couldn't find any package by regex 'libvala-0.20-0'
wirgi@wirgi-VB:~$ sudo apt-get install valac
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libvala-0.28-0 valac-0.28 valac-0.28-vapi
The following NEW packages will be installed:
  libvala-0.28-0 valac valac-0.28 valac-0.28-vapi
0 upgraded, 4 newly installed, 0 to remove and 173 not upgraded.`
**then..**
`wirgi@wirgi-VB:~$ cd ~
wirgi@wirgi-VB:~$ mkdir quad
wirgi@wirgi-VB:~$ cd ~/quad
wirgi@wirgi-VB:~/quad$ git clone https://github.com/stronnag/mwptools.git
Cloning into 'mwptools'...
remote: Counting objects: 3901, done.
remote: Compressing objects: 100% (58/58), done.
remote: Total 3901 (delta 27), reused 0 (delta 0), pack-reused 3843
Receiving objects: 100% (3901/3901), 53.29 MiB | 615.00 KiB/s, done.
Resolving deltas: 100% (2917/2917), done.
Checking connectivity... done.
wirgi@wirgi-VB:~/quad$ git clone https://github.com/cleanflight/blackbox-tools
Cloning into 'blackbox-tools'...
remote: Counting objects: 5494, done.
remote: Total 5494 (delta 0), reused 0 (delta 0), pack-reused 5494
Receiving objects: 100% (5494/5494), 41.87 MiB | 584.00 KiB/s, done.
Resolving deltas: 100% (2306/2306), done.
Checking connectivity...done.`
**then..**
`wirgi@wirgi-VB:~/quad$ cd ~/quad/mwptools/
wirgi@wirgi-VB:~/quad/mwptools$ make
make -C mspsim
make[1]: Entering directory '/home/wirgi/quad/mwptools/mspsim'
valac -D HAVE_FIONREAD -D NOPUSHFRONT -X -O2 -X -s --thread --target-glib=2.36 --pkg posix --pkg linux --pkg gio-2.0 --pkg gtk+-3.0 --pkg libxml-2.0 --pkg json-glib-1.0 -o mspsim  mspsim.vala ../common/mwp_xml.vala .`
----------# **---- jump---**-----

`make[1]: Leaving directory '/home/wirgi/quad/mwptools/cf-cli'
make -C horizon
make[1]: Entering directory '/home/wirgi/quad/mwptools/horizon'
cc -O2 `pkg-config --cflags gtk+-2.0` ath.c gtkartificialhorizon.c -o mwp_ath `pkg-config --libs gtk+-2.0`
ath.c: In function ‘publish_plug’:
ath.c:74:5: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     (void)write(fileno(stdout), buf, strlen(buf));
     ^
cp mwp_ath ~/bin/
cp: cannot create regular file ‘/home/wirgi/bin/’: Not a directory
Makefile:9: recipe for target 'mwp_ath' failed
make[1]: *** [mwp_ath] Error 1
make[1]: Leaving directory '/home/wirgi/quad/mwptools/horizon'
Makefile:9: recipe for target 'horizon' failed
make: *** [horizon] Error 2
wirgi@wirgi-VB:~/quad/mwptools$ `

stronnag, 2016-03-07:
Fixed in latest commit

wirgi, 2016-03-07:
I run this:   installing dependencies 
then
`wirgi@wirgi-VB:~/quad$ git clone https://github.com/stronnag/mwptools.git`
`wirgi@wirgi-VB:~/quad$ git clone https://github.com/cleanflight/blackbox-tools`
`wirgi@wirgi-VB:~/quad$ sudo apt-get install ruby-full`
`wirgi@wirgi-VB:~/quad/mwptools$ make`
`wirgi@wirgi-VB:~/quad/mwptools$ sudo make sysinstall`
```
wirgi@wirgi-VB:~/quad/mwptools$ mwp
pci id for fd 8: 80ee:beef, driver (null)
libGL error: core dri or dri2 extension not found
libGL error: failed to load driver: vboxvideo
```
```
2016-03-07T19:25:25+0100 mwp startup
2016-03-07T19:25:30+0100 libchamplain 0.12.11 may not draw maps at scale > 16
2016-03-07T19:25:30+0100 Consider downgrading, upgrading or building from source
2016-03-07T19:26:00+0100 Running blackbox_decode failed (is it on the PATH?)
Failed to execute child process "blackbox_decode" (No such file or directory)
```
where am I wrong?
Thanks

stronnag, 2016-03-07:
It would appear that you have not compiled and installed blackbox-tools

It may also be that virtualbox is not installed or configured correctly, as it doesn't appear to be able to load the video driver. I can't help with that.

wirgi, 2016-03-07:
Haiaiii! Sorry! true. I do not have compiled and installed blackbox-tools.
I make this?
~/quad/blackbox-tools make
~/quad/blackbox-tools sudo make install

stronnag, 2016-03-07:
I don't think there is a make install target for blackbox

$make
$ sudo cp obj/blackbox-decode /usr/bin/

stronnag, 2016-03-07:
Users of ArchLinux will now find a mwptools-git package in the AUR. This will install blackbox-tools-git as a dependency. 

At the moment, no documentation is installed, which some may consider a good thing.

wirgi, 2016-03-07:
ok! the bb-log window, is now opening up. 
now problem is zooming aerial map show  Design Camera
in the bb-log windows I can open normal txt log file?

arch linux users are super specialists ..of code.
it took me 17 minutes to find the terminal in ubuntu

stronnag, 2016-03-07:
@wirgi, i didn't understand the last comment 'Design Camera'. 
From the bblog window, you should be able to select a LOG*.TXT file, then select the index within the log, then OK. 
Can you try and explain the problem again?

wirgi, 2016-03-07:
in MapQuest Open Aerial   see Earth map but to  zoom more of 11 disappears Earth map .

stronnag, 2016-03-07:
I understand now! That is a limitation of open aerial. You can use OpenStreetMap to max 18, or Bing to 19. See the 'Quadkey proxy' section in the manual. That's as easy as I can make it.

wirgi, 2016-03-07:
"See the 'Quadkey proxy' section in the manual"  
which manual?

stronnag, 2016-03-07:
The mwptools manual
mwptools/docs/mwptools.pdf

stronnag, 2016-03-14:
closing this as the new makefile seems to be working

closed 2016-03-14 by stronnag