stronnag/mwptools #7: map sources - quadkey - bing maps

wirgi, 2016-03-08:
CIAO!   I can get some help here? **thanks**

"Define the “map-sources” key in the settings, this is the name of a JSON file in 
~/.config/mwp defining the map sources."
**this one?**   **which key, what value ?**
org.mwptools.planner.gschema.xml

```
<key name="map-sources" type="s">
      <summary>Additional Map sources</summary>
      <default>""</default>
      <description>JSON file defining additional map sources</description>
    </key>
```

"3 Ensure that the gsettings key 'map-sources' is set to allow user defined map sources;" **???**

"4 Ensure that the gsettings key 'quaduri' that defines the real quadkeys source is also set (there is 
no default) is shown below, #Q# indicates where the per tile quadkey value goes). "  **???**
**this one?**
```
<key name="quaduri" type="s">
      <summary>URI for the quadkey proxy</summary>
      <default>""</default>
      <description>A URI for the real source of quadkey tile. Use %s
      to indicate where the actual quadkey goes.</description>
    </key>
```

"6. Enable “Bing Proxy” as the map source, set it as default if you wish " ???

wirgi, 2016-03-09:
I got some progress 
Proxy is started but....
```
wirgi@wirgi-VB:~$ mwp
pci id for fd 8: 80ee:beef, driver (null)
libGL error: core dri or dri2 extension not found
libGL error: failed to load driver: vboxvideo
2016-03-09T02:13:40+0100 mwp startup
2016-03-09T02:13:40+0100 libchamplain 0.12.11 may not draw maps at scale > 16
2016-03-09T02:13:40+0100 Consider downgrading, upgrading or building from source
2016-03-09T02:13:40+0100 Starting proxy thread 
2016-03-09T02:13:40+0100 Invalid quadkeys URI (
http://a0.ortho.tiles.virtualearth.net/tiles/a#Q
#.png?g=131)
wirgi@wirgi-VB:~$
```

stronnag, 2016-03-09:
If the message is exactly as you've pasted it, then the quaduri gsettings value is invalid because it appears to have one or two newline characters embedded. You must paste exactly the value listed in the mwptools manual or the quaduri line from the gsettings sample file:
````
mwptools/samples/mwp_gsettings.sh
````
This file lists all the gsettings I use, obviously the home position default will not be useful to you.

wirgi, 2016-03-10:
ciao! thanks for the time that you've given me.
i finally run bingmap!  but I mixed up the file permissions.
if I run "mwp" proxy start and mwp works --bash report "Removing /home/wirgi/.config/mwp/.layout.xml 0"
if i run "sudo mwp" proxy not start and bash report "Removing /home/wirgi/.config/mwp/.layout.xml 0"
what am I doing wrong? 
I also installed arch linux Manjaro and mwp 
automatically installed without problems.

stronnag, 2016-03-11:
You really should not run things with sudo that do not need it, all that happens is that you corrupt the permissions in your home directory. The message is probably because you can no longer access your own files.

````
sudo chown -R wirgi:wirgi ~/
````
may fix it.

Thanks for confirming that the Arch (clone) install is clean.

stronnag, 2016-03-14:
Closing this **support** ticket. Please feel free to contact me via the email address in the manual on support matters.

closed 2016-03-14 by stronnag