stronnag/mwptools #8: replay_bbox_ltm.rb : `initialize': Bad file descriptor

DzikuVx, 2016-04-03:
I'm unable to run replay_bbox_ltm.rb on Linux Mint 17.3 with Xfce (Debian)

Tried both ruby2.0 and ruby 2.2. Every time I get
```
replay_bbox_ltm.rb --fd 18 -i 1 -t 3 -f /home/pawel/Downloads/BBlog sobota 2.TXT
```
```
/usr/bin/replay_bbox_ltm.rb:374:in `initialize': Bad file descriptor (Errno::EBADF)
	from /usr/bin/replay_bbox_ltm.rb:374:in `new'
	from /usr/bin/replay_bbox_ltm.rb:374:in `<main>'
```

All other tools seems to be working fine

stronnag, 2016-04-03:
I'd guess it's not escaping the spaces in the file name properly. Can
you try it with a name without spaces?

-jonathan

DzikuVx, 2016-04-03:
I also thought it might be it, but looks like not
```
pawel@Min ~ $ replay_bbox_ltm.rb --fd 18 -i 1 -t 3 -f /home/pawel/Downloads/bb2.TXT
/usr/bin/replay_bbox_ltm.rb:374:in `initialize': Bad file descriptor (Errno::EBADF)
	from /usr/bin/replay_bbox_ltm.rb:374:in `new'
	from /usr/bin/replay_bbox_ltm.rb:374:in `<main>'
```
File exists, blackbox explorer reads it

stronnag, 2016-04-03:
You can't just run --fd from the command line, because it needs fd 18
to exist. And there are multiple issues with file names with spaces
that I need to fix.

So:

$ exec 18<> /tmp/foo
$ bbox-replay/replay_bbox_ltm.rb --fd 18  -i 1 -t 3 -f  "/tmp/Test Log.TXT" 

will fail elsewhere.

-jonathan

DzikuVx, 2016-04-03:
OK, don't know what I did by replay started to work after restarting. Closing.

closed 2016-04-03 by DzikuVx

stronnag, 2016-04-03:
No, there was a bug in the replay script involving file names with spaces, fixed by cdf0199bb94a79c9010bc3ec1238edbdcbb16d9a

DzikuVx, 2016-04-03:
Well, I did not pulled yet :) Glad to been able to find a bug without finding it after all :)