stronnag/mwptools #9: Compile Error

ohhorob, 2016-04-04:
Fedora linux guest on VMWare Fusion (Mac OS X)

> libsoup-2.4.vapi:955.58-955.73: error: The type name `GLib.Cancellable' could not be found
> 		public abstract uint get_proxy_uri_sync (Soup.URI uri, GLib.Cancellable? cancellable, out Soup.URI proxy_uri);
> 		                                                       ^^^^^^^^^^^^^^^^
> Compilation failed: 90 error(s), 0 warning(s)
> Makefile:14: recipe for target 'qproxy' failed
> make[1]: *** [qproxy] Error 1
> make[1]: Leaving directory '/home/rob/mwptools/qproxy'
> Makefile:9: recipe for target 'qproxy' failed
> make: *** [qproxy] Error 2

ohhorob, 2016-04-04:
[make.log.txt](https://github.com/stronnag/mwptools/files/203223/make.log.txt)

[image]

stronnag, 2016-04-04:
Thanks for report. I'm surprised as Fedora 23 is one of my test beds. Let me fire up the VM.

ohhorob, 2016-04-04:
Checking the vala dependencies.

> $ sudo dnf install vala vala-devel vala-tools vala-doc
> [sudo] password for rob: 
> Last metadata expiration check performed 0:00:03 ago on Mon Apr  4 13:02:24 2016.
> Package vala-0.30.0-1.fc23.x86_64 is already installed, skipping.
> Package vala-devel-0.30.0-1.fc23.x86_64 is already installed, skipping.
> Package vala-tools-0.30.0-1.fc23.x86_64 is already installed, skipping.
> Package vala-doc-0.30.0-1.fc23.noarch is already installed, skipping.
> Dependencies resolved.
> Nothing to do.
> Complete!

stronnag, 2016-04-04:
That looks OK. It's building fom the expected source (thanks Ubuntu).
````
[jrh@fedora qproxy]$ make
valac --pkg libsoup-2.4 --pkg posix -o qproxy genproxy.vala
/home/jrh/Projects/mwptools/qproxy/genproxy.vala.c: In function ‘gen_proxy_default_handler’:
/home/jrh/Projects/mwptools/qproxy/genproxy.vala.c:537:12: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    _tmp29_ = _tmp28_->data;
            ^
[jrh@fedora qproxy]$ ls -l 
total 44
-rw-rw-r--. 1 jrh jrh  4733 Mar 13 18:48 genproxy.vala
-rw-rw-r--. 1 jrh jrh   380 Mar 13 18:48 Makefile
-rwxrwxr-x. 1 jrh jrh 24008 Apr  4 21:04 qproxy
-rw-rw-r--. 1 jrh jrh  4604 Mar 13 18:48 qproxy.vala
````
````
[jrh@fedora qproxy]$ valac --version
Vala 0.30.0
[jrh@fedora qproxy]$ pkg-config --modversion libsoup-2.4
2.52.2
````
I'm confused!

ohhorob, 2016-04-04:
```
$ git branch
* master
```

```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	cf-cli/cf-cli
	cf-cli/cf-cli-ui
	horizon/mwp_ath
	mspsim/mspsim
	mwp/mwp
	pidedit/pidedit
	samples/ublox-test/ublox-cli
	samples/ublox-test/ublox-geo
	switchedit/switchedit

nothing added to commit but untracked files present (use "git add" to track)
```

```
$ valac --version
Vala 0.30.0
```

```
$ pkg-config --modversion libsoup-2.4
2.52.1
```

ohhorob, 2016-04-04:
```
[rob@localhost mwptools]$ cd qproxy
[rob@localhost qproxy]$ make
valac --pkg libsoup-2.4 --pkg posix -o qproxy genproxy.vala
libsoup-2.4.vapi:103.38-103.59: error: The type name `GLib.SocketConnectable' could not be found
	public class Address : GLib.Object, GLib.SocketConnectable {
	                                    ^^^^^^^^^^^^^^^^^^^^^^

```

stronnag, 2016-04-04:
Yes. This is very strange it's internal to vala's dependency handling. Does
````
valac --pkg libsoup-2.4 --pkg posix --pkg gio-2.0 --pkg glib-2.0  -o qproxy genproxy.vala
````
change anything. It should not, as /usr/share/vala-0.30/vapi/libsoup-2.4.deps should define the gio dependency, and glib is a given.

ohhorob, 2016-04-04:
No errors!

```
[rob@localhost qproxy]$ valac --pkg libsoup-2.4 --pkg posix --pkg gio-2.0 --pkg glib-2.0  -o qproxy genproxy.vala
/home/rob/mwptools/qproxy/genproxy.vala.c: In function ‘gen_proxy_default_handler’:
/home/rob/mwptools/qproxy/genproxy.vala.c:537:12: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    _tmp29_ = _tmp28_->data;
            ^
[rob@localhost qproxy]$
```

stronnag, 2016-04-04:
And you have the file /usr/share/vala-0.30/vapi/libsoup-2.4.deps, which just contains 'gio-2.0' ?

ohhorob, 2016-04-04:
```
[rob@localhost qproxy]$ ls -alh /usr/share/vala-0.30/vapi/libsoup-2.4.deps
-rw-r--r--. 1 root root 8 Sep 18  2015 /usr/share/vala-0.30/vapi/libsoup-2.4.deps
[rob@localhost qproxy]$ cat /usr/share/vala-0.30/vapi/libsoup-2.4.deps
gio-2.0
[rob@localhost qproxy]$
```

stronnag, 2016-04-04:
And how about?
````
valac --pkg libsoup-2.4 --pkg posix --pkg gio-2.0 -o qproxy genproxy.val
````

ohhorob, 2016-04-04:
```
[rob@localhost qproxy]$ valac --pkg libsoup-2.4 --pkg posix --pkg gio-2.0 -o qproxy genproxy.vala
/home/rob/mwptools/qproxy/genproxy.vala.c: In function ‘gen_proxy_default_handler’:
/home/rob/mwptools/qproxy/genproxy.vala.c:537:12: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
    _tmp29_ = _tmp28_->data;
            ^
[rob@localhost qproxy]$
```

stronnag, 2016-04-04:
pushed a new Makefile with --pkg gio-2.0 as it's harmless

ohhorob, 2016-04-04:
```
git pull
make clean
make
```
No errors!

closed 2016-04-04 by ohhorob