stronnag/mwptools #10: Can't open mwp after rebuilding

ohhorob, 2016-05-02:
I pulled latest changes from git (tracking github master branch)

Peformed make clean, then make, and finally sudo make install.

Now when trying to run mwp from the "Activities" list in Gnome, or from terminal, the following error shows in the system log..

> [rob@localhost mwptools]$ ls -alh $(which mwp)
> -rwxr-xr-x. 1 root root 484K May  2 14:57 /usr/bin/mwp
> [rob@localhost mwptools]$ mwp
> 2016-05-02T16:35:51-0700 mwp startup
> 
> (mwp:4011): GLib-GIO-ERROR **: Settings schema 'org.mwptools.planner' does not contain a key named 'dwell-time'
> Trace/breakpoint trap (core dumped)
>

ohhorob, 2016-05-03:
Digging around a bit, I'm not sure what I've done wrong, or forgotten to do, but there is a stale schema (doesn't contain 'dwell-time') in my ~/.local/ path.

I deleted that and I MWP will open again.

stronnag, 2016-05-03:
So it's a feature. I should add a redundant key ever so often, just to weed out broken installations.

There's a script ```samples/clean-local.sh``` that will flush out any remnant of my misguided local install experiment.

closed 2016-05-03 by stronnag

ohhorob, 2016-05-04:
No worries, but what was broken about the installation? What would have I done originally to have that old schema there, and stopping me from launching after a `make sudo install`?

stronnag, 2016-05-04:
The deprecated 'make local' install would have installed the schema under .local. If this is found before the new version in /usr/share/ then it breaks.