stronnag/mwptools #11: Can't save preferences

stronnag, 2016-08-25:
via user HoChiMinh on RCG http://www.rcgroups.com/forums/showpost.php?p=35565467&postcount=80
"In Preferences/Device List, I deleted some default devices and added /dev/ttyUSB0 then Apply. 
But it does not appear in the Main screen. Exit and re-run mwp makes no difference. 
Did I do something wrong?"

Please tell me:

* OS 
* How you installed e.g. `make; sudo make install`
* The exact value you tried to set (may cut & paste it)
* The output of  (from a  terminal)
````
gsettings list-recursively org.mwptools.planner
````
If you in a terminal enter:
````
gsettings set org.mwptools.planner device-names "['/dev/ttyUSB0']"
````
what then happens when you run mwp?

HCM2000, 2016-08-27:
Ok This issue seems to have resolved.
What I did:
1. Open Edit/Preferences 
2. Make changes, ie. 
a. delete all devices in device list and added /dev/ttyUSB0 
b. change Def. Loiter = 15
c. change Def. Alt = 15
3. Click Apply
Look at main screen - nothing change i.e. full device list still shown and /dev/ttyUSB0 is not there and Def. Loiter = 30, Def. Alt = 20

Exit and rerun mwp. Nothing changed. All default values are shown.

Repeat steps 1,2,3. 
Then Open Edit/Preferences and click Apply again.
This time everything sticks!

OS: Ubuntu 16.04 LTS running  via VMware Player
System installed:  $ sudo make install

stronnag, 2016-08-27:
One point. I would not run mwp with sudo, there is no need to, and it may result in some areas of your home directory being owned by root and you don't have access, resulting in difficult to diagnose issues (like preferences not saving).

HCM2000, 2016-08-27:
Thanks.
If I run mwp without sudo then it will not connect to /dev/ttyUSB0.
When I click on Connect it pops up:
**Unable to open serial device: /dev/ttyUSB0
Reason: Bad file descriptor (9)**

stronnag, 2016-08-27:
Normally people hit that problem first in the configurator.

Your user needs to be a member of the group that owns the USB device, dialout on Ubuntu.
````
sudo usermod -a -G dialout <username>
````
Reopening, so I remember to update the error message.

HCM2000, 2016-08-27:
Sorry. I'm totally a dummy here.

Still get same error pop up when trying to connect to USB device.

hcm@tvm:~/Downloads/mwptools$ sudo usermod -a -G dialout hcm
hcm@tvm:~/Downloads/mwptools$ mwp
2016-08-27T13:33:14+1000 mwp startup version: local / 2016-08-13
2016-08-27T13:33:14+1000 libchamplain 0.12.13

(mwp:5065): GLib-WARNING **: /build/glib2.0-8EA4QF/glib2.0-2.48.1/./glib/giounix.c:410Error while getting flags for FD: Bad file descriptor (9)

stronnag, 2016-08-27:
You need to log out / log in for the group to be picked up.

stronnag, 2016-09-02:
closing, not an mwp issue --- and the error message has been updated

closed 2016-09-02 by stronnag