stronnag/mwptools #12: Mission not saved in flight controller after power off

HCM2000, 2016-08-27:
OS: Ubuntu 16.04 LTS running via VMware Player
System installed: $ sudo make install
FC: Naze32 rev6 full version
Firmware: inav 1.1 
Quadcopter is flying perfectly with Alt-Hold, Pos-Hold, RTH.

Here are the steps I did:
1. create mission
2. upload mission
3. message box "Mission Validated" appears
4. click ok
5. exit mwp, but FC remains connected to USB.
6. rerun mwp (sudo mwp)
7. connect to FC (successful. I can see flight view updated)
8. download mission
9. mission is shown in Tote

But, if I power off the FC i.e. unplug USB/battery, then reconnect USB/battery and try to download mission, nothing is shown in Tote.

It seems mission is lost after power recycled.

Thoughts please.

stronnag, 2016-08-27:
This is a feature of iNav (multiwii did saved missions in non-volatile memory, which was useful). The feature has been discussed but not implemented.

You can save the mission in mwp of course.

One point. I would not run mwp with sudo, there is no need to, and it may result in some areas of your home directory being owned by root and you don't have access, resulting in difficult to diagnose issues (like preferences not saving).

HCM2000, 2016-08-27:
I see. Thanks. So this means I will have to bring a laptop to the field, upload mission and fly. The problem is I don't have a laptop :-(

Re running mwp without sudo - it will not connect to /dev/ttyUSB0.
When I click on Connect it pops up:
**Unable to open serial device: /dev/ttyUSB0
Reason: Bad file descriptor (9)**

stronnag, 2016-08-27:
Please raise an issue on iNav about persistent missions :)

DzikuVx, 2016-08-27:
@stronnag @HCM2000 there is aleready iNavFlight/inav#79
It is even planned

stronnag, 2016-09-02:
closing, not an mwp issue

closed 2016-09-02 by stronnag