stronnag/mwptools #13: Is there a possibility for this program to run on OS X?

liutairan, 2016-12-30:
This tool looks really great, but is there a possibility to run on mac OS X or migrate it to OS X ? Thanks!

stronnag, 2016-12-30:
Some users have had success running mwp within an Ubuntu or Fedora VM using parallels. Unfortunately, the mapping layer and its dependencies (libchamplain, clutter) are not well supported on platforms other than Linux and FreeBSD, so the chances of a native port for either Windows or OSX is currently zero.

liutairan, 2016-12-30:
@stronnag  OK, I see. Thanks~

closed 2016-12-30 by liutairan

stronnag, 2016-12-30:
You're welcome to start another issue if you need help in getting it to work in a VM. Showing some evidence of having read the installation notes (in the docs directory) for the chosen distribution will help!

liutairan, 2016-12-30:
@stronnag  Thanks, I just got it work in VM. I got stuck with some error "VMware: vmw_ioctl_command error Invalid argument." but I finally solved that. It looks great!

stronnag, 2016-12-30:
Cool. Don't hesitate to ask questions in the RCG thread so other users can benefit from the answers. Posting in RCG about any OSX specific problems / workarounds would be welcome too.

liutairan, 2016-12-30:
Sure.