stronnag/mwptools #14: What is the leg speed for iNav

liutairan, 2017-01-04:
@stronnag Hi, I met a word "leg speed" when I read the manual, but I just don't understand what does it mean. I did some search online but almost found nothing about it. So what is that speed?

Thanks~

stronnag, 2017-01-04:
It's the speed for that leg of the mission. Normally, missions are flown at the speed given by the CLI parameter `set nav_max_speed = 500`. The leg speed setting overrides that (downwards); you cannot exceed it.

liutairan, 2017-01-04:
@stronnag  OK, I see now. Is there a way to set the real fly speed through mwptools? Or is iNav capable of flying with a specified speed?

Another question is some action attributes/message types are not supported for iNav, is that a problem of iNav or just not implemented in mwptools?

Thanks~

stronnag, 2017-01-04:
There will be a way, a new message will be in iNav 1.6 to support this. It was pushed to iNav development branch today, and will be in mwp by the weekend.

As for as attributes, mwp implements those that the considered useful by the author:), and are not better handled in the configurator. I'm not volunteering for the pain that Bart suffers in trying to keep ez-gui up with 3 different firmwares and supporting every possible configuration item. Suggestions are always welcome.

liutairan, 2017-01-04:
@stronnag  Great! Thank you so much for what you have done for this project. 

I was just trying to figure out what the limitation comes from. Actually I am working with a project with more complex things, so I still need to write some codes to develop new functions. Previously I met some problems with the ground control station, mainly about how to send out valid information to FC. I really learned a lot from your software. Thank you so much! 

I am currently using python to build up a GUI program to do the GCS job; I will post it to GitHub if I am able to finish it.

DzikuVx, 2017-01-04:
@liutairan also Configurator for INAV 1.6 will support more "navigation" settings

liutairan, 2017-01-04:
@DzikuVx Great, thank you~ Looking forward for it.

stronnag, 2017-01-04:
[image]

DzikuVx, 2017-01-04:
Nice!

closed 2017-01-05 by liutairan