stronnag/mwptools #15: mspsim seems not installed

liutairan, 2017-01-05:
Previously I successfully installed mwp, but seems like the other tools were not installed at all. 
I can start mwp, but mspsim only gives me "mspsim: command not found".
Then I tried to make it in mspsim folder, but got some error:

/tmp/ccIjM3TF.o: In function `mw_serial_setup_fd':
serial-device.vala.c:(.text+0x903): undefined reference to `set_fd_speed'
/tmp/ccIjM3TF.o: In function `mw_serial_open':
serial-device.vala.c:(.text+0xb36): undefined reference to `open_serial'
serial-device.vala.c:(.text+0xc7a): undefined reference to `get_error_text'
/tmp/ccIjM3TF.o: In function `mw_serial_close':
serial-device.vala.c:(.text+0x16ab): undefined reference to `close_serial'
/tmp/ccIjM3TF.o: In function `mw_serial_error_counter':
serial-device.vala.c:(.text+0x392): undefined reference to `flush_serial'
collect2: error: ld returned 1 exit status
error: cc exited with status 256
Compilation failed: 1 error(s), 0 warning(s)
Makefile:13: recipe for target 'mspsim' failed
make: *** [mspsim] Error 1

stronnag, 2017-01-05:
mea culpa, I decided to 'simplify' things by removed stuff that is rarely used. I will fix it.

liutairan, 2017-01-05:
I am using Ubuntu 16.04 in VM.

liutairan, 2017-01-05:
@stronnag You really reply fast~ I see. Thanks~

closed 2017-01-05 by liutairan

stronnag, 2017-01-05:
fixed in commit b2357f7