stronnag/mwptools #17: How to arm without the radio/ using wireless module?

liutairan, 2017-03-15:
Because I don't have a hand held radio controller and want to use the wireless module to do all the control, so I wonder how to arm the FC board and make it take off?

I am using a pair of xbees to do the wireless data exchange (MSP), and I am able to use xbee to get data from the FC board, upload missions, etc. In the past, I sent raw RC values to the board to replace the radio controller so that I could arm and take off. 

But to make it fly with waypoints, I still need to arm it, right? Does the process look like: 1. upload waypoints/ missions with mwp; 2. use rc arm it; 3. then the quadcopter take off by itself?

Then how to do this if I don't have the rc? I tried to send raw rc values after I upload missions, but still not able to arm it.

liutairan, 2017-03-15:
I know this is kind of off-topic, but just wish to get some answers of how other people doing it. Thanks~

liutairan, 2017-03-15:
I am using SP Racing F3 deluxe with ublox M8N GPS. Firmware is inav 1.5RC3 self compiled enabled MSP and sonar(but sonar not connected yet).
UART2 connected to GPS; UART3 connected to XBee with MSP settings.

stronnag, 2017-03-15:
There was a bug in some earlier version of iNav, maybe 1.5.x, where arm with MSP Rx didn't work. Try 1.6. 

Then you should be able to upload a mission, arm, then engage WP mode (via a switch / MSP Rx) and the machine will fly the mission.

stronnag, 2017-03-15:
And obviously, you still have to meet inav arming conditions (no. of satellites, level etc.)

liutairan, 2017-03-15:
@stronnag OK, I will try 1.6 first. In the configurator, all the pre-arming conditions are green, and I have 15-18 satellites connection.

liutairan, 2017-03-15:
@stronnag I tried inav 1.6.1. I think I still need to send out arm command by raw rc values all the times, otherwise failsafe will turn on immediately once I stopped sending out raw rc values. Is it possible to hold the last valid value and not go to failsafe so that I only need to send out arm command once?

liutairan, 2017-03-15:
Because wireless modules may not have enough bandwidth or processing speed to send out raw rc as while as getting data from FC. So to share the wireless module rather than use another radio controller, I guess I can only send out arm command once.

stronnag, 2017-03-15:
How are you arming, I would have thought for stick arm it would be a OK to hold it for a second, but I'm no expert on MSP_RX.

liutairan, 2017-03-15:
@stronnag  I use aux1 to arm, I use MSP send out raw rc values (8 values for roll, pitch, throttle, yaw, aux1, aux2, aux3, and aux4) to arm.

stronnag, 2017-03-15:
Then if it's a AUX switch, in "real life" it would have be be asserted in the 'on' position the whole time it's armed -- at least to my thinking.

liutairan, 2017-03-15:
Yeah, I think you are right. I will try the stick arm then. Thanks~

liutairan, 2017-03-15:
Well, I tried to use the stick arm, it can arm it, but after arm that I still need to have valid raw rc values send out, otherwise it will go to failsafe again. Feel frustrated. Thanks anyway, I will check other possibilities.

closed 2017-03-15 by liutairan

liutairan, 2017-03-15:
@stronnag I checked that page and tested. MSP_SET_RAW_RC needs to be sent every 0.48s in my test. Probably because of hardware or software reasons. But anyway, 0.48s is already enough for me and this is the "greatest news" I heard today. Lol. Thank you so much~