stronnag/mwptools #18: Ubuntu 16.04 compatibility?

mgeeforce, 2017-03-15:
Just tried to build on 16.04 LTS Desktop and got the following error:

`mwp.vala:1441.23-1441.46: error: The name `get_monitor_at_window' does not exist in the context of `Gdk.Display?'
            var mon = dp.get_monitor_at_window(window.get_window());
                      ^^^^^^^^^^^^^^^^^^^^^^^^
mwp.vala:1441.17-1441.67: error: var declaration not allowed with non-typed initializer
            var mon = dp.get_monitor_at_window(window.get_window());
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
mwp.vala:1442.24-1442.26: error: The name `mon' does not exist in the context of `MWPlanner.activate'
            var rect = mon.get_geometry();
                       ^^^
mwp.vala:1442.17-1442.41: error: var declaration not allowed with non-typed initializer
            var rect = mon.get_geometry();
                ^^^^^^^^^^^^^^^^^^^^^^^^^
mwp.vala:1443.29-1443.32: error: The name `rect' does not exist in the context of `MWPlanner.activate'
            conf.window_p = rect.width*80/100;
                            ^^^^`

Did install all the dependencies as per the doc. Thoughts?

stronnag, 2017-03-15:
I broke it when I moved from a deprecated GTK API (in Arch). I'll fix it ...

stronnag, 2017-03-15:
Fixed by https://github.com/stronnag/mwptools/commit/e57d43abf0bbd907a94933badef2fa921f444585

mgeeforce, 2017-03-15:
pulled, tested and issue is resolved. Thanks!

stronnag, 2017-03-15:
Confirmed fixed

closed 2017-03-15 by stronnag