stronnag/mwptools #20: mwp from AUR and qproxy

alfag4, 2017-04-05:
Hello, i have installed mwptools from AUR using yaourt and qproxy does not start when launch mwp. To have Bing Map running correctly i need manually run qproxy and then launch mwp.
From Terminal receive this:

* (mwp:702): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-QT2hp4Vzhm: Connection refused
2017-04-05T18:42:50+0200 mwp startup version: 3c32b1d / 2017-04-05
2017-04-05T18:42:51+0200 libchamplain 0.12.15
2017-04-05T18:42:51+0200 Forcing proxy offline [NMSTATE_UNKNOWN]
2017-04-05T18:42:51+0200 Starting proxy thread (offline)

stronnag, 2017-04-05:
You don't need qproxy as a separate process (it is handled on an internal thread now) for Bing.
You can still use it however, e.g. for 'other providers'. 

It works for me on Arch both ways (with / without qproxy). The line 'Starting proxy thread means it is using the internal thread (not qproxy).

Can you please supply the results of
````
gsettings get org.mwptools.planner map-sources
gsettings get org.mwptools.planner quaduri 
````
and the contents of the ~/.config/mwp/sources.json (or the name from the first gsettings command).

alfag4, 2017-04-05:
For me works only if using "qproxy <uri> <port>"
Below my settings. Thanks!

map-sources:
'sources.json'

quaduri:
'http://h0.ortho.tiles.virtualearth.net/tiles/h#Q#.png?g=131'

sources.json:
{
  "sources": [
    {
	"id": "BingProxy",
	"name": "Bing Proxy",
	"license": "(c) Microsoft Corporation, (c) NAVTEQ, (c) Image courtesy of NASA",
	"license_uri": "http://www.bing.com/maps/",
	"min_zoom": 0,
	"max_zoom": 19,
	"tile_size": 256,
	"projection": "MERCATOR",
	"uri_format": "http://localhost:21303/quadkey-proxy/#Z#/#X#/#Y#.png",
	"warning" : "The only user changeable part of the uri is the port number (21303)"
    }
  ]
}

stronnag, 2017-04-05:
That works here too (without qproxy). So what happens, do you get the OSM world or just a blank map?

As it thinks you're offline and won't go outside do you have a existing cache (I guess so if qproxy works)
````
du -sh ~/.cache/champlain/BingProxy/
````
Is there a network connection and it is managed by Network Manager?

stronnag, 2017-04-05:
I've added a gsetting key to disable polling network-manager for network status. 
````
gsettings set org.mwptools.planner ignore-nm true
````
It would be interesting to see if that fixes the problem (you'll need to reinstall via yaourt).

alfag4, 2017-04-05:
Perfect!! Now works well :)
Infact i haven't NetworkManager installed.

stronnag, 2017-04-05:
Strange that the Network Manager dbus API returns anything. I should ignore NM results if I get NMSTATE_UNKNOWN then.

alfag4, 2017-04-05:
I try to install NetworkManager and set to false the ignore-nm and works also.
No errors messages
* (mwp:904): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-0jqXCTWvvT: Connection refused
2017-04-05T20:16:59+0200 mwp startup version: 2193045 / 2017-04-05
2017-04-05T20:16:59+0200 libchamplain 0.12.15
2017-04-05T20:16:59+0200 Starting proxy thread

stronnag, 2017-04-05:
Thanks. The next update will probably make the ignore-nm redundant, but no harm in it staying. Thanks for testing the last version.

alfag4, 2017-04-05:
Thanks to you for the fast support :)

closed 2017-04-05 by stronnag