stronnag/mwptools #21: mwp should have a pause button during blackbox replay

giacomo892, 2017-04-06:
As for the title.

stronnag, 2017-04-06:
Indeed it should, this is not the first suggestion. Unfortunately as the replay is provided by a separate process, it's non-trivial to solve. We shall see, thanks for raising the issue.

stronnag, 2017-04-07:
Initial cut at this in 6d52e7473fe9f67b5ed79f073e56f7d5d8a375ee. Pressing <SPACE> during replay will pause / resume the replay,

giacomo892, 2017-04-08:
tested and it works.

stronnag, 2017-04-08:
Congratulations on upgrading to a modern distro.

giacomo892, 2017-04-08:
I've only tried it on VM 🍡 💃 👯

stronnag, 2017-04-09:
Closing as implemented and tested

closed 2017-04-09 by stronnag

giacomo892, 2017-04-09:
Maybe also adding a pause button is not a totally bad idea :)

stronnag, 2017-04-09:
If the main aim of the application was to replay log perhaps. 
But it's just a side effect.

giacomo892, 2017-04-09:
Is your mainly usage to upload missions and monitor them in real time?

stronnag, 2017-04-09:
The clue's in the name, so yes planning and monitoring; mwp log replay was late in the game, blackbox replay more so. Not to say they are not useful functions (probably more so than I expected), but not the main aim.