stronnag/mwptools #24: Can't get it to make ublox

davidflypei, 2017-10-13:
I've been trying to get this to install and it just keeps throwing this error.

david@david-VirtualBox:/usr/bin/mwptools-master$ make
make -C samples/ublox-test
make[1]: Entering directory '/usr/bin/mwptools-master/samples/ublox-test'
glib-compile-resources --target=ublox-resources.c --generate-source ublox.gresources.xml
can't write to file ublox-resources.cMakefile:21: recipe for target 'ublox-resources.c' failed
make[1]: *** [ublox-resources.c] Error 1
make[1]: Leaving directory '/usr/bin/mwptools-master/samples/ublox-test'
Makefile:8: recipe for target 'samples/ublox-test' failed
make: *** [samples/ublox-test] Error 2

stronnag, 2017-10-13:
I'd guess you've tried once as root and that previously created `ublox-resources.c`; now as a normal user you can't write it. Actually, as a normal user, building stuff in `/usr/bin` is not a recommended way of working.

May I suggest (as normal user (david)):
````
cd
git clone https://github.com/stronnag/mwptools.git
cd mwptools
make && sudo make install
````
Then:
````
sudo rm -rf /usr/bin/mwptools-master
````
If this doesn't work, add to this issue, otherwise please close it once you've built the project.

````

davidflypei, 2017-10-13:
Looks like that did the trick! I'm not fluent in Ubuntu. Thanks.

closed 2017-10-13 by davidflypei