stronnag/mwptools #25: Compile fails asking for gdk >= 3.22

Nandox7, 2017-10-14:
Version listed in the requirements is gtk >= 3.18 still when compiling it complains for a higher version.

mwp.vala:1684.23-1684.46: error: Gdk.Display.get_monitor_at_window is not available in gdk-3.0 3.18.9. Use gdk-3.0 >= 3.22
            var mon = dp.get_monitor_at_window(window.get_window());
                      ^^^^^^^^^^^^^^^^^^^^^^^^
mwp.vala:1685.24-1685.39: error: Gdk.Monitor.get_geometry is not available in gdk-3.0 3.18.9. Use gdk-3.0 >= 3.22
            var rect = mon.get_geometry();
                       ^^^^^^^^^^^^^^^^

Trying to compile in Elementary OS 0.4.1, that has 3.18 and won't receive 3.22 for quite some time.

stronnag, 2017-10-14:
I'll take a look at it. Unfortunately, newer OS (Arch, Fedora) need > 3.18 for wayland. I may have to install an Elementary OS VM in order to see how to do this cleanly.

Nandox7, 2017-10-14:
ElementaryOS has gtk 3.18.9  has you see in the error so it should be fine.
But from the error seems some methods are asking for >=3.22

stronnag, 2017-10-14:
No, it's too old. I have updated the documentation.

stronnag, 2017-10-14:
As this is c. the unsupported Ubuntu 14.04 era, you may find that
````
git checkout last_14.04_build && make && sudo make install 
````
will work.

stronnag, 2017-10-14:
OK, seems quite simple to sort out. I need to check that the fix doesn't break the other supported systems, after I've been flying ... Check back in a few hours.

[image]

Nandox7, 2017-10-14:
Ah cool I'll try that.
If you have a patch I can try as well. :)

stronnag, 2017-10-14:
Fixed by 3273e9a

closed 2017-10-15 by stronnag