stronnag/mwptools #26: Telemetry broken when 6 GPS Satellites fixed

b14ckyy, 2017-10-23:
When GPS has fixed 6 or more satellites, Complete telemetry is broken. All Data modules freeze and activity light stops blinking. Reproducible with iNav 1.7.3 on SPRacing F3 Deluxe, SPRacing F3 Acro and Naze32 rev6. Same with different baud rates. Also tested on 1 PC and also one VM with Ubuntu 17.10
It seems to work with iNav 1.8-RC1  but only was able to get 7 satellite in my flat. 

Should be investigated if MWP or iNav Bug.

stronnag, 2017-10-23:
I very much doubt this is a mwp issue. But please post your iNav CLI diff.

b14ckyy, 2017-10-23:
I used the SPRacing F3 Acro for verifying bug (Deluxe is used in my Camera Copter) and reflashed with full Erase. So it is iNav 1.7.3 completely in factory state, with only GPS on UART3 with 115200 or 57600 baud and activated uBlox Protocol without ground station support. 

# diff

# version
# INAV/SPRACINGF3 1.7.3 Sep  9 2017 / 10:20:09 (974440df)

# resources

# mixer

# servo

# servo mix

# feature
feature GPS

# beeper

# map

# name

# serial
serial 2 2 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# master
set acc_hardware = MPU6050
set mag_hardware = NONE
set baro_hardware = NONE
set pitot_hardware = NONE

# profile
profile 1

#

stronnag, 2017-10-23:
Not helpful without telemetry settings

b14ckyy, 2017-10-23:
Update:
After attaching the FC with iNav 1.8 and flashing back to 1.7.3, it worked fine with 8 Satellite fix. But after close and reopen MWP, the same issue is back. 

Will it help to send dump? 
Using only MSP telemetry. Same issue btw over USB and also 3DR radio.

`# serial
serial 0 1 115200 38400 0 115200
serial 1 0 115200 38400 0 115200
serial 2 2 115200 115200 0 115200

# master
set looptime = 2000
set gyro_sync = OFF
set gyro_sync_denom = 2
set align_gyro = DEFAULT
set gyro_hardware_lpf = 42HZ
set gyro_lpf_hz = 60
set moron_threshold = 32
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 1
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 1
set vbat_adc_channel = 1
set rssi_adc_channel = 3
set current_adc_channel = 2
set airspeed_adc_channel = 0
set acc_notch_hz = 0
set acc_notch_cutoff = 1
set align_acc = DEFAULT
set acc_hardware = MPU6050
set acc_lpf_hz = 15
set acczero_x = 0
set acczero_y = 0
set acczero_z = 0
set accgain_x = 4096
set accgain_y = 4096
set accgain_z = 4096
set rangefinder_hardware = NONE
set align_mag = DEFAULT
set mag_hardware = NONE
set mag_declination = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0
set mag_calibration_time = 30
set baro_hardware = NONE
set baro_use_median_filter = ON
set pitot_hardware = NONE
set pitot_use_median_filter = ON
set pitot_noise_lpf =  0.600
set pitot_scale =  1.000
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_invert = OFF
set rc_smoothing = ON
set serialrx_provider = SPEK1024
set sbus_inversion = ON
set rx_spi_rf_channel_count = 0
set spektrum_sat_bind = 0
set rx_min_usec = 885
set rx_max_usec = 2115
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SPIFLASH
set min_throttle = 1150
set max_throttle = 1850
[... 125 lines cut ...]
set telemetry_inversion = ON
set frsky_default_latitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = METRIC
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set smartport_uart_unidir = OFF
set smartport_fuel_percent = OFF
set ibus_telemetry_type = 0
set ltm_update_rate = NORMAL
set ledstrip_visual_beeper = OFF
set i2c_speed = 400KHZ
set cpu_underclock = OFF
set debug_mode = NONE
set acc_task_frequency = 500
set attitude_task_frequency = 250
set async_mode = NONE
set throttle_tilt_comp_str = 0
set input_filtering_mode = OFF
set mode_range_logic_operator = OR
set stats = OFF
set stats_total_time = 0
set stats_total_dist = 0`

stronnag, 2017-10-23:
No, because it's not a mwp issue

b14ckyy, 2017-10-23:
OK. so I'll wait for 1.8 or use RC1.

stronnag, 2017-10-23:
fixed in d3acd29

stronnag, 2017-10-23:
It was an mwp bug, tried to use an too new MSP message with old firmware. Thanks for finding it.

closed 2017-10-23 by stronnag