stronnag/mwptools #28: Unexpected removal of .layout.xml breaks the dock

stronnag, 2017-11-14:
Via user Alle_Ita on RCG https://www.rcgroups.com/forums/showpost.php?p=38633222&postcount=212

stronnag, 2017-11-14:
This may take a while ....

stronnag, 2017-11-14:
OK, so lets start be seeing what's happening in the configuration directory. Please post the output of:
````
ls -lR ~/.config/mwp
````

AlleIta, 2017-11-14:
Here the output:
/home/alle/.config/mwp:
totale 4
-rw-rw-r-- 1 alle alle 437 nov  7 15:43 sources.json

stronnag, 2017-11-14:
Sorry, my mistake:
````
ls -lRa ~/.config/mwp

````

AlleIta, 2017-11-14:
/home/alle/.config/mwp:
totale 16
drwxrwxr-x  2 alle alle 4096 nov 14 22:08 .
drwx------ 18 alle alle 4096 nov  7 15:51 ..
-rw-rw-r--  1 alle alle   37 nov 14 22:08 .layout.xml
-rw-rw-r--  1 alle alle  437 nov  7 15:43 sources.json

stronnag, 2017-11-14:
Good. `.layout.xml` should be much bigger, like about 3kB. Can you attach it please.

AlleIta, 2017-11-14:
I changed the extension to .txt otherwise it is not possible to attach the file.
Please remove the .txt
[.layout.xml.txt](https://github.com/stronnag/mwptools/files/1472611/default.layout.xml.txt)

stronnag, 2017-11-14:
Thanks. This is a puzzle, more tomorrow.

stronnag, 2017-11-14:
So. if I remove my .layout.xml and use yours (which contains no data), I get:
````
2017-11-14T21:46:56+0000 Removing /home/jrh/.config/mwp/.layout.xml 0
````
as expected, and then this:
[image]

All the dock icons are in the vertical bar, I can click them to open them  in the dock and close the items I don't want (so they return to the View menu). Then if I use File/Quit, the layout is saved (destroying the window does not save the layout, this is a feature).
````
$ ls -lstra ~/.config/mwp/.layout.xml ## note the file size ....
4 -rw-r--r-- 1 jrh jrh 3895 Nov 14 21:52 /home/jrh/.config/mwp/.layout.xml
````
And I have my favourite layout again.
[image]

AlleIta, 2017-11-14:
When you use my .layout.xml, does the item appear in the view menu by closing it in the dock?
Actually, if I close an item in the dock then I can not open it anymore in the view menu.
I did not tried to file/quit after closing an item in the dock....I have only tried to destroy the windows to be sure to have all the items back in the dock.

stronnag, 2017-11-15:
Yes, it does. So why don't you try the attached [layout.xml.txt](https://github.com/stronnag/mwptools/files/1475596/layout.xml.txt)
````
cp layout.xml.txt ~/.config/mwp/.layout.xml
`````

AlleIta, 2017-11-15:
I tried the your file attached and this is what happen.
If I launch mwp, I can see your layout and in the view menu I can see the items you disabled.
I can enable them from the view menu.
If I enable them from the view menu, as soon as I close the items they disappears and they do not appear anymore in the view menu....and I come back at my initial situation.
[.layout.xml.tar.gz](https://github.com/stronnag/mwptools/files/1476346/default.layout.xml.tar.gz)

Enclosed you can find the .layout.xml after disabling the battery monitor and File->quit.

AlleIta, 2017-11-15:
I did other tests and if I disable an item in the dock, the item does not appear in the view menu.
If I file->quit and open again mwp the item appears in the view menu.
Should it works in this way?

stronnag, 2017-11-15:
No, your system is somewhat broken. Is it native or in a VM. Non of my machines / VMs on Arch, Fedora, FreeBSD, Ubuntu or Elementary (I thing that's the lot) behave in that way.

AlleIta, 2017-11-15:
it is a native system. I have a double partition (windows 10 and Ubuntu 16.04 LTS).
Now if I replace the .layout.xml with my original one, I do not have anymore the:
- 2017-11-14T22:04:15+0100 Removing /home/alle/.config/mwp/.layout.xml 0
How I can fix it and let mwp behave how it should? all the dependencies are installed. I checked them and made a git pull && make && sudo make install

stronnag, 2017-11-15:
I really have no idea. What desktop are you using ? Unity ?

AlleIta, 2017-11-15:
The one installed with Ubuntu 16.04 LTS

alle@Portatile:~$ ls -l /usr/share/xsessions/
totale 4
-rw-r--r-- 1 root root 204 ago 22  2016 ubuntu.desktop

stronnag, 2017-11-15:
I just changed by Ubuntu 16.04 VM from gnome to ubuntu-desktop. Makes no difference, mwp behaves as designed. I'm really out of ideas of what to try next.

AlleIta, 2017-11-16:
Do you think that by removing completely mwptools and build and install it again from new could solve this issue? If you think that is worth a try, can you please tell me how to completely remove mwp?

stronnag, 2017-11-16:
Not yet, I have a few things to try later.

Fyi, there is an uninstall script. I think the INSTALL file describes where it is, samples maybe.

AlleIta, 2017-11-16:
OK, waiting for the next things to try.

stronnag, 2017-11-16:
OK, first how it works at the moment. mwp is set up so that when you open the 'View' menu, the application is signalled and sets the dock items' status. This appears to work for every mwp user on multiple OS, apart from you for some reason.

So please do the following:
````
git pull
git checkout gdl-test
git pull # probably not necessary ....
make && sudo make install
````
Then run mwp from a terminal (if necessary please use my `.layout` from yesterday) so there is a mixture of open / icon / closed dock items. Don't change the dock, just open the View menu. Then quit mwp and send me the terminal log. It should look like:
````
$ mwp
2017-11-16T17:14:05+0000 mwp startup version: mwp local / 2017-11-16 0.320.620
2017-11-16T17:14:05+0000 default options: --rings 50,20
2017-11-16T17:14:05+0000 Using speech api 1 [espeak]
2017-11-16T17:14:05+0000 libchamplain 0.12.16
2017-11-16T17:14:05+0000 Starting external qproxy process
2017-11-16T17:14:05+0000 Starting proxy thread
2017-11-16T17:14:11+0000 setting status from menu view
````
The last line is from the View menu signal. I'd like to know if you see it.

Then please run mwp again, as below, with the `--dock-test` flag. This disables the 'View' clicked signal and instead updates the menu every time the dock changes (somewhat unnecessarily, as just moving a dock item causes the layout change event to file).

So in this second test, please:
* Close one of the dock items (the battery from my layout would be fine)
* Check the menu, if the battery item is enabled, please add it back to the dock bar and then the dock

Again, please collect the terminal log for me, and also describe what you see in the menu. The log should look something like:
````
$ mwp --dock-test
2017-11-16T17:15:03+0000 mwp startup version: mwp local / 2017-11-16 0.320.620
2017-11-16T17:15:03+0000 default options: --rings 50,20
2017-11-16T17:15:04+0000 Using speech api 1 [espeak]
2017-11-16T17:15:04+0000 libchamplain 0.12.16
2017-11-16T17:15:04+0000 Starting external qproxy process
2017-11-16T17:15:04+0000 Starting proxy thread
2017-11-16T17:15:20+0000 setting status from layout change
2017-11-16T17:15:27+0000 setting status from layout change
2017-11-16T17:15:27+0000 setting status from layout change
2017-11-16T17:15:32+0000 setting status from layout change
````

stronnag, 2017-11-16:
OK, it's an Ubuntu bug. If you have the app menu set (one menu at the top), the menu update event is not sent. If you have normal menus (menu in the window) then it works correctly (like it does not every other OS). I'm not fixing this. You can fix Ubuntu's stupidity by adding to `/etc/environment`:
````
UBUNTU_MENUPROXY=0
````

stronnag, 2017-11-16:
Closing this issue as it's an Ubuntu bug.

closed 2017-11-16 by stronnag

AlleIta, 2017-11-16:
Ok. 
I do not have one menu at the top. 
I have the menu in the window and not the app menu set.
In the /etc folder I can not find any environment folder.
I tried: sudo echo "UBUNTU_MENUPROXY=0" >> /etc/environment
but I have "bash: /etc/environment: permission denied"

stronnag, 2017-11-16:
````
sudo -i
echo "UBUNTU_MENUPROXY=0" >> /etc/environment
````

stronnag, 2017-11-16:
Looks like a menu at the top in the screen shot

AlleIta, 2017-11-16:
still "bash: /etc/environment: permission denied"

AlleIta, 2017-11-16:
sorry, it was my fault.
I succed with two steps....
- sudo -i
- echo "UBUNTU_MENUPROXY=0" >> /etc/environment

When I start mwp, still the same behaviour

stronnag, 2017-11-16:
You at least log out and log in

AlleIta, 2017-11-16:
Ops...sorry, my second fault. I'm newbie.
I did a reboot and it works perfectly.
Thank you very much for your help and patience.
I appreciate.
Thank you again.

stronnag, 2017-11-16:
No problem. Actually, I probably will provide a workaround.

stronnag, 2017-11-16:
With the latest commit, you should be able to build mwp as:
````
DOPTS="-D BROKEN_MENUS" make && sudo make install   
````
and it should  work even with Ubuntu's global app menu (so you can remove `UBUNTU_MENUPROXY=0` from `/etc/environment`)

AlleIta, 2017-11-17:
OK....super!
Thank you

stronnag, 2017-11-17:
Later, there will be a completely automatic (detected at run time) fix

stronnag, 2017-11-17:
It's now fixed (I think). No special branches, no strange build commands.
````
git checkout master # one off, just in case
git pull && make && sudo make install
````

AlleIta, 2017-11-17:
Ok, thank you.
I will test if it works on my computer.
Can you please tell me how to delete the UBUNTU_MENUPROXY=0" in the /etc/environment?

stronnag, 2017-11-17:
Just edit with any text editor. There's a very simple editor called nano on Ubuntu.
Remove the line, CTRL-O (^O) to save the file, CTRL-X (^X) to exit. Done.

AlleIta, 2017-11-17:
Ok, it works great on my computer!
Thank you