stronnag/mwptools #29: mwp should have an option to forward telemetry

stronnag, 2017-11-21:
as https://www.rcgroups.com/forums/showpost.php?p=38671422&postcount=224

stronnag, 2017-11-21:
This should take the usual mwp device definition, `/dev/ttyUSB0@19200`, `udp://localhost:4567` etc,

stronnag, 2017-11-21:
@markandkymward

There is now a branch 'forward-to'
````
git pull
git checkout forward-to
git pull
make && sudo make install
````
Then

`mwp --forward-to /dev/ttyUSB42` # assuming `/dev/ttyUSB42` is the forwarding device
(any mwp device name, so)
````
mwp --forward-to /dev/ttyUSB17@4800
mwp --forward-to udp://wol:3000 # for those of us with computers named after 'pooh' characters
````

* Only forwards LTM (for now)

markandkymward, 2017-11-22:
back to getting that segmentation fault.  troubleshooting...

markandkymward, 2017-11-22:
pi@raspberrypi:~/git/mwptools $ mwp
2017-11-21T21:22:52-0500 mwp startup version: mwp 3518622 / 2017-11-21 0.325.753
2017-11-21T21:22:52-0500 default options: --rings 10,20 -S 64 --centre 41.2141,-73.0502 --serial-device /dev/ttyUSB0@19200 --connect /dev/ttyUSB0@19200 
2017-11-21T21:22:53-0500 Using speech api 1 [espeak]
2017-11-21T21:22:54-0500 libchamplain 0.12.15

markandkymward, 2017-11-22:
pi@raspberrypi:~/git/mwptools $ gsettings list-recursively org.mwptools.planner
org.mwptools.planner default-map ''
org.mwptools.planner wdw-p 512
org.mwptools.planner led '#60ff00'
org.mwptools.planner audio-bearing-is-reciprocal false
org.mwptools.planner mavrth ''
org.mwptools.planner vlevels ''
org.mwptools.planner gpsintvl uint32 2000
org.mwptools.planner speech-api 'espeak'
org.mwptools.planner auto-follow true
org.mwptools.planner checkswitches false
org.mwptools.planner font-fv 12
org.mwptools.planner default-layout ''
org.mwptools.planner mavph ''
org.mwptools.planner show-scary-warning false
org.mwptools.planner default-loiter uint32 30
org.mwptools.planner uilang ''
org.mwptools.planner display-dms false
org.mwptools.planner poll-timeout uint32 500
org.mwptools.planner set-head-is-b0rken false
org.mwptools.planner max-home-delta 2.5
org.mwptools.planner quaduri ''
org.mwptools.planner ah-size 32
org.mwptools.planner fctype 'auto'
org.mwptools.planner atstart ''
org.mwptools.planner rth-autoland true
org.mwptools.planner default-zoom uint32 17
org.mwptools.planner ah-invert-roll false
org.mwptools.planner wdw-h 1056
org.mwptools.planner wpmodifier3 uint32 0
org.mwptools.planner dwell-time uint32 300
org.mwptools.planner ignore-nm false
org.mwptools.planner wpmodifier uint32 1
org.mwptools.planner espeak-voice 'en'
org.mwptools.planner heartbeat ''
org.mwptools.planner atexit ''
org.mwptools.planner audio-on-arm true
org.mwptools.planner compat-version '2.3 pre8'
org.mwptools.planner default-altitude uint32 8
org.mwptools.planner default-nav-speed 2.5
org.mwptools.planner media-player 'paplay'
org.mwptools.planner display-distance uint32 1
org.mwptools.planner wdw-w 1920
org.mwptools.planner rings-colour '#ffffff20'
org.mwptools.planner update-interval 1.0
org.mwptools.planner device-names ['/dev/3dr', '/dev/rfcomm1', '/dev/mega2560', '/dev/ttyACM0@115200', '/dev/pts/0']
org.mwptools.planner log-path ''
org.mwptools.planner auto-restore-mission false
org.mwptools.planner default-longitude -73.046999999999997
org.mwptools.planner baudrate uint32 115200
org.mwptools.planner mission-path ''
org.mwptools.planner default-latitude 41.215499999999999
org.mwptools.planner log-on-arm false
org.mwptools.planner log-save-path ''
org.mwptools.planner display-speed uint32 3
org.mwptools.planner speak-interval uint32 1
org.mwptools.planner stats-timeout uint32 30
org.mwptools.planner map-sources ''
org.mwptools.planner tote-float-p true

stronnag, 2017-11-22:
Interesting bug. Here it just hangs. `---connect` connects to the first device in the gsettings `device-names` list; in your list its `/dev/3dr`, which is unlikely to exist --- I should sort out the defaults. I think you meant `--auto-connect`, which connects to the device on the command line. This is really confusing; there should only be `connect` option, which should behave sanely.

stronnag, 2017-11-22:
22cd652 in the `forward-to` branch may sort this out. Looks like it was trying to use something that may be have been initialised :(

markandkymward, 2017-11-23:
That works.

stronnag, 2017-11-23:
Thanks. Looking forward to learning how well `--forward-to` works with the Ghetto station.

markandkymward, 2017-11-23:
I’ll report back. It’s in pieces waiting on the arduino. I’d really like to be able to pull telemetry from my Taranis   but this will work. I’m looking for a solution that’ll exceed the range of the 3DRs. I have a TBS crossfire on order. The crossfire offers a Bluetooth out option for multiple protocols including Mavlink. 

Mark Ward
Chief Pilot, Sikorsky Stratford Flight Center
w: 203-386-3397
c: 203-414-3485

stronnag, 2017-11-23:
Another cheap, should be better than 3DR / HC-12 option: LoRa https://github.com/stronnag/mwptools/wiki/Using-LoRa-for-iNav-Telemetry

stronnag, 2017-11-25:
I merged this into master, we can deal with any issues there, and deleted the `forward-to` branch, so you will need to `git checkout master` next time you want to pull / build.

Some time ago (the other open issue) I requested (from anyone) a Taranis telemetry dump; I'd be delighted to receive the same and implement this is mwp.

markandkymward, 2017-11-25:
I took a look at the LoRa device you suggested.  Looks impressive.  Bought 2.  I’d be glad to help with the Taranis dump.   How best to do that?

From: stronnag
Sent: Thursday, November 23, 2017 2:47 PM
To: stronnag/mwptools
Cc: markandkymward; Mention
Subject: Re: [stronnag/mwptools] mwp should have an option to forwardtelemetry (#29)

Another cheap, should be better than 3DR / HC-12 option: LoRa https://github.com/stronnag/mwptools/wiki/Using-LoRa-for-iNav-Telemetry

markandkymward, 2017-11-25:
Oh, and also – I’d be glad to help with the taranis dump.  Not sure how to do that.

From: stronnag
Sent: Saturday, November 25, 2017 1:51 PM
To: stronnag/mwptools
Cc: markandkymward; Mention
Subject: Re: [stronnag/mwptools] mwp should have an option to forwardtelemetry (#29)

I merged this into master, we can deal with any issues there, and deleted the forward-to branch, so you will need to git checkout master next time you want to pull / build.
Some time ago (the other open issue) I requested (from anyone) a Taranis telemetry dump; I'd be delighted to receive the same and implement this is mwp.

stronnag, 2017-11-26:
https://github.com/stronnag/mwptools/issues/19

markandkymward, 2017-11-26:
sorted out my serial comms issues.  bluetooth is a real bear on the RPI.  I reread your post above and realized why I wasn't getting data - you only transmit LTM, therefore my copter has to be armed to send data in my config.  when I armed it on the bench the data streamed freely.  building ghettostation this week.  I have everything all printed out, just need to assemble and troubleshoot.  I'll look into the taranis dump also.

stronnag, 2017-11-26:
As far as I know, the GhettoStation onlu uses LTM, hence the forward policy; currently it forwards all LTM, which may (or may not) confuse the GS, as iNav LTM is a superset of standard LTM. Depending on how the GS behaves, the future (user selectable) options could be:

* Forward all LTM (extant policy)
* Forward GS compatible LTM 
* Forward GS compatible MSP & LTM
* Forward everything

Let me know if you need any changes; this is an interesting project. A write up on the iNav or mwp wiki would be appreciated when its working.

stronnag, 2017-12-02:
There is now 'forward` gsetting:
[image]

| Key | Usage |
| ---- | ---- |
| none | | 
| LTM |  all LTM messages |
| minLTM | LTM understood by GhettoStation (G,A,S frames) |
| all |  all messages (LTM and MSP) |

markandkymward, 2017-12-05:
Works great on the ghetto station.  I’ve connected it via Bluetooth and  I was scratching my head over —forward-to /dev/rfcomm0. Turns out MWP  didn’t want to do that. A quick ln -s /dev/rfcomm0 /dev/tyyS4 (or whatever) did the trick. I moved off of the raspberry pi in favor of the Asus tinkerboard and I have a very usable field workstation. I’ll give more details on the inav/ mwp wiki once I have the project completed and field tested.

stronnag, 2017-12-05:
Excellent. There's no reason why mwp should not like /dev/rfcomm0 as the serial port handling is common across forwarding and normal telemetry reception. You should also be able to use the Blue Tooth address of the BT device, either work here for my particular device:
````
mwp --forward-to 00:14:03:11:35:16 
````
and
````
mwp --forward-to /dev/rfcomm1@9600 
````

markandkymward, 2017-12-09:
It does.  I had pooched my settings somewhere along the line.  Connects like a magnet on all of the above.  Waiting on snow right now so no flying for a bit.  The craft>mwp>ghettostation connection appears very stable.  I'm looking into my new LoRA commlink now.

stronnag, 2017-12-16:
closing, as feature complete, for now.

closed 2017-12-16 by stronnag