stronnag/mwptools #33: mwp mission upload error

amitmanish, 2018-01-12:
Hello stronnag, I am trying to implement mission planning using mwptool but always get error saying "number of waypoints(4) exceeds maximum(0) ". I am using naze32 rev6 board with inav 1.6 version.

stronnag, 2018-01-12:
My trivial answer is iNav 1.6 is really old; my slightly less trivial answer is that this was fixed recently, at least for iNav 1.73.

* What build of mwp are you running ? (from help/about)
* Let me try a 1.6 hex (huum, technical archaeology:)

amitmanish, 2018-01-13:
Thanks for quick reply. I will update inav firmware and try again.

stronnag, 2018-01-13:
Your choice, either update mwptools, which will now work with 1.6, or update the firmware (which is probably a good idea anyway, a lot has improved since 1.6).

amitmanish, 2018-01-14:
Thanks again.

closed 2018-01-14 by amitmanish