stronnag/mwptools #35: MWP --forward-to second bluetooth device appears in-op TBS Crossfire source, GhettoStation receiver.

markandkymward, 2018-02-04:
Data Source - Taranis 9XD with Crossfire transmitter.  Bluetooth device on CRSF is set to MAVEmu.  MWP ground station is connected to CRSF via bluetooth device (/dev/rfcomm0). MWP receives and interprets the mavlink data perfectly.  "Forward-To" appears to connect to GhettoStation BT device (/dev/rfcomm1) OK, but it appears that no data is forwarded.  BT device was checked outside of the data-loop for proper serial link via serial monitor using "echo and cat."  BT link is 9600 baud.  Multiple baud rates and settings were attempted via changes to the BT device (HC-05).    Tried gsettings set org.mwptools.planner forward "minLTM."    When run, logging  similar to the following is output.  Again, MWP receives and processes the info OK.

2018-02-04T16:55:32-0500 Connected /dev/rfcomm0
2018-02-04T16:55:32-0500 set forwarder /dev/rfcomm1
2018-02-04T16:55:32-0500 LTM/Mavlink mode
2018-02-04T16:55:32-0500 Comm error count 5
2018-02-04T16:55:32-0500 MAVCRC Fail, got b58a != 8121 [1 c8] (cmd=147, len=36)
2018-02-04T16:55:32-0500 init icon 0
2018-02-04T16:55:32-0500 Armed 0
2018-02-04T16:55:32-0500 Using espeak for speech
2018-02-04T16:55:32-0500 Comm error count 6
2018-02-04T16:55:32-0500 MAVCRC Fail, got 5709 != eced [33 44] (cmd=109, len=9)
2018-02-04T16:55:32-0500 Comm error count 7
2018-02-04T16:55:32-0500 MAVCRC Fail, got 985e != acf5 [1 c8] (cmd=147, len=36)
2018-02-04T16:55:33-0500 Comm error count 8
2018-02-04T16:55:33-0500 MAVCRC Fail, got be62 != 8ac9 [1 c8] (cmd=147, len=36)
2018-02-04T16:55:34-0500 Comm error count 9
2018-02-04T16:55:34-0500 MAVCRC Fail, got 38de != c75 [1 c8] (cmd=147, len=36)
2018-02-04T16:55:34-0500 Comm error count 10
2018-02-04T16:55:34-0500 MAVCRC Fail, got 3511 != 1ba [1 c8] (cmd=147, len=36)
2018-02-04T16:55:35-0500 Comm error count 11
2018-02-04T16:55:35-0500

stronnag, 2018-02-04:
mwp doesn't forward MAVLINK (but MSP and LTM). It could do; due to the way that mwp does forwarding (it rewrites each message) it will need a bit of thought,but it's easily doable. 

I'll have a look at the MAVLink that Ghettostation consumes, and see what would be useful to forward.

2 CRCs / second is more that I'd expect, but not catastrophic. 

Leave it with me.

stronnag, 2018-02-04:
And mwp doesn't know about MAVlink message 147, and MAVlink has a RC seed for each message;so it's not surprising that it errors each 147. So really the error rate is acceptably low. I'll sort that out too.

Any chance you can capture a stream of mavlink off the CRSF to see exactly that it's outputting ?

markandkymward, 2018-02-04:
Hex dump.  CRSF BT set to MAVemu. GPS not receiving signal.

0000000 7f17 09fe 3389 6d44 0000 0000 a1b6 0002
0000020 3900 fecc 8a09 4433 006d 0000 b600 5f9f
0000040 0000 a4f7 09fe 338b 6d44 0000 0000 9db6
0000060 005a c200 fea2 8c09 4433 006d 0000 b600
0000100 56a1 0000 e690 1ffe 018d 01c8 0000 0000
0000120 0000 0000 0000 8f00 c801 001e 0000 e300
0000140 9ba5 c3bc 2a64 4dbc 3815 0040 0000 0000
0000160 0000 0000 0000 2000 feed 9024 c801 9d93
0000200 000a ff00 ffff ffff 007f 0000 0000 0000
0000220 0000 0000 0000 0000 0000 0000 0000 0000
0000240 0000 9c28 fe94 9109 4433 006d 0000 b600
0000260 51a1 0000 0556 09fe 3392 6d44 0000 0000
0000300 a1b6 004c 4600 fece 9309 4433 006d 0000
0000320 b600 45a1 0000 49cf 09fe 3394 6d44 0000
0000340 0000 a1b6 003e 1800 fea3 9509 4433 006d
0000360 0000 b600 399f 0000 3fba 09fe 3396 6d44
0000400 0000 0000 a1b6 0033 4300 24d5 fe36 9809
0000420 4433 006d 0000 b600 2b9d 0000 b1d5 09fe
0000440 3399 6d44 0000 0000 a1b6 0026 5100 9ce7
0000460 c801 001e 0000 e300 9ba5 c3bc 2a64 4dbc
0000500 3815 0040 0000 0000 0000 0000 0000 f100
0000520 fe0b 9d24 c801 9d93 000a ff00 ffff ffff
0000540 007f 0000 0000 0000 0000 0000 0000 0000
0000560 0000 0000 0000 0000 0000 c928 feb7 9e09
0000600 4433 006d 0000 b600 20a3 0000 cac4 0000
0000620 32fa 09fe 33a0 6d44 0000 0000 9db6 0018
0000640 e100 fe44 a109 4433 006d 0000 b600 00a1
0000660 ff00 fe59 a209 4433 006d 0000 b600 0e9f
0000700 0000 2786 09fe 33a3 6d44 0000 0000 9fb6
0000720 000b 3b00 fe37 a409 4433 006d 0000 b600
0000740 05a3 0000 4a1f 09fe 33a5 6d44 0000 0000
0000760 a1b6 0064 2200 a9a7 c801 001e 0000 e300
0001000 9ba5 c3bc 2a64 4dbc 3815 0040 0000 0000
0001020 0000 0000 0000 9300 fe28 aa24 c801 9d93
0001040 000a ff00 ffff ffff 007f 0000 0000 0000
0001060 0000 0000 0000 0000 0000 0000 0000 0000
0001100 0000 f528 fe91 ab09 4433 006d 0000 b600
0001120 60a1 0000 34c0 09fe 33ac 6d44 0000 0000
0001140 96b6 005d 3700 fe67 ad09 4433 006d 0000
0001160 b600 579f 0000 9417 09fe 33ae 6d44 0000
0001200 0000 9db6 0051 3b00 fe4a af09 4433 006d
0001220 0000 b600 4aa1 0000 9484 09fe 33b0 6d44
0001240 0000 0000 9fb6 0043 bd00 fe91 b109 4433
0001260 006d 0000 b600 419d 0000 c0a9 09fe 33b2
0001300 6d44 0000 0000 9db6 003f 9000 b5b7 c801
0001320 001e 0000 e300 9ba5 c3bc 2a64 4dbc 3815
0001340 0040 0000 0000 0000 0000 0000 4000 fe14
0001360 b624 c801 9d93 000a ff00 ffff ffff 007f
0001400 0000 0000 0000 0000 0000 0000 0000 0000
0001420 0000 0000 0000 0000 9e28 fe89 b709 4433
0001440 006d 0000 b600 3e9f 0000 30f0 09fe 33b8
0001460 6d44 0000 0000 9fb6 003d d900 958a fe72
0001500 ba09 4433 006d 0000 b600 3ca3 0000 c857
0001520 09fe 33bb 6d44 0000 0000 9fb6 0038 b000
0001540 8067 09fe 33bd 6d44 0000 0000 9fb6 0033
0001560 c800 fea9 be09 4433 006d 0000 b600 329f
0001600 0000 364d 09fe 33bf 6d44 0000 0000 9fb4
0001620 0030 8700 c279 c801 001e 0000 e300 9ba5
0001640 c3bc 2a64 4dbc 3815 0040 0000 0000 0000
0001660 0000 0000 3400 fe2e c324 c801 9d93 000a
0001700 ff00 ffff ffff 007f 0000 0000 0000 0000
0001720 0000 0000 0000 0000 0000 0000 0000 0000
0001740 9b28 fe1d c409 4433 006d 0000 b600 2d9d
0001760 0000 732d 09fe 33c5 6d44 0000 0000 9fb4
0002000 002b 8300 fe58 c609 4433 006d 0000 b600
0002020 279f 0000 44df 09fe 33c7 6d44 0000 0000
0002040 9db6 0025 cb00 7815 fe3c c909 4433 006d
0002060 0000 b400 229f 0000 bd3a 09fe 33ca 6d44
[... 118 lines cut ...]
0005640 f249 09fe 333f 6d44 0000 0000 a1b6 0054
0005660 6e00 fea9 4409 4433 006d 0000 b600 529f
0005700 0000 0bc7 09fe 3345 6d44 0000 0000 9fb4
0005720 0050 0a00 fe44 4609 4433 006d 0000 b600
0005740 4ba1 0000 79b8 baa6 09fe 09fe 334a 6d44
0005760 0000 0000 9fb6 0043 d300 fe35 4b09 4433
0006000 006d 0000 b600 3e9f 0000 9af3 09fe 334c
0006020 6d44 0000 0000 9fb6 0039 5500 4fbd c801
0006040 001e 0000 e300 9ba5 c3bc 2a64 4dbc 3815
0006060 0040 0000 0000 0000 0000 0000 4100 fe4f
0006100 5024 c801 9d93 000a ff00 ffff ffff 007f
0006120 0000 0000 0000 0000 0000 0000 0000 0000
0006140 0000 0000 0000 0000 0728 fe14 5109 4433
0006160 006d 0000 b400 35a1 0000 23b9 09fe 3352
0006200 6d44 0000 0000 9fb6 0031 5400 fe6f 5309
0006220 4433 006d 0000 b600 28a1 0000 9e15 fe8f
0006240 5509 4433 006d 0000 ab00 209f 0000 b7a6
0006260 0000 2d58 09fe 0157 00c8 0000 f395 09fe
0006300 3358 0044 7b00 fe2c 5909 4433 006d 0000
0006320 b600 189f 0000 7539 015c 1ec8 0000 0000
0006340 a5e3 bc9b 64c3 bc2a 154d 4038 0000 0000
0006360 0000 0000 0000 0000 a990 24fe 015d 93c8
0006400 0a9d 0000 ffff ffff 7fff 0000 0000 0000
0006420 0000 0000 0000 0000 0000 0000 0000 0000
0006440 0000 2800 3752 09fe 335e 6d44 0000 0000

stronnag, 2018-02-05:
Perfect, thanks.

stronnag, 2018-02-05:
As of now there's a test branch with (mwpid 1.036.603, commit bcdcbab7) that works if you use /dev/rfcommX. Damn, I need to pay attention when you tell me stuff doesn't work, you can't currently use BT addresses. Actually, the whole comms sub-system is based on the premise there's only one serial device. I need to fix that next.

Now, here I have
Tricopter, unconditional mavlink => 3DR => /dev/rfcomm1
Computer1, /dev/rfcomm1 forwarded to /dev/rfcomm2

/dev/rfcomm2 is bridged to a HC-12 radio

Computer 2 has a USB connected HC-12 radio.

Same picture on both computers, via four radios.

stronnag, 2018-02-05:
Forwarding to a BT address is now disabled; device nodes (/dev/rfcommX) work OK.

In order to forward Mavlink, it is necesary to set the forward key to either 'all' or 'minMav' (currently synonymous)
````
$ gsettings range  org.mwptools.planner forward
enum
'none'
'LTM'
'minLTM'
'minMAV'
'all'
````
I'm not convinced that 9600 is fast enough for Mavlink; I see data loss / backlog on the final screen in the above chain (the 3DR is 19200, the HC-12 is 9600), whereas with a faster forward link, the two screens update simultaneously (this is using iNav's Mavlink telemetry as the source). 

Also, in your sample, there is  > 15% data loss whereas at 19200 I got just one CRC error in more than an hour's testing.

From your sample:
````
$ ./hxreader mavex.txt # simple test tool for parsing a mav hexdump
...
OK   (109 6d) 9 68 33 44
Fail (109 6d) 9 69 440a 440d :  fe 09 45 33 44 6d 00 00 00 00 b4 9f 50 00 00 0a 44
...
135 msgs:  good 111,  bad 24
````

stronnag, 2018-02-05:
merged into development ... so you should use that
````
git pull
git checkout development
git pull
make && sudo make install
````

markandkymward, 2018-02-06:
Stronnag - Thanks for this.  I have to pause testing the GhettoStation setup because I seem to have trashed yet another HC05 serial bluetooth device.  Another pile is on order.  I did manage to get output transferred to my second computer before the HC05 died.  Those devices are so finicky.
I'll be back to you after the postman does his thing.

markandkymward, 2018-02-06:
I found my old stash of HC05s.  I really need to clean up my shop.  I verified that I am indeed getting connectivity via the --forward-to.  BT baud rate set at 19200 and used --forward-to /dev/rfcommx@19200 switch.  Set gsettings org.mwptools.planner forward all.  Ghettostation TM input rate set at 19200.  Tried all applicable protocols, multiwii, lighttelemetry and mavlink.  No joy on connecting.  Same setup was previously working when receiving LTM data via 3DR radio.  Looks like either the "MAVEmu" data packets aren't quite Mavlink or something is happening in the MWP forwarding?

stronnag, 2018-02-06:
Please try the latest development commit. A few improvement, including BT forwarding working with BT addresses as well as /dev/rfcommX device nodes, tolerance of the forward device failing / not keeping up and retrying opening the forwarding device  if it fails.

btw: your earlier note that using a BT address for forwarding caused mwp to lock up or slow down was most likely a symptom of the the link being too slow and blocking everything, which is now fixed.

If you're still having problems with 1.037.686 (023080b) or later, please post the mwp console output here.

stronnag, 2018-02-06:
Also, as of 1.037.686 (023080b), gsettings forward 'minMav' only sends the mavlink understood by ghetto-station, while 'all' will forward everything.

markandkymward, 2018-02-06:
'minMAV' did the trick.  Works like a champ. In summary - ground station running a tinkerboard w/tinker OS.  Receives telemetry data from short range bridge (fancy term for bluetooth) on the TBS Crossfire. TBS Crossfire is configured to output MAVemu data.  This is on /dev/rfcomm0.  Ghettostation antenna with HC05 BT adapter is connected to the groundstation via /dev/rfcomm1.  mwp --forward-to /dev/rfcomm2 command line.  So as long as I have Crossfire connectivity, I have valid gps position data being fed to the tracker.

stronnag, 2018-02-12:
closing, as the specific dev branch is now merged into master

closed 2018-02-12 by stronnag