stronnag/mwptools #36: mwp should have a remote control (dbus) interface

stronnag, 2018-02-13:
Currently, mwp's dbus server only does one thing, load a buffer of XML as a mission (for mwp-area-planner 'publish'). 

This capability should be expanded to (e.g.)

* Upload / verify a mission
* start / stop serial comms
* other stuff

stronnag, 2018-02-14:
First pass ...

````
# ask mwp to list the serial devices it knows about, returns an array of device names, here there are three.
$ samples/mwp-dbus-test.sh get_devices
   array [
      /dev/ttyUSB0      00:14:03:11:35:16 3dr-bridge      30:14:12:02:16:64 HC12-bridge   ]
#  ask mwp if it's connected, false means it's not ...
$ samples/mwp-dbus-test.sh connection_status
      boolean false
# instruct mwp to connect to the first device from the device list, the return status of true means it succeeded
$ samples/mwp-dbus-test.sh connect_device /dev/ttyUSB0
   boolean true
# instruct mwp to open a mission file from the file system, returns the number of points in the mission (0 if it fails).
$ samples/mwp-dbus-test.sh load_mission  ~/missions/nm_small_field.mission
   uint32 10
# instruct mwp to upload the mission to the FC, the return value is the number of  way points **validated **. 
# If the return is 0, the upload was not valid (serial disconnected, no mission loaded in mwp)
# if the return is < 0, then the upload failed or failed to validate.  
$ samples/mwp-dbus-test.sh upload_mission false
   int32 10
````

stronnag, 2018-02-14:
And here's the ruby example (as in `samples/mwp-dbus.rb`)

````ruby
#!/usr/bin/ruby
require 'dbus'

# Create bus and service object
bus = DBus::SessionBus.instance
service = bus.service("org.mwptools.mwp")
mwp = service.object("/org/mwptools/mwp")

# Test if it's up.
# Rather than abort, we could start an instance of mwp if appropriate
#
begin
  pif = mwp["org.freedesktop.DBus.Peer"]
  pif.Ping
rescue
  abort "Service unavailable"
end

# Set the default interface
mwp.default_iface = "org.mwptools.mwp"

# Get the devices known to mwp
devs = mwp.GetDevices
devstr = devs[0].join(", ")
puts "Devices #{devstr}"

# See if we're connected to th FC, for the Connect* methods, the final
# item in the returned array is the connection status. For
# consistency, we can therefore test constat[-1]

constat = mwp.ConnectionStatus
if constat[-1]
  puts "Connected to #{constat[1]}"
else
  puts "Not connected, trying first device ... #{devs[0][0]}"
  constat = mwp.ConnectDevice(devs[0][0])
  puts "Connected state now #{constat[-1]}"
end

# If a mission file was given load it and upload to the FC
if ARGV[0] and File.exists?(ARGV[0])
  npts = mwp.LoadMission ARGV[0]
  puts "Loaded mission from #{ARGV[0]}, #{npts[0]} mission points"
  if constat[-1]
    nwpts = mwp.UploadMission true
    puts "Uploaded #{nwpts[0]} waypoints to the FC"
  end
end

# dump out the interface definitions
#puts mwp.introspect
````

stronnag, 2018-02-16:
merged into master, closing

closed 2018-02-16 by stronnag

ashearin, 2018-03-08:
So I've been attempting to utilize this, I have mwp housed on an ubuntu server and am trying to get the python script to work. I'm not sure if I am not using it right (still learning linux in general) but when I use the command:

mwptools/samples/mwp-dbus.py test.mission

I get the error:

org.freedesktop.DBus.Error.NoServer: Failed to connect to socket /tmp/dbus-UGZwyWh6Ei: Connection refused

I've played around a bit and have the xserver working, but I'm not sure why it doesn't give me the results of the script onto the shh terminal screen. What am I doing wrong?

stronnag, 2018-03-09:
This problem happens on older dbus implementations due to the need to access the session dbus that's actually used by mwp. This is solved automagically on newer distros (e.g. Arch, Fedora 27). On ancient Ubuntu, you can use something like the following script to find the required session bus:
````
if [[ -n $SSH_CLIENT ]]; then
  # Choose something that will be running the same session as mwp
  # the desktop manager (gnome-shell, compiz, xfce, budgie etc) is a good
  # choice
  APP=compiz # for ancient Ubuntu with Unity desktop
 BUS=$(cat /proc/$(pgrep $APP -u "$USER")/environ | grep -z "^DBUS_SESSION_BUS_ADDRESS=")
 # echo $BUS
 export $BUS
fi
````

* Choose a app you know will be running 
* As this will only apply to ssh sessions, you can put it in `.bashrc`.

Otherwise, save the above to a file, e.g. `$HOME/getbus.sh`, then
````
$ source $HOME/getbus.sh
$ env | grep DBUS
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-46NaEK7rYU
$ ~/Projects/mwptools/samples/mwp-dbus.py ./area.mission
Loaded ./area.mission with 17 mission points
...
````

stronnag, 2018-03-09:
Note that the above ~hack~ workaround is only necessary for Ubuntu 16.04; 17.04 does not need it.

ashearin, 2018-03-13:
I'm still getting the same error when running mwp-dbus.py with a test file, I'm not sure if my dbus settings are messed up, but when I run env | grep DBUS, there's no output. I tried putting the code snippet you provided in .bashrc (and skipping the source $HOME/getdbus. sh command) as well as in its own file and had that same results. I'm Including the terminal output below. 

Upon opening connection:
declare -x DISPLAY="localhost:11.0"
declare -x HOME="/home/ubuntu"
declare -x LANG="en_US.UTF-8"
declare -x LOGNAME="ubuntu"
declare -x MAIL="/var/mail/ubuntu"
declare -x OLDPWD
declare -x PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
declare -x PWD="/home/ubuntu"
declare -x SHELL="/bin/bash"
declare -x SHLVL="1"
declare -x SSH_CLIENT="132.170.212.20 55768 22"
declare -x SSH_CONNECTION="132.170.212.20 55768 172.31.22.93 22"
declare -x SSH_TTY="/dev/pts/2"
declare -x TERM="xterm"
declare -x USER="ubuntu"
declare -x XDG_DATA_DIRS="/usr/local/share:/usr/share:/var/lib/snapd/desktop"
declare -x XDG_RUNTIME_DIR="/run/user/1000"
declare -x XDG_SESSION_ID="164"

$ env | grep DBUS 
$ (no output)

$ mwptools/samples/mwp-dbus.py ./fake.mission
org.freedesktop.DBus.Error.NoServer: Failed to connect to socket /tmp/dbus-FH69KfXlqZ: Connection refused

I appreciate your help

ashearin, 2018-03-19:
I apologize if this is a dumb question, how do I find/choose an appropriate app to use? used ps -ef to get this list:

UID        PID  PPID  C STIME TTY          TIME CMD
root         1     0  0 Mar09 ?        00:00:05 /sbin/init
root         2     0  0 Mar09 ?        00:00:00 [kthreadd]
root         3     2  0 Mar09 ?        00:00:01 [ksoftirqd/0]
root         5     2  0 Mar09 ?        00:00:00 [kworker/0:0H]
root         7     2  0 Mar09 ?        00:00:00 [rcu_sched]
root         8     2  0 Mar09 ?        00:00:00 [rcu_bh]
root         9     2  0 Mar09 ?        00:00:00 [migration/0]
root        10     2  0 Mar09 ?        00:00:04 [watchdog/0]
root        11     2  0 Mar09 ?        00:00:00 [kdevtmpfs]
root        12     2  0 Mar09 ?        00:00:00 [netns]
root        13     2  0 Mar09 ?        00:00:00 [perf]
root        14     2  0 Mar09 ?        00:00:00 [xenwatch]
root        15     2  0 Mar09 ?        00:00:00 [xenbus]
root        17     2  0 Mar09 ?        00:00:00 [khungtaskd]
root        18     2  0 Mar09 ?        00:00:00 [writeback]
root        19     2  0 Mar09 ?        00:00:00 [ksmd]
root        20     2  0 Mar09 ?        00:00:01 [khugepaged]
root        21     2  0 Mar09 ?        00:00:00 [crypto]
root        22     2  0 Mar09 ?        00:00:00 [kintegrityd]
root        23     2  0 Mar09 ?        00:00:00 [bioset]
root        24     2  0 Mar09 ?        00:00:00 [kblockd]
root        25     2  0 Mar09 ?        00:00:00 [ata_sff]
root        26     2  0 Mar09 ?        00:00:00 [md]
root        27     2  0 Mar09 ?        00:00:00 [devfreq_wq]
root        30     2  0 Mar09 ?        00:00:00 [kswapd0]
root        31     2  0 Mar09 ?        00:00:00 [vmstat]
root        32     2  0 Mar09 ?        00:00:00 [fsnotify_mark]
root        33     2  0 Mar09 ?        00:00:00 [ecryptfs-kthrea]
root        49     2  0 Mar09 ?        00:00:00 [kthrotld]
root        50     2  0 Mar09 ?        00:00:00 [bioset]
root        51     2  0 Mar09 ?        00:00:00 [bioset]
root        52     2  0 Mar09 ?        00:00:00 [bioset]
root        53     2  0 Mar09 ?        00:00:00 [bioset]
root        54     2  0 Mar09 ?        00:00:00 [bioset]
root        55     2  0 Mar09 ?        00:00:00 [bioset]
root        56     2  0 Mar09 ?        00:00:00 [bioset]
root        57     2  0 Mar09 ?        00:00:00 [bioset]
root        58     2  0 Mar09 ?        00:00:00 [bioset]
root        59     2  0 Mar09 ?        00:00:00 [bioset]
root        60     2  0 Mar09 ?        00:00:00 [bioset]
root        61     2  0 Mar09 ?        00:00:00 [bioset]
root        62     2  0 Mar09 ?        00:00:00 [bioset]
root        63     2  0 Mar09 ?        00:00:00 [bioset]
root        64     2  0 Mar09 ?        00:00:00 [bioset]
root        65     2  0 Mar09 ?        00:00:00 [bioset]
root        66     2  0 Mar09 ?        00:00:00 [bioset]
root        67     2  0 Mar09 ?        00:00:00 [bioset]
root        68     2  0 Mar09 ?        00:00:00 [bioset]
root        69     2  0 Mar09 ?        00:00:00 [bioset]
root        70     2  0 Mar09 ?        00:00:00 [bioset]
root        71     2  0 Mar09 ?        00:00:00 [bioset]
root        72     2  0 Mar09 ?        00:00:00 [bioset]
root        73     2  0 Mar09 ?        00:00:00 [bioset]
root        74     2  0 Mar09 ?        00:00:00 [scsi_eh_0]
root        75     2  0 Mar09 ?        00:00:00 [scsi_tmf_0]
root        76     2  0 Mar09 ?        00:00:00 [scsi_eh_1]
root        77     2  0 Mar09 ?        00:00:00 [scsi_tmf_1]
root        82     2  0 Mar09 ?        00:00:00 [ipv6_addrconf]
root        83     2  0 Mar09 ?        00:00:00 [bioset]
root        96     2  0 Mar09 ?        00:00:00 [deferwq]
root       248     2  0 Mar09 ?        00:00:00 [raid5wq]
root       280     2  0 Mar09 ?        00:00:00 [bioset]
root       304     2  0 Mar09 ?        00:00:00 [jbd2/xvda1-8]
root       305     2  0 Mar09 ?        00:00:00 [ext4-rsv-conver]
root       363     2  0 Mar09 ?        00:00:00 [kworker/0:1H]
root       379     2  0 Mar09 ?        00:00:00 [iscsi_eh]
root       385     2  0 Mar09 ?        00:00:00 [ib_addr]
[... 22 lines cut ...]
root      1161     1  0 Mar09 ?        00:00:12 /usr/lib/accountsservice/account
root      1191     1  0 Mar09 ?        00:00:00 /sbin/mdadm --monitor --pid-file
root      1193     1  0 Mar09 ?        00:00:00 /usr/lib/policykit-1/polkitd --n
root      1236     1  0 Mar09 tty1     00:00:00 /sbin/agetty --noclear tty1 linu
root      1243     1  0 Mar09 ttyS0    00:00:00 /sbin/agetty --keep-baud 115200
root      2137     2  0 Mar09 ?        00:00:00 [xfsalloc]
root      2138     2  0 Mar09 ?        00:00:00 [xfs_mru_cache]
ubuntu   12371     1  0 Mar13 ?        00:00:00 /lib/systemd/systemd --user
ubuntu   12373 12371  0 Mar13 ?        00:00:00 (sd-pam)
root     12625     1  0 Mar13 ?        00:00:00 /usr/bin/dbus-daemon --fork --pr
ubuntu   12677     1  0 Mar13 ?        00:00:00 /usr/bin/dbus-daemon --fork --pr
root     12700     1  0 Mar13 ?        00:00:00 /usr/bin/dbus-daemon --fork --pr
root     13388     1  0 Mar13 ?        00:00:00 /usr/bin/dbus-daemon --fork --pr
root     13464     1  0 Mar13 ?        00:00:00 /usr/bin/dbus-daemon --fork --pr
root     18163     2  0 Mar15 ?        00:00:05 [kworker/0:1]
root     26355     2  0 Mar18 ?        00:00:00 [kworker/u30:1]
root     27119     2  0 01:19 ?        00:00:00 [kworker/u30:2]
root     27588     2  0 06:38 ?        00:00:00 [kworker/0:0]
root     28720  1140  0 16:52 ?        00:00:00 sshd: ubuntu [priv]
ubuntu   28774 28720  0 16:52 ?        00:00:00 sshd: ubuntu@pts/0
ubuntu   28777 28774  0 16:52 pts/0    00:00:00 -bash
ubuntu   28875 28777  0 17:15 pts/0    00:00:00 ps -ef

I've tried a few things to choose an appropriate app, entered in a few: PID's and UID's, but I always get something along the lines of 
                         cat: /proc//environ: No such file or directory

stronnag, 2018-03-19:
In order to remotely control mwp, a precondition is that mwp is running in an X session on the console. I don't any evidence that there is (a) an X server or (b) a user session.

If the original question that started this thread was really not about mwp, but more generally "I need to be  able to upload a mission to iNav on a headless server" then please clarify what you're asking (that would be a valid question, to which the answer is "it's a trivial bit of coding"). So what's the real requirement ? don't be shy about asking.

ashearin, 2018-03-19:
So after a quick google search, I'd say I need to be able to upload a mission to iNav on a headless server, but I'll describe a little more about our situation and hopefully clear things up a little:

We have a webapp and associated database, and a headless ubuntu VM provided by AWS (where we have MWP installed) and our hope was to be able to upload MWP mission scripts onto a locally connected drone from any PC with internet access.

stronnag, 2018-03-19:
OK, soon there will be a mwp tool to upload a mission to an connected FC, it's will be called `muppet`, something like:
````
$ muppet --mission xmlmission-file.xml --device /dev/ttyUSB0
```` 
If you don't give it a device name, it will auto-discover it.
The `xmlmission-file.xml` is a standard mwp / ezgui XML mission file (or mwp JSON mission file).

Does that do what you need?

ashearin, 2018-03-19:
It does, I really appreciate the help.

stronnag, 2018-03-19:
Continued in #38, somewhat experimental

Conclusion 1 for your school project:

* Specifying software requirements is hard and needs careful thought and precise definition