stronnag/mwptools #37: How to setup Serial over Wifi

b14ckyy, 2018-03-05:
Because I don't need long range telemetry on all vehicles (like planes) I prefer to use a ESP8266 Wireless Module to connect with my laptop or smartphone wirelessly. 

On android with INAV Mission Planner it works out of the box and absolutely perfect. With my Windows Laptop  I need an extra tool to connect a virtual COM-port over WiFi but this also works fine to use INAV Configurator. 

Unfortunately I was not able yet to connect this virtual COM port stable to VMWare and use it on my Ubuntu VM with mwptools. I also have another laptop (an old netbook) only with ubunto and specially configured for mwptools only but I was not able to get the Serial Port connected over Wifi and use it with mwptools. 

I found this manual for all different systems that told to use "socat" for this but I just don't understand this 2-line manual. My Linux skills are like.... a Windows User? :D 

It would be great if you could make a manual for Linux dummies like me to get a connection over Wifi with an ESP8266.

stronnag, 2018-03-05:
You should not need socat (that fakes up a serial port for the configurator or CLI). mwp talks TCP/IP directly (both TCP and UDP). There is a somewhat detailed description in the iNav wiki https://github.com/iNavFlight/inav/wiki/iNavFlight-Missions#esp8266-1

If you need more help, can you please tell me which how the ESP8266 is configured? I have one somewhere with the vendor firmware on it.

b14ckyy, 2018-03-05:
I used the Manual from here: http://ez-gui.com/manual/multiwii-clearflight-wifi-to-ezi-gui-how-to/
Firmware is jeelabs ESP-Link: https://github.com/jeelabs/esp-link
Not sure atm if i used release or alpha. I flashed a few ESP8266 a few months ago for a friend and he gave me one of them. 
The manual you sent is only talking about UDP to ESP-Link. I'll try this tomorrow. How to connect over TCP? I don't have any experience with UDP and don't even know what the parts of the address mean: udp://:14014/esp-air:14014

stronnag, 2018-03-05:
if you're using ESP-Link firmware, then the first part of the wiki link tells you how to do that, just set the device on mwp to `tcp://host:port`; the wiki link quotes *exactly* the same example as the ezgui example, the mwp device for that would be `tcp://192.168.4.1:23`.

b14ckyy, 2018-03-05:
ah okay it is the same command. 
ok i will try this tomorrow. this should then also work through the VM if NAT bridge is enabled. 
Thank you very much :)

stronnag, 2018-03-05:
Sure, the above pre-supposes that you've successfully created the wifi link and can:
* Ping the device and
* Access it via `nc` or `telnet` as described on the github page (you should be able to get the CLI in this way).

stronnag, 2018-03-05:
And yes, it should work from the VM with the NAT bridge.

b14ckyy, 2018-03-06:
I tried it with TCP Connection right Now. But MWP crashes immediately. Tested on Ubuntu 1604 LTS on my Netbook and also in VMware with Ubuntu 17.10. On 16.04 it Shows me an crash Report the first time but I don't know how to Export it. Ping works fine.

stronnag, 2018-03-06:
works just fine here.

Please run mwp from a terminal and paste all the output here.

b14ckyy, 2018-03-06:
`2018-03-06T11:22:05-0800 Using Xwayland for safety:)
2018-03-06T11:22:05-0800 mwp startup version: mwp 8e7c69e / 2018-02-20 1.047.684
2018-03-06T11:22:05-0800 Using speech api 2 [speechd]
2018-03-06T11:22:06-0800 libchamplain 0.12.16
2018-03-06T11:22:26-0800 Starting Bing proxy 
2018-03-06T11:22:26-0800 Removing /home/marc/.config/mwp/.layout.xml 0
2018-03-06T11:22:41-0800 Try connect tcp://192.168.4.1:23
2018-03-06T11:22:41-0800 Connected tcp://192.168.4.1:23
2018-03-06T11:22:41-0800 Serial ready
2018-03-06T11:22:41-0800 set mrtype=14 cap =3c
2018-03-06T11:22:41-0800 Feature Mask [10402082] : telemetry false, gps true
Gleitkomma-Ausnahme (Speicherabzug geschrieben)
`
Log just stops here with Floating Point error

CLI Status Through Windows with Serial bridge: (Using INAV 1.7.3 because it is the last Version for my old NAZE32, my both SPR F3 FC still in shipping)

`# status
System Uptime: 564 seconds
Voltage: 0 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, GPS=OK
Stack size: 2048, Stack address: 0x20005000
I2C Errors: 0, config size: 1420, max available config: 2048
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 6, cycle time: 2015, PID rate: 496, RX rate: 49, System rate: 9

# `

stronnag, 2018-03-06:
Thanks. It's not due to the serial bridge. It may be because you're running wayland 
`2018-03-06T11:22:05-0800 Using Xwayland for safety:)`
(running wayland rather than X on something as old as Ubuntu is asking for trouble), or it could be something related to NAZE32 and an old iNav version. Let me have a look, I have a naze32 somewhere.

b14ckyy, 2018-03-06:
This was in VM with Ubuntu 17.10.
Just done the same on 16.04 and this line does not appear. I think then it uses X? 
Anyway it Fails at the same Point. Unfortunately my F4 FC from my race quad is fried so cannot test with it. And I have only a 20mm Micro FC and a brushed FC laying around but both don't have UART breakouts. Only Rx Pins for Receiver. I'll ask a friend to bring me one of his SPR F3 at work tomorrow so I can do some more Tests with 1.7.3 and also 1.9 to see if it is my Setup or possibly the ESP link.

stronnag, 2018-03-06:
I very much doubt it's the esp-link. I don't have a NAZE32 anymore, but I do have a CC3D with 1.7.3. No problem  in an Ubuntu VM (16.04 or 17.04) using a serial / TCP bridge.  On the netbook, does it work with a direct USB cable?

stronnag, 2018-03-06:
And if it doesn't work with the USB cable, or you'd be prepared to test out something:
````
$ cd <your mwp directory>
$ git pull
$ git checkout development
$ git pull # perhaps not necessary
$ make && sudo make install
$ # try the esp link again, from the terminal and  post the output.
````
then
````
$ git checkout master
````

b14ckyy, 2018-03-06:
Now Connection is established but MSP runs into timout Loop. But at least MWP does not crash anymore:

```
2018-03-06T12:39:07-0800 Using Xwayland for safety:)
2018-03-06T12:39:07-0800 mwp startup version: mwp 1716b87 / 2018-03-06 1.065.727
2018-03-06T12:39:07-0800 Using speech api 2 [speechd]
2018-03-06T12:39:07-0800 libchamplain 0.12.16
2018-03-06T12:39:08-0800 Starting Bing proxy 
2018-03-06T12:39:08-0800 Failed to save layout, remains in /tmp/.mwp.EWR2EZ.xml
2018-03-06T12:39:37-0800 Try connect tcp://192.168.4.1:23
2018-03-06T12:39:37-0800 Connected tcp://192.168.4.1:23
2018-03-06T12:39:37-0800 Serial ready
2018-03-06T12:39:37-0800 set mrtype=14 cap =3c
2018-03-06T12:39:37-0800 Feature Mask [10402082] : telemetry false, gps true
2018-03-06T12:39:37-0800 Flash claims to be 0 bytes!!
2018-03-06T12:39:37-0800 MSP Timeout (MSP_CMDS_DATAFLASH_SUMMARY)
2018-03-06T12:39:37-0800 Flash claims to be 0 bytes!!
2018-03-06T12:39:38-0800 MSP Timeout (MSP_CMDS_DATAFLASH_SUMMARY)
2018-03-06T12:39:38-0800 Flash claims to be 0 bytes!!
```

stronnag, 2018-03-06:
Thanks. NAZE32 claims to have a data flash, then lies about the size, at least in 1.7.3. Let me fix that.

b14ckyy, 2018-03-06:
With USB it is the same issue. INAV configurator works fine. 
Do you mean the Blackbox log dataflash? the 16MBit Flash on my naze is broken and not detected anymore, also in INAV configurator not accessible. That could cause this Report. if MWP ignores that, it should possibly work.

stronnag, 2018-03-06:
Would you please go through the same process:
````
$ cd <your mwp directory>
$ git pull
$ git checkout development
$ git pull # perhaps not necessary
$ make && sudo make install
$ # try the esp link again, from the terminal and  post the output.
````
It should now complete the startup (I forgot to move to the next state when the flash was 0). 
If you could post the output again, and hopefully I can merge development into master.

b14ckyy, 2018-03-06:
USB:

```
2018-03-06T12:53:48-0800 Using Xwayland for safety:)
2018-03-06T12:53:48-0800 mwp startup version: mwp b4c1c5a / 2018-03-06 1.065.748
2018-03-06T12:53:48-0800 Using speech api 2 [speechd]
2018-03-06T12:53:49-0800 libchamplain 0.12.16
2018-03-06T12:53:49-0800 Starting Bing proxy 
2018-03-06T12:53:49-0800 Failed to save layout, remains in /tmp/.mwp.6042EZ.xml
2018-03-06T12:53:59-0800 Try connect /dev/ttyUSB0
2018-03-06T12:53:59-0800 Connected /dev/ttyUSB0
2018-03-06T12:53:59-0800 Serial ready
2018-03-06T12:53:59-0800 set mrtype=14 cap =3c
2018-03-06T12:53:59-0800 Feature Mask [10402082] : telemetry false, gps true
2018-03-06T12:53:59-0800 Flash claims to be 0 bytes!!
2018-03-06T12:53:59-0800 INAV v1.7.3  NAZE (974440df)
2018-03-06T12:53:59-0800 Generate navconf 13
2018-03-06T12:53:59-0800 Masks arm 1 angle 2 horz 4 ph 200 rth 400 wp 800
2018-03-06T12:53:59-0800 WP_GETINFO: 30/0/0
2018-03-06T12:53:59-0800 Activeboxes 4 00004002
2018-03-06T12:53:59-0800 Arming flags: Calibrating,Navigation unsafe (0a00), load 5% MSP_CMDS_STATUS_EX
2018-03-06T12:53:59-0800 Sensors: Acc Baro  (0003)
2018-03-06T12:53:59-0800 switch val == 00004002 (00000006)
2018-03-06T12:53:59-0800 message => No GPS detected
2018-03-06T12:53:59-0800 alert /usr/share/mwp/beep-sound.ogg
2018-03-06T12:53:59-0800 no GPS, sensor = 0x3
2018-03-06T12:53:59-0800 Timer cycle for 4 items, 25 => 59 bytes
2018-03-06T12:53:59-0800 Start poller
2018-03-06T12:53:59-0800 Arming flags: Navigation unsafe (0800), load 6% MSP_CMDS_STATUS_EX
2018-03-06T12:54:09-0800 Arming flags: Navigation unsafe,H/W fail (8800), load 6% MSP_CMDS_STATUS_EX
2018-03-06T12:54:09-0800 sensor health 8003 1 0
2018-03-06T12:54:09-0800 init icon 14
2018-03-06T12:54:09-0800 alert /usr/share/mwp/bleet.ogg
2018-03-06T12:54:09-0800 request sensor info
2018-03-06T12:54:09-0800 Sensor status 0
2018-03-06T12:54:14-0800 Serial doom replay 0
2018-03-06T12:54:14-0800 15s, rx 23465b, tx 8400b, (1597b/s, 572b/s) to 0 wait 0, avg poll loop 51 ms messages 1399
2018-03-06T12:54:14-0800 Close fd -1 on G_IO_IN|G_IO_ERR|G_IO_HUP (19)
```

WiFi:

USB:

```
2018-03-06T12:53:48-0800 Using Xwayland for safety:)
2018-03-06T12:53:48-0800 mwp startup version: mwp b4c1c5a / 2018-03-06 1.065.748
2018-03-06T12:53:48-0800 Using speech api 2 [speechd]
2018-03-06T12:53:49-0800 libchamplain 0.12.16
2018-03-06T12:53:49-0800 Starting Bing proxy 
2018-03-06T12:53:49-0800 Failed to save layout, remains in /tmp/.mwp.6042EZ.xml
2018-03-06T12:53:59-0800 Try connect /dev/ttyUSB0
2018-03-06T12:53:59-0800 Connected /dev/ttyUSB0
2018-03-06T12:53:59-0800 Serial ready
2018-03-06T12:53:59-0800 set mrtype=14 cap =3c
2018-03-06T12:53:59-0800 Feature Mask [10402082] : telemetry false, gps true
2018-03-06T12:53:59-0800 Flash claims to be 0 bytes!!
2018-03-06T12:53:59-0800 INAV v1.7.3  NAZE (974440df)
2018-03-06T12:53:59-0800 Generate navconf 13
2018-03-06T12:53:59-0800 Masks arm 1 angle 2 horz 4 ph 200 rth 400 wp 800
2018-03-06T12:53:59-0800 WP_GETINFO: 30/0/0
2018-03-06T12:53:59-0800 Activeboxes 4 00004002
2018-03-06T12:53:59-0800 Arming flags: Calibrating,Navigation unsafe (0a00), load 5% MSP_CMDS_STATUS_EX
2018-03-06T12:53:59-0800 Sensors: Acc Baro  (0003)
2018-03-06T12:53:59-0800 switch val == 00004002 (00000006)
2018-03-06T12:53:59-0800 message => No GPS detected
2018-03-06T12:53:59-0800 alert /usr/share/mwp/beep-sound.ogg
2018-03-06T12:53:59-0800 no GPS, sensor = 0x3
2018-03-06T12:53:59-0800 Timer cycle for 4 items, 25 => 59 bytes
2018-03-06T12:53:59-0800 Start poller
[... 28 lines cut ...]
2018-03-06T13:04:02-0800 Generate navconf 13
2018-03-06T13:04:02-0800 Masks arm 1 angle 2 horz 4 ph 200 rth 400 wp 800
2018-03-06T13:04:02-0800 WP_GETINFO: 30/0/0
2018-03-06T13:04:02-0800 Activeboxes 4 00200002
2018-03-06T13:04:02-0800 Arming flags: Not level (0100), load 7% MSP_CMDS_STATUS_EX
2018-03-06T13:04:02-0800 Sensors: Acc Baro GPS  (000b)
2018-03-06T13:04:02-0800 switch val == 00200002 (00000006)
2018-03-06T13:04:02-0800 init icon 14
2018-03-06T13:04:02-0800 Timer cycle for 7 items, 43 => 107 bytes
2018-03-06T13:04:02-0800 Start poller
2018-03-06T13:04:26-0800 alert /usr/share/mwp/sat_alert.ogg
2018-03-06T13:04:27-0800 alert /usr/share/mwp/sat_alert.ogg
2018-03-06T13:04:29-0800 Arming flags: Not level,Navigation unsafe (0900), load 6% MSP_CMDS_STATUS_EX
2018-03-06T13:04:31-0800 Arming flags: Not level (0100), load 6% MSP_CMDS_STATUS_EX
2018-03-06T13:04:33-0800 MSP Timeout (MSP_CMDS_ALTITUDE)
2018-03-06T13:04:42-0800 alert /usr/share/mwp/sat_alert.ogg
2018-03-06T13:04:43-0800 Arming flags: Not level,Navigation unsafe (0900), load 6% MSP_CMDS_STATUS_EX
2018-03-06T13:05:02-0800 Arming flags: Not level (0100), load 7% MSP_CMDS_STATUS_EX
2018-03-06T13:05:03-0800 alert /usr/share/mwp/sat_alert.ogg
2018-03-06T13:05:04-0800 Arming flags: Not level,Navigation unsafe (0900), load 6% MSP_CMDS_STATUS_EX
2018-03-06T13:05:09-0800 Arming flags: Not level (0100), load 6% MSP_CMDS_STATUS_EX
2018-03-06T13:05:18-0800 alert /usr/share/mwp/sat_alert.ogg
2018-03-06T13:05:19-0800 Arming flags: Not level,Navigation unsafe (0900), load 6% MSP_CMDS_STATUS_EX
```
Connection works Now fine, both USB and Wifi. But when Battery is connected and GPS active, it permanently jumps between GPS fix false and true. Inav Config over USB is fine. When GPS off then correctly Shows sensor warning.

b14ckyy, 2018-03-07:
Update:
I tested again now and it seems to work for status info, mission upload and download and so on. the only issue is that the NAV Status is not correctly submitted. It switches between fix true and false 5 times per second and shows no info update. All other data seems to be shown correctly. Even over USB connection it has the same issue. the log looks in both cases like the wifi log above. If this is fixed, all should be fine.

stronnag, 2018-03-07:
mwp is correctly displaying how iNav handles this particular field (which is not the same as multiwii, unfortunately). In iNav, it's basically a heartbeat, while the fix status is shown elsewhere.

You are better just using the Flight View widget with iNav, and using the Nav Status space for something else, like the artificial horizon or battery status. 

Anyway, I think we have addressed the original point of the issue, so I shall close it.

closed 2018-03-07 by stronnag