stronnag/mwptools #41: Version number script does not handle GPG-signed commits

nmaggioni, 2018-05-11:
The [version number generation script](https://github.com/stronnag/mwptools/blob/7aced427e226fe5c5ea63c779083af9519329fec/mwp/mkvers.sh#L4) fails during installation if the latest commit was signed with a GPG key (e.g.: `git commit -S`).

Strings with quotes inside them are probably imported unescaped and the rest of the code gets wrapped inside them until EOF, causing unparseable output.

stronnag, 2018-05-16:
I cannot reproduce this. Signed and unsigned commits behave the same. The only git information used by mwp is the output of `git log -1 --format="%h"` which does not change with signing (at least here). 

Can you please provide an example of the output you get from from `git log -1 --format="%h"` where a commit is signed, and  the `git log -1 --format="%h"`  output contains quotes?

nmaggioni, 2018-05-18:
I tend to forget that I've enabled showing signatures in logs with my config (`log.showsignature=true`)...
With that option the output of `git log -1 --format="%h"` includes verbose GPG signature details and breaks C strings.

I've found that `git rev-parse --short HEAD` does not have this problem, being it a plumbing command.

commit 569b1a92 (2018-05-18): Recognize verbose signed commits in version script

Fixes #41
closed 2018-05-18 by stronnag