stronnag/mwptools #43: Too long timeout (UDP)

seeul8er, 2018-06-07:
Hi!
I very much enjoy the tool you have written. 
I try to integrate/make it usable with my own project called "DroneBridge" which is a communication link.
"End-user-device" (app/laptop/mwptools) to ground station communication currently is based on UDP.

If mwptools requests a message it waits for about one second for a requested message (according to Wireshark) bevor it requests a new one (timeout). As the radio link loses packets from time to time, this behaviour leads to one-second delays/interruptions. I think the user experience suffers a lot from that, for the fact that just one packet was lost.

**As this is a UDP only issue (i guess) I'd like to make following suggestions:**
- Make timeout configurable (at least for UDP connections)
- Make requests per second configurable - MSP is polled and that way users can optimize the settings for their hardware/link capacity.

_Another thing that came up:_
I love the fact that it auto-detects the switch from MSP to LTM. DroneBridge delivers both protocols via separate UDP streams. Would it be possible to have a second socket for LTM only? The moment LTM is detected on socket 2 you could pause/deactivate socket 1 for MSP.
I'll write a new issue if necessary. Just want to test the waters.

stronnag, 2018-06-08:
So at the moment, mwp runs a 500ms timer loop. This is adequate for slow radios where the poll / response time is perhaps 130 - 140 ms. The timeout is defined by the *existing* setting `poll-timeout`
```
gsettings list-recursively org.mwptools.planner | grep poll
org.mwptools.planner poll-timeout uint32 500
``` 
As the timeout is checked for > `poll-timeout `, this leads to the 1 second you see in wireshark.
```
gsettings set org.mwptools.planner poll-timeout  250
# or any value less than 500
```
Would leave a 500ms timeout / retry, due the resolution of the timer loop.

There is a (small) overhead in scheduling the timer loop unnecessarily, but reducing it to 100ms is reasonable. With `poll-timeout` set to < 100, that would give a 0.1s minimum timeout.

seeul8er, 2018-06-22:
Fair enough. I can live with that. DroneBridge for ESP32 already offers such an option and the rest of the software (DroneBridge for RPi) can be configured to do that as well.

I wrote a parser that can handle MSP & LTM at the same time. I have not tested the "on the fly switch" from one protocol to the other but it should work. [You can check it out here.](https://github.com/seeul8er/DroneBridge/blob/master/db_esp32/main/msp_ltm_serial.c) & [Here is a use case.](https://github.com/seeul8er/DroneBridge/blob/master/db_esp32/main/db_esp32_control.c#L110) Maybe you find it somehow useful. It uses code from iNAVs MSP parser so it must be licensed under GPL I guess. But that shouldn't be a problem since mwptools is GPL3 as well.

stronnag, 2018-06-22:
mwp has automatic protocol switching for about two years now, thank you.

stronnag, 2018-08-12:
closing, as the timeout default has been reduced.

closed 2018-08-12 by stronnag