stronnag/mwptools #45: Feature Request: Please make flight statistics window user configurable.

ninja-zx11, 2018-07-19:
Please make flight statistics window user configurable so that user can add their needed values like max altitude,Max gps distance from the armed position,max amps etc.Thanks.

stronnag, 2018-07-20:
As of 2018-07-20, the `development` branch has an expanded flight stats display. 
Max current will not be added until the 2.0 BB format is finalised and documented.

[image]

ninja-zx11, 2018-07-21:
I appreciate your work!! Thanks for your support.

stronnag, 2018-07-21:
I now appear to understand both the old and new bbox battery stuff

[image]

I think this is one of yours.  Development branch as of 2018-07-21 17:10 ish.

Requires blackbox patched with `mwptools/samples/consolidated-blackbox-inav.dif`

ninja-zx11, 2018-07-28:
This is awesome!!
So maximum range is max straight line distance from the launch location?Thanks for adding this feature.

stronnag, 2018-07-28:
Range is the maximum straight line distance from home.

I'm going to close this; if these's anything other parameters you think of, please reopen the issue.

closed 2018-07-28 by stronnag