stronnag/mwptools #46: mwptools shutting down by itself while trying to open a log file.

ninja-zx11, 2018-07-20:
After updating mwptools by using command "git pull && make && sudo make install" ,its shutting down by itself after i try to open a log file.I am running ubuntu 18.04 LTS on a latest oracle VM virtualbox on win10.It was working fine before the update.

As i am not familiar with the Linux,please guide me how to get the log file(if needed to diagnose).Thanks

stronnag, 2018-07-20:
The blackbox log file please. 
Actually, there will so be a $HOME/mwp_stderr_2018-07-20.txt log as well, that will be useful too.

ninja-zx11, 2018-07-20:
Here are the files.
[blackbox_log_2018-07-19_064530.TXT](https://github.com/stronnag/mwptools/files/2215126/blackbox_log_2018-07-19_064530.TXT)
[mwp_stderr_2018-07-20.txt](https://github.com/stronnag/mwptools/files/2215128/mwp_stderr_2018-07-20.txt)

stronnag, 2018-07-20:
Ok, sadly it works here, sort of. There are a number of invalid / corrupt entries in the log, which mwp should survive (it does here), however it fails to find a location -- but nothing bad happens.

Apologies for the piecemeal requests; which Linux (disto, release (i.e Ubuntu xx.yy)) please. 

[image]

ninja-zx11, 2018-07-20:
Yes that's strange.It was working fine with the same log file before update.I am using ubuntu18.04 LTS.

No worries and please let me know if any other files needed.Thanks.

stronnag, 2018-07-20:
Unfortunately, it works in my Ubuntu 18.04 VM (hosted on Linux, using kvm). Both the extant `master` and the updated `development` branch (which does a better job of working out which are the valid logs -- actually non-trivial). 

If possible, you might have a try with `development`; if nothing else (if it doesn't crash), the work in process flight stats.

**development, Ubuntu 18.04**
[image]

ninja-zx11, 2018-07-20:
Okay i will try development branch.Could you please guide me how can i install development branch.

stronnag, 2018-07-20:
Apologies, I assume far too much (in the following, `mwptools` is the mwptools source directory).
```
cd mwptools
git pull
git checkout development
git pull
make && sudo make install
```
To get  back to master, substitute `master` for `development`

ninja-zx11, 2018-07-20:
I think i need to read more about linux commands and stuff.Sorry for asking to spoon feed:(

I tried to install development but got this error.
`ninja@ninja-VirtualBox:~/Projects/mwptools$ git pull
Already up to date.
ninja@ninja-VirtualBox:~/Projects/mwptools$ git checkout development
error: pathspec 'development' did not match any file(s) known to git.`

stronnag, 2018-07-20:
At the moment master and development are the same.
```
cd mwptools
git checkout master
git pull
make && sudo make install
```
btw did you originally create the project from a zip file or `git clone` ?

stronnag, 2018-07-20:
and what does
```
git branch -r
```
show ?

ninja-zx11, 2018-07-20:
I followed the installation video to install.

> ninja@ninja-VirtualBox:~/Projects/mwptools$ git branch -r
  origin/HEAD -> origin/master
  origin/master

stronnag, 2018-07-20:
Thanks. I'll have to have a think about that --- tomorrow.

ninja-zx11, 2018-07-20:
No worries.I will also try to install it again on my home computer as this is my work PC.Thanks and have a great weekend.

stronnag, 2018-07-21:
My git-fu improves!

The default instruction only set up the master branch (in order in minimise the initial setup) .
 In order to access other branches:
```
git remote set-branches origin '*'
git fetch -v
```
Then:
```
$ git branch
  development
* master
```
And one can checkout development
```
$  git checkout development
Switched to branch 'development'
Your branch is up to date with 'origin/development'.
```

stronnag, 2018-07-21:
This is probably fixed in both master and development now.

stronnag, 2018-07-26:
A few hours ago I was almost convinced that this was a problem with the way that mwp was doing a number of asynchronous event; now my suspects are:

* Video drivers (not liking the background zoom)
* Virtualbox / Win 10
* Combination of the above
* Something else to do with async I/O

So let's see. The easiest way is to start from the bottom. So I've split out the bbox processing into a simple standalone application (well two actually, doing the same thing in slightly different ways).

So please download the attached Zip file, then (in an Ubuntu terminal)
[bbox_test.zip](https://github.com/stronnag/mwptools/files/2233145/bbox_test.zip)

```
unzip bbox_test.zip
cd bbox_test/
make
# FILE is blackbox file, GEOUSER is a geonames user name (optional)
 ./runme.sh FILE GEOUSER 
```
Then post back `mwp_bbox0.log` and `mwp_bbox1.log` 

You should see something like:
```
$ ./runme.sh blackbox_log_2018-07-19_064530.TXT a_user_name
Analysing log ...
Getting base location from index 6 49.162187 -122.494655 true
Getting data for log 6
Geonames 49.162187 -122.494655 America/Vancouver
6 05:24.249 2018-07-14 18:23:16 PDT
Getting data for log 7
7 05:28.304 2018-07-14 18:30:26 PDT
Getting data for log 8
8 05:58.073 2018-07-14 18:38:00 PDT
Getting data for log 9
9 00:07.803 2018-07-14 18:45:48 PDT
Getting data for log 10
10 05:02.661 2018-07-14 18:46:02 PDT
Getting data for log 11
11 04:15.928 2018-07-14 18:54:15 PDT
File contains 11 entries
```

ninja-zx11, 2018-07-26:
Here are the files.Thanks
[mwp_bbox0.log](https://github.com/stronnag/mwptools/files/2233280/mwp_bbox0.log)
[mwp_bbox1.log](https://github.com/stronnag/mwptools/files/2233281/mwp_bbox1.log)

stronnag, 2018-07-26:
OK, so Ubuntu on VBox handles the async IO correctly as well; next step is to turn off the background map zoom while analysing logs. For this we need the latest development branch.
```
cd mwptools
git remote set-branches origin '*'
git fetch -v
git checkout development
git pull
make && sudo make install
```
Now run it from the terminal with a specific environment (with and without the async map zoom).
```
# Async zoom, may crash
MWP_NOLOG_REDIRECT=1 mwp 2>> /tmp/bbsync.log
# No async zoom, see if this helps
MWP_NOLOG_REDIRECT=1 MWP_BB_NOZOOM=1 mwp 2>> /tmp/bbnosync.log
```
And let me have the two log files `/tmp/bbsync.log` and `/tmp/bbnosync.log` please, and let me know what happens.

ninja-zx11, 2018-07-26:
Its working now!!
Both Async zoom and No Async zoom worked from the terminal.
I think newer development build did the trick.Any idea why it wasn't working before?
I also tried to run from the desktop icon and its working now.
[bbsync.log](https://github.com/stronnag/mwptools/files/2233519/bbsync.log)
[bbnosync.log](https://github.com/stronnag/mwptools/files/2233521/bbnosync.log)

stronnag, 2018-07-26:
Thanks. No idea what the previous problem was. I now need to tidy a couple of things and document some new setting options then I'll merge it to master, tomorrow.
```
# back to master 
git checkout master
```

stronnag, 2018-07-26:
I think we're done on this, closing.

closed 2018-07-26 by stronnag