stronnag/mwptools #47: Dropdown for waypoint type in mission editor

Ant73, 2018-09-04:
When the FC is connected, the dropdown to change the waypoint type in the mission editor will close straight away. If the FC is disconnected, the dropdown stays open and the wpt type can be changed (e.g RTH). Right clicking on the wpt in the map view works whether the FC is connected or not.

FC = Omnibus F4 V3
Linux = Linux Mint 18.3

stronnag, 2018-09-04:
I believe this was corrected some time ago in dee85fc14f4ee3a328c4a3751e7ace597229e48e of 2018-03-03. This would be indicated as build v1.062.557 in the 'About' dialogue. 

Can you please confirm the build you're using, and consider updating if it's older.
```
cd mwptools
git pull && make && sudo make install
```
 If you're running a newer build of mwp, I'll need to think about what I need to ask about the 'not supported' Mint 18.

Ant73, 2018-09-05:
I only just installed it a few days ago. I think it is the latest 2018-08-28 1.238.317

stronnag, 2018-09-05:
Yes, it's a bug. It needs the previous fix extended to the drop down. In progress.

commit 1867babd (2018-09-05): * fc-get: Only check for incompatible firmware when a version is
  present in the saved file.

* mwp: Pause poller when tote WP type dropdown is selected (fixes #47)
* mwp: Resume poller when tote edit is cancelled
closed 2018-09-05 by stronnag