stronnag/mwptools #51: modify install instructions on wiki

wx4cb, 2019-03-29:
not an "issue" par se, but an observation. Didn't want to edit the wiki directly but i can if you want me to. In the Install page, there's an easier way (single command) to install the ubuntu deps.

I would also recommend adding the "-y" switch to all the "apt install" lines that way the user doesn't get prompted for "are you sure?" :D

anyways, right now you have this:

```
copy the ubuntu-deps.txt file
cp docs/ubuntu-deps.txt /tmp/ubuntu-deps.sh
make it executable
chmod +x /tmp/ubuntu-deps.sh
```

and run the script (this will take some time). Answer Y on all questions.
sudo /tmp/ubuntu-deps.sh

You can replace it with this

```
cd mwptools
sudo bash docs/ubuntu-deps.txt
```

stronnag, 2019-03-29:
You're welcome to edit the wiki (and raise a PR for the dependencies documents).

wx4cb, 2019-03-29:
done and i'll do the txt file after i've finished testing this out. seems to run fine under "ubuntu for windows" other than what i noted in the other issue case.

I also have some instructions for installing the X server that you would need on windows 10 for it to run I can add those into a wiki page if you want once i figure it all out

stronnag, 2019-03-29:
Cool. I'd been hoping for a while that someone brave person would make it work in WSL.
Are you using VcXsrv?

Anyway, a wiki page on getting it working on Win10/WSL would be wonderful.

wx4cb, 2019-03-29:
yea vcxsrv, but right now im getting a bunch of mavcrc errors (MSP and LTM at 57600) and also i can't get the maps to load, i'm guessing that's an OpenGL thing?

also as you can see some of the text isn't correct on the right..... i'm not supposing there's a way to adjust the map size? or the font size? my Laptop resolution is 1366*768

[image]

wx4cb, 2019-03-29:
i think you need to add me as a contributor?

stronnag, 2019-03-29:
> yea vcxsrv, but right now im getting a bunch of mavcrc errors (MSP and LTM at 57600) and also i can't get the maps to load, i'm guessing that's an OpenGL thing?

Most likely OpenGL. Damn it. Years ago I briefly had access to a Windows (7?) box and built mwp in cygwin, where it worked OK (even the map). But for most Windows users, cygwin is a harder option than a VM.
 
> also as you can see some of the text isn't correct on the right..... i'm not supposing there's a way to adjust the map size? or the font size? my Laptop resolution is 1366*768

You could try
```
gsetting set org.mwptools.planner font-fv 12
```
(this is what I have on my 1366*768 chromebook)

stronnag, 2019-03-29:
> i think you need to add me as a contributor?

No problem. Where would you like?

wx4cb, 2019-03-29:
not sure what you mean... i guess dev branch ? not sure how to do it. still trying to learn github (outside of cloning :D) lol

stronnag, 2019-03-29:
You can contribute PRs by:
* clone the repository
* In your fork
  * create a new branch from mwptools/development
  * make your changes
  * commit the changes to your fork
  * create a PR against mwptools/development
  * I merge your change (for which you are credited as the contributor)

i.e the same procedure as for iNav https://github.com/iNavFlight/inav/blob/master/docs/development/Development.md#using-git-and-github

wx4cb, 2019-03-29:
ok a little faffing and learning but i think i got it heh

closed 2019-03-29 by wx4cb