stronnag/mwptools #52: ubuntu on windows

wx4cb, 2019-03-29:
just a comment.... 

running "ubuntu on windows" not the vm, but what comes with windows 10. it seems to compile and install fine however it looks like pkg-config doesn't like a command line argument.

```
Unknown option --list-package-names
Must specify package names on the command line

```

```
make[1]: Leaving directory '/home/charlesb/src/mwptools/mwp'
make -C bbox-replay
make[1]: Entering directory '/home/charlesb/src/mwptools/bbox-replay'
Unknown option --list-package-names
Must specify package names on the command line
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/charlesb/src/mwptools/bbox-replay'
make -C qproxy
make[1]: Entering directory '/home/charlesb/src/mwptools/qproxy'
Unknown option --list-package-names
Must specify package names on the command line

ETC ETC.
```

stronnag, 2019-03-29:
The message is harmless (a bug, but harmless; it suppresses another harmless warning on more modern distros). I'll fix it.

wx4cb, 2019-03-29:
ok cool..

stronnag, 2019-03-29:
fixed by 0105b74

wx4cb, 2019-03-29:
ok so compiles fine on wsl, but when running on it i get this and i'm unable to resize the window.

(mwp:20489): Gtk-WARNING **: 17:30:01.546: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node button, owner GtkButton)

(mwp:20489): Gtk-WARNING **: 17:30:01.548: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node button, owner GtkButton)

wx4cb, 2019-03-29:
seems to be associated to the fact there's no "dbus" on WSL. when i try to change settings in the preferences menu:

```
(mwp:20555): dconf-WARNING **: 17:40:32.043: failed to commit changes to dconf: Failed to execute child process “dbus-launch” (No such file or directory)

(mwp:20555): dconf-WARNING **: 17:40:32.044: failed to commit changes to dconf: Failed to execute child process “dbus-launch” (No such file or directory)

(mwp:20555): dconf-WARNING **: 17:40:32.045: failed to commit changes to dconf: Failed to execute child process “dbus-launch” (No such file or directory)

(mwp:20555): dconf-WARNING **: 17:40:32.047: failed to commit changes to dconf: Failed to execute child process “dbus-launch” (No such file or directory)

```

wx4cb, 2019-03-29:
plus the fact there's no window manager to handle the resize

stronnag, 2019-03-29:
IIRC, running stuff that requires a complex desktop environment on WSL is unsupported (by WSL).
btw, another user was also attempting to get it working in WSL via the RCG forum, see https://www.rcgroups.com/forums/showpost.php?p=41157749&postcount=447 and subsequent posts.I think he also got stuck on dbus.

stronnag, 2019-03-29:
> plus the fact there's no window manager to handle the resize

Indeed, the GTKButton warnings are harmless, I get them on my 1024x600 eeepc (but everything still works there).

wx4cb, 2019-03-29:
dbus is fixed.

stronnag, 2019-03-29:
:-)

wx4cb, 2019-03-29:
and you don't need a complex window environment, just a window manager that can handle resizing. even the venerable TWM will work :D

stronnag, 2019-03-29:
Isn't it possible to run VXsrv in a mode where it uses the Windows window manager? 
"Multiple Windows" mode?

wx4cb, 2019-03-29:
yes that's correct, but the issue is, you're not handling the WM_RESIZE by the look of it in msp

stronnag, 2019-03-29:
No, because that's the WM's job. I don't have to on any of the supported OS / distros. Resize works without the application having to do anything.

wx4cb, 2019-03-29:
that's what i'm saying... the WM handles the physical resize, but the app "should" handle the signal that it's been resized so it can adjust any "controls" that it creates.... that's the way it's supposed to work anyway.

The  framework that you're using may handle all of that for you,

stronnag, 2019-03-29:
gtk handles that (and makes it positively difficult for the app to do so). This is not my first gtk application.

stronnag, 2019-03-29:
PRs to improve the situation in WSL will be gratefully accepted as long as they are non-intrusive and don't affect / break traditional platforms (Linux, FreeBSD).

wx4cb, 2019-03-29:
yea cygwin works differently from WSL. and as far as PRs etc, i'm purposely not modifying the code (mainly as i don't understand vala - or whatever it's in :D)

wx4cb, 2019-03-29:
ok fixed it... like i thought, it's an OpenGL issue. and its VxSrvs issue...

wx4cb, 2019-03-29:
[image]

wx4cb, 2019-03-29:
@stronnag seems to work fine.. i'll play tomorrow when i maiden my Mini talon. so this should auto switch over to decoding the LTM telemetry when i arm right ?

stronnag, 2019-03-30:
Nice, very nice. If you can get the font in the flight view to size then the whole app will probably also resize. The fonr **should** get a hint if you can move the panel divider right then left. Or just go down a number with:

```
gsetting set org.mwptools.planner font-fv 11
```
As long you have MSP / LTM switching set in the FC, mwp should switch automatically.

wx4cb, 2019-03-30:
for some reason windows won't let me change it

```
charlesb@HP_Encoder:~$ gsettings get org.mwptools.planner font-fv
12
charlesb@HP_Encoder:~$ gsettings set org.mwptools.planner font-fv 9
charlesb@HP_Encoder:~$ gsettings get org.mwptools.planner font-fv
12
charlesb@HP_Encoder:~$ sudo gsettings set org.mwptools.planner font-fv 9
[sudo] password for charlesb:
charlesb@HP_Encoder:~$ gsettings get org.mwptools.planner font-fv
12
charlesb@HP_Encoder:~$ sudo gsettings get org.mwptools.planner font-fv
12
```

stronnag, 2019-03-30:
Damn, that's sub-optimal.

stronnag, 2019-03-30:
Possibly due to the lack of unix-sockets in WSL (coming later, IIRC). A potential workaround in https://www.reddit.com/r/bashonubuntuonwindows/comments/51f7ni/how_to_run_gnome_terminal_on_wsl/ ? See also comments in /etc/dbus-1/session.conf and /usr/share/dbus-1/session.conf (distro / age dependence).

wx4cb, 2019-03-30:
for sure.. i'm thinking it's a font issue. ie the font you're using doesn't exist. i just changed the schema.xml file and changed the default size to 9 and it seems that worked a little better.

I dunno why i change change it in gsetting, but im guessing it's a limitation of the way it works. 

not a big deal, i have it looking nicely at the minute, and we can always have an alternate schema file with smaller fonts for this. plus it would probably be fine if i had a typical 1920*1080 resolution on this laptop

[image]

wx4cb, 2019-03-30:
well i'm having no dbus errors right now. i'll take a look at that reddit link later and see what happens

stronnag, 2019-03-30:
If gsettings is the only obstacle, I'll implement a simple file based key=value store for WSL.

wx4cb, 2019-03-30:
right now, that's what it appears. and like i say, it's probably a permissions thing. 

The only other thing i've found right now is that windows doesn't seem to store permissions for the tty devices so that you have to change the permissions for the TTY device every time. but that's another thing i'm working on a workaround for.

Is there any way to specify the tty device on the command line?

stronnag, 2019-03-30:
```
mwp --help
```
hence:
```
mwp --device=/dev/ttyUSB0@115200
```

wx4cb, 2019-03-30:
ok then that works.. I can create a simple script that will do it all for ya

stronnag, 2019-03-30:
In theory (and practice on Linux/BSD) you can put persistent command line options in
```
~/.config/mwp/cmdopts
```
e.g.

```
$ cat ~/.config/mwp/cmdopts 
# Default options for mwp
--rings 50,20
#--voice-command "spd-say -t female2 -e"
--wayland
#--debug-flags=2
--dont-maximise
#-S 8192
#--voice-command "ftest file:///home/jrh/Projects/flite/cmu_us_clb.flitevox"
```

wx4cb, 2019-03-30:
for sure, it exists in WSL too (already saved the layout). but there's a start script in my fork which works well.

wx4cb, 2019-03-30:
https://github.com/wx4cb/mwptools/tree/master/WSL

wx4cb, 2019-03-30:
so attitude worked, but there was no other data coming through. the stderr log as full of CRC errors.

```
2019-03-30T11:17:27-0400 MAVCRC Fail, got f574 != cea7 (cmd=147, len=36)
2019-03-30T11:17:27-0400 Comm error count 939
2019-03-30T11:17:27-0400 MAVCRC Fail, got 5e1c != 65cf (cmd=147, len=36)
2019-03-30T11:17:27-0400 Comm error count 940

```

wx4cb, 2019-03-30:
ok so i think setup was part of it.... tried on a different aircraft with MSP/LTM and i got further. logs attached, but it seems it's losing the MSP heartbeat or something. I can still go into the CLI window and get status etc when it's doing that, but it doesn't seem to recover

[mwplog.tar.gz](https://github.com/stronnag/mwptools/files/3025779/mwplog.tar.gz)

stronnag, 2019-03-30:
mwp has the assumption that it's talking to something that behaves like inav. 

* MSP is query - response protocol
* Telemetry is unsolicited
* The two are mutually exclusive; if a telemetry packet is received, then don't bother with MSP again until the telemetry times-out. 

I have 5 years of logs that work like that (MW, then iNav). What ever you're talking to doesn't behave like that. 

The unsolicited mavlink interspersed in the MSP conversion includes stanzas that iNav never outputs (e.g. MSGID 147), and which mwp doesn't understand so the outcome is sadly predictable. mwp *only* handles the mavlink that inav generates, not the (large) universe of mavlink syntax.

From `mwp_2019-03-30_134927.raw` the `#` bits are commentary: 

```
/tmp/mx/mwp_2019-03-30_134927.raw
offset 0.00115 len 9 o                            ## output MSP, 64 == MSP.IDENT, expect MSP back
"$X<\x00d\x00\x00\x00\x8F"
24 58 3c 00 64 00 00 00 8f 
offset 0.001855 len 119 i                   ## Instead get a bunch of concatenated Mavlink packets (ATTITUDE, 
Mav 30 :"\xFE\x1C`\x01\xC8\x1E\x00\x00\x00\x00\x82\xE2\xC7<\xC3d*\xBC\x9Aw\x9C<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x97\xF9\xFE$a\x01\xC8\x93\x06\x00\x00\x00\xFF\xFF\xFF\xFF\xFF\x7F\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00:S\xC6\xFE\x1Fb\x01\xC8\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xC4;<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00:KA"
fe 1c 60 01 c8 1e 00 00 00 00 82 e2 c7 3c c3 64 2a bc 9a 77 9c 3c 00 00 00 00 00 00 00 00 00 00 00 00 97 f9 fe 24 61 01 c8 93 06 00 00 00 ff ff ff ff ff 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 00 3a 53 c6 fe 1f 62 01 c8 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c4 3b 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 3a 4b 41 
offset 0.033444 len 43 i
Mav 30 :                 ## the end of this packet also contains part of the MSP response ...
"\xFE\x1Cc\x01\xC8\x1E\x00\x00\x00\x00\x82\xE2\xC7<\xC3d*\xBC\x9Aw\x9C<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0029$X>\x00d\x00\a"
fe 1c 63 01 c8 1e 00 00 00 00 82 e2 c7 3c c3 64 2a bc 9a 77 9c 3c 00 00 00 00 00 00 00 00 00 00 00 00 32 39 24 58 3e 00 64 00 07 
offset 0.058224 len 9 i  ## and here's the rest of the MSP response, which mwp recognises
"\x00\xE7\x03\x00<\x00\x00\x80\xDA"
00 e7 03 00 3c 00 00 80 da 
offset 0.059476 len 9 o   ## as mwp moves on to the next part of its sequence ...
"$X<\x00\x01\x00\x00\x00E"
24 58 3c 00 01 00 00 00 45 
offset 0.06787 len 83 i    ## and gets a bunch of unsolicited mavlink back
Mav 147 :"\xFE$d\x01\xC8\x93\x06\x00\x00\x00\xFF\xFF\xFF\xFF\xFF\x7F\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00:\xFA3\xFE\x1Fe\x01\xC8\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xC4;<\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00:\x95\x1C"
fe 24 64 01 c8 93 06 00 00 00 ff ff ff ff ff 7f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 00 3a fa 33 fe 1f 65 01 c8 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c4 3b 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 3a 95 1c 
offset 0.114695 len 5 i
"$X>\x00\x01"
24 58 3e 00 01 
offset 0.116224 len 7 i
"\x00\x03\x00\x00\x02\x03'"
00 03 00 00 02 03 27 
offset 0.117529 len 9 o
"$X<\x00\n\x00\x00\x00\xDD"
```

wx4cb, 2019-03-30:
that's weird because mavlink isnt enabled on this (F722 wing with inav 2.1) its msp/ltm - i dont know where that might be coming from.

only thing i can think of is that the crossfire is injecting mavlink into it? 

[image]

stronnag, 2019-03-30:
It seems like it. There are mavlink packets in the data stream that iNav does not generate (e.g. 147, battery status), iNav reports voltage in the mavlink system status packet.

wx4cb, 2019-03-30:
@stronnag  is this what you use? mine would have to be 433 mhz due to the crossfire

https://www.readytoflyquads.com/3dr-radio-telemetry-kit-915mhz-532

stronnag, 2019-03-30:
In effect yes (I'm using some older 100mW, 433Mhz units).

stronnag, 2019-03-30:
mwp had a out of date Mavlink CRC seed table, which explains the (erroneous) CRC errors for MAVLink MSGID 147. fixed in 80d1477. So now the log is spammed with an unrecognised message (147) vice CRC errors. However, the high volume, unsolicited Mavlink is corrupting the MSP replies, which is why mwp gets stuck.

wx4cb, 2019-03-30:
Is there any way to drop them for now?

stronnag, 2019-03-30:
They are dropped (mwp does not know how to process them), the problem is that they appear to swamp the link. With the CRC fixed, if I replay your raw log:
```
### In one terminal:
mwp -d udp://:3000 --no-poll
## .... hit "connect" in mwp ...
### and in second, replay the log 
samples/parseraw.rb -u 3000 -i /tmp/mx/mwp_2019-03-30_134902.raw 
```
mwp's debug log shows the following, we just get swamped with Mavlink MSGID 147, and there are no replys after MISC is processed. Below this output, there's what normally happens (3DR via a BT bridge), after the inital 'what are you' conversation, mwp settles down in MSP status polling. No telemetry, as I'm indoors with no fix.

```
2019-03-30T21:29:20+0000 Try connect udp://:3000
2019-03-30T21:29:20+0000 Connected udp://:3000
2019-03-30T21:29:20+0000 Serial ready
2019-03-30T21:29:22+0000 MAVlink telemetry
2019-03-30T21:29:22+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:22+0000 Process MSP MSP_CMDS_IDENT
2019-03-30T21:29:22+0000 set mrtype=3 cap =3c
2019-03-30T21:29:22+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:22+0000 Process MSP MSP_CMDS_API_VERSION
2019-03-30T21:29:22+0000 Using MSP v2
2019-03-30T21:29:22+0000 Process MSP MSP_CMDS_NAME
2019-03-30T21:29:22+0000 Model name: "WX4CB-Harrier"
2019-03-30T21:29:22+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_INAV_MIXER
2019-03-30T21:29:23+0000 V2 mixer 8 1
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_BOARD_INFO
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_FC_VARIANT
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_FEATURE
2019-03-30T21:29:23+0000 Feature Mask [30490c82] : telemetry true, gps true, current true
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_BLACKBOX_CONFIG
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_FC_VERSION
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_COMMON_SET_TZ
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_BUILD_INFO
2019-03-30T21:29:23+0000 INAV v2.1.0  MATEKF722SE (65b0ec14)
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_BOXNAMES
2019-03-30T21:29:23+0000 Generate navconf 13
2019-03-30T21:29:23+0000 Masks arm 1 angle 2 horz 4 ph 800 rth 2000 wp 4000
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Process MSP MSP_CMDS_MISC
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 Comm error count 1
2019-03-30T21:29:23+0000 MAVCRC Fail, got 0 != 914f (cmd=1, len=31)
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:23+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:24+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:25+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:26+0000 MAVLink : 147 / 93 (36bytes)
2019-03-30T21:29:26+0000 MAVLink : 147 / 93 (36bytes)
[... 210 lines cut ...]
2019-03-30T21:41:24+0000 Start poller
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_NAV_POSHOLD
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_ANALOG2
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_NAV_STATUS
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_RAW_GPS
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_COMP_GPS
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_GPSSTATISTICS
2019-03-30T21:41:24+0000 Process MSP MSP_CMDS_ATTITUDE
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_ALTITUDE
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_INAV_STATUS
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_NAV_STATUS
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_RAW_GPS
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_COMP_GPS
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_GPSSTATISTICS
2019-03-30T21:41:25+0000 Process MSP MSP_CMDS_ATTITUDE
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_ALTITUDE
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_INAV_STATUS
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_NAV_STATUS
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_RAW_GPS
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_COMP_GPS
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_GPSSTATISTICS
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_ATTITUDE
2019-03-30T21:41:26+0000 Process MSP MSP_CMDS_ALTITUDE
2019-03-30T21:41:27+0000 Process MSP MSP_CMDS_INAV_STATUS
```

stronnag, 2019-03-30:
There is always the possibility that there is a bug in CSFR whereby uploading the MSP requests through it provokes CSFR into spewing out unsolicited MSGID 147.

stronnag, 2019-03-30:
And FYI, as of a few minutes ago (37b9e274f), the "MSP processed" messages in the log are generated with
```
mwp --debug-flags 4
```

wx4cb, 2019-03-30:
Well it cant be inav as its 2 different  Flight control is the only common point other than the laptop is the crossfire

And what I'm going to do is run and F TDI adapter on the same uart from the flight controller direct into the laptop to see for sure where it's coming from

wx4cb, 2019-03-30:
I posted a question in the TB S crossfire Facebook group if I get no answer I will create a support ticket

wx4cb, 2019-03-31:
@stronnag how about these? https://store.mrobotics.io/product-p/mro-sikv2airgnd-mr.htm

I really want to be able to use the crossfire though if i can, don't really want to have to use 3 bands for radio links lol

I've also created a support ticket at TBS as well

wx4cb, 2019-03-31:
ok so i just got home, tried it right out of the box with an F35 (the previous other FCs were f405 and f722 wings) with the same crossfire setup - in fact the same model on the crossfire module as the harrier and it works fine under ubuntu straight off. i'm gong to try the same setup under WSL

wx4cb, 2019-03-31:
ok so this is weird.... with the EXACT same setup... i get the same thing as I did with the other 2 flight controllers.
[mwp-f35.tar.gz](https://github.com/stronnag/mwptools/files/3026123/mwp-f35.tar.gz)

wx4cb, 2019-03-31:
definitely weird and definitely coming from somewhere... if you run it over the USB connection it's perfectly fine. Run it via crossfire on WSL - mavlink errors  even with your CRC changes I get what you get - message 147.

what I don't understand is why it works fine under "real" ubuntu via crossfire. there is the audio device issue which there's nothing much can be done about that without some pulseaudio hack - beyond the scope of this.

any thoughts?

stronnag, 2019-03-31:
> any thoughts?

Not really. Given the thoroughness of Mavlink's CRCs and that fact that the MSGID 147 packet is 'perfectly formed' (i.e. valid data, including the fields that are set to -1 or MAXINT16 for 'no data'); then it has to be generated by something that understands MAV MSGID 147. That precludes iNav or mwp. I somehow doubt that Microsoft added a hidden Mavlink generator as a WSL 'easter egg'.

wx4cb, 2019-03-31:
Thats the weird thing. Doesnt happen on ununtu direct, but under wsl lol.

wx4cb, 2019-03-31:
@stronnag by jove i think i fixed it.....

I have no reason or rhyme as to why it was happening... and there's still a couple of MSGID 147 kick in which i beleive are the transmitter voltage, but I completely reflash the transmitter module and the receiver on my F35 test board and it worked properly.

so right now, without audio support from WSL/MSFT the only thing that appears to be misfunctional is audio.

wx4cb, 2019-03-31:
https://youtu.be/rtARALW7blI

working fine over crossfire bluetooth. no GPS on this FC so that data wouldn't work. Will try it flying over the next couple of days

stronnag, 2019-03-31:
Excellent. And settings / preferences. A workaround (rather than patching the gschema.xml) might be to use the 'vendor overrides' to fixup some of the more important settings, see  http://www.techytalk.info/customize-default-desktop-environment-settings-gnome-centric-linux-distributions/
e.g. `org.mwptools.planner.gschema.override`

```
[org.mwptools.planner]
fv-font=10
default-map='Bing Proxy'
```
etc.

wx4cb, 2019-03-31:
interesting.. how difficult would it be to put a fv-font font size in the preferences dialog?

wx4cb, 2019-03-31:
once i've got this working reliably i'll PR my fork in for you, aksi.... i might try learning vala (or w/e language it's written in to see if i can extend it and put a servo tracking module in. I have the maestro module driver already written and working in python (as that's what I know :D) but as we already have the gps of where we are and where we took off (i'm assuming you're handling arming - hence setting the home point) then getting the bearing is simple - again another function already written 🗡

wx4cb, 2019-03-31:
is there any way to slow down the data rate?

[40 more replies]

closed 2019-04-19 by stronnag