stronnag/mwptools #55: MSP Timeouts on arming

wx4cb, 2019-05-05:
for some reason it wasn't logging even though it was turned on, but I have it setup MSP / LTM, it works fine, I arm, it sees the switch to LTM and then just times out with MSP comm failure, it restarts the telemetry poller and then just sits there trying to get MSP_INAV_STATUS - which obviously it will never get because the aircraft is flying and sending LTM.

in the mean time for the rest of the flight until disarm, when it gets MSP again, it just keeps running the last alt/sats/voltage callouts it got.

Is there any way to revert back to the last state it was in for the telemetry poller? ie, if it's disarmed or in msp and times out it goes back to that. but if it already knows it's armed and is getting (eg LTM) telemetry, to go back and look for LTM?

This wasn't under WSL, this was under ubuntu 18.10

stronnag, 2019-05-05:
Without any kind of log (`mwp_stderr_yyyy-mm-dd.txt`) , I can't do much other than say unhelpfully, "it always works for me ...". 

It really doesn't matter if the fallback  MSP starts again; once mwp actually receives a valid MSP **or** LTM packet it will process it and evince the state.

* Crossfire ??

You can always invoke `mwp --raw-log` which will log every serial transaction.

stronnag, 2019-05-05:
If it is the case that crossfire just barfs on sending LTM, you could just turn if off in the FC and endure the slower polled MSP whether armed or not.

wx4cb, 2019-05-05:
i realise you need a log, i just wanted to get input. I don't know why it wasn't logging. i have one from yesterday when i was testing on the bench. Just wondering what it's process is.

and yes, i was testing more on crossfire. i dropped down to 2.43 to test some things on this and also the lua script.

stronnag, 2019-05-05:
so it seems like CRSF is not passing LTM; nothing mwp can do here. In the event that LTM arrives, it will be processed (as that's how the normal MSP => LTM handover happens).

wx4cb, 2019-05-05:
@stronnag - not sure. i do remember seeing that it saw the LTM arm message in the stderr. and it "recorded" the home position... battery capacity started going up (or was it current, not sure. then it just got comm error and reset the poller and ended up sitting there waiting for msp packets.

i figured out why it wasn't logging i think. i created a .desktop file so i could just double click. but when i start that "in a terminal" it doesn't actually log anything. i'll try later this week manually see what happens and also switch over to mavlink.

stronnag, 2019-05-05:
In a terminal it should log. Until recently just to the terminal, now (since 2019-04-16) it logs to the terminal and  `$(pwd)/mwp_stderr-yyyy-mm-dd.txt` when run from a terminal. Note : in the current directory  when you invoked mwp.

wx4cb, 2019-05-05:
yea i know. but i created a desktop icon thingy and i screwed it up, so it was starting in a "terminal" but wasnt logging anything as i think it took over stderr. it's my fault not mwps obviously. but it's chucking it down now and im out of town for the next couple days. ive switched it over to msp/mavlink to see if that makes any difference.

stronnag, 2019-05-05:
`find ~ -iname mwp_stderr-2019-05-05.txt` might  yet surprise you

wx4cb, 2019-05-05:
already tried that. oh well...

closed 2019-06-18 by stale[bot]