stronnag/mwptools #58: Data Review if possible:

wx4cb, 2019-07-06:
just upgraded to crossfire 3.0 - beta and i just grabbed some data into a log so i was wondering if you could just look through it, or show me how to do it, to see if the modules are still spaffing mavlink packets.

inav 2.1 is set to msp/ltm @ 57600. 

it seems to play ok and in my short 100ft test i had a couple of timeouts but seemed to do ok. 

```
2019-07-06T11:22:43-0400 Process MSP MSP_CMDS_NAV_STATUS
2019-07-06T11:22:43-0400 LTM telemetry
2019-07-06T11:22:43-0400 Armed 0
2019-07-06T11:22:43-0400 Initialised "espeak" for speech
2019-07-06T11:22:43-0400 New LTM Mode Manual (0) 1 0s 0.000000 0.000000 0 0
2019-07-06T11:22:44-0400 Set home 29.164957 -81.048481 (LTM OFrame true)
2019-07-06T11:23:08-0400 set max amps 35 1691 (32)
2019-07-06T11:23:18-0400 Comm error count 10
2019-07-06T11:23:18-0400 CRC Fail, got 0 != 28 (cmd=65607)
2019-07-06T11:23:22-0400 Comm error count 11
2019-07-06T11:23:22-0400 expected header0 (f7)
2019-07-06T11:23:27-0400 Comm error count 12
2019-07-06T11:23:27-0400 CRC Fail, got 255 != 216 (cmd=65614)
2019-07-06T11:23:34-0400 Comm error count 13
2019-07-06T11:23:34-0400 CRC Fail, got 1 != 222 (cmd=65607)
2019-07-06T11:23:34-0400 set max amps 1691 1762 (36)
2019-07-06T11:23:35-0400 Comm error count 14
2019-07-06T11:23:35-0400 CRC Fail, got 65 != 85 (cmd=65619)
2019-07-06T11:23:35-0400 Comm error count 15
2019-07-06T11:23:35-0400 CRC Fail, got 36 != 67 (cmd=65607)
2019-07-06T11:23:38-0400 Comm error count 16
2019-07-06T11:23:38-0400 CRC Fail, got 1 != 207 (cmd=65607)
2019-07-06T11:23:41-0400 Alert: sat_alert.ogg (8)
2019-07-06T11:23:41-0400 GPS stalled
2019-07-06T11:23:41-0400 Restart poller on telemetry timeout
2019-07-06T11:23:41-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:42-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:43-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:44-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:45-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:46-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:47-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:48-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:49-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:50-0400 MSP Timeout (MSP_CMDS_INAV_STATUS)
2019-07-06T11:23:51-0400 Comm error count 17
2019-07-06T11:23:51-0400 expected header0 (cf)
2019-07-06T11:23:51-0400 LTM telemetry
2019-07-06T11:23:52-0400 Comm error count 18
2019-07-06T11:23:52-0400 CRC Fail, got 2 != 154 (cmd=65601)
2019-07-06T11:24:00-0400 Comm error count 19
2019-07-06T11:24:00-0400 CRC Fail, got 253 != 151 (cmd=65601)
2019-07-06T11:24:07-0400 Comm error count 20
2019-07-06T11:24:07-0400 CRC Fail, got 249 != 169 (cmd=65601)
2019-07-06T11:24:15-0400 Comm error count 21
2019-07-06T11:24:15-0400 CRC Fail, got 0 != 182 (cmd=65615)
2019-07-06T11:24:16-0400 set max amps 1762 1787 (42)
2019-07-06T11:24:18-0400 Comm error count 22
2019-07-06T11:24:18-0400 expected header0 (37)
2019-07-06T11:24:26-0400 Comm error count 23
2019-07-06T11:24:26-0400 CRC Fail, got 65 != 113 (cmd=65614)
2019-07-06T11:24:26-0400 Comm error count 24
2019-07-06T11:24:26-0400 MAVCRC Fail, got fffb != 32c (cmd=36, len=255)
2019-07-06T11:24:44-0400 Comm error count 25
2019-07-06T11:24:44-0400 CRC Fail, got 0 != 46 (cmd=65619)

```

[mwp_2019-07-06_112119.raw.txt](https://github.com/stronnag/mwptools/files/3364819/mwp_2019-07-06_112119.raw.txt)

[mwp_stderr_2019-07-06.txt](https://github.com/stronnag/mwptools/files/3364818/mwp_stderr_2019-07-06.txt)

stronnag, 2019-07-06:
Hi,

Looks a lot better than before. Some CRC errors and occasional big latency, but in general much better.

The easiest way to look at it is the `samples/parseraw.rb` tool. For example, you can play the raw log back into mwp:

In one terminal:
```
# Don't poll MSP, listen on UDP port 3000, auto-start comms
mwp --no-poll -d udp://:3000 -a
``` 
In another terminal:
```
# replay the raw log:
# only inbound messages, to UDP port 3000, save output to a file.
samples/parseraw.rb -i -u 3000 /tmp/mwp_2019-07-06_112119.raw.txt  2>&1 | tee /tmp/wx1.txt
```
The raw log replays into mwp and displays / stores the (albeit crude) raw parser output. 

You can look at the output at your leisure, and maybe checkout latencies:
```
grep Sleep /tmp/wx1.txt | sort -r | less
```

As an aside, the raw log just writes out 'records' of what mwp has received, each record begin (assume it's a non-ancient log and starts "v2\n":
```
struct {
 double time_offset; // seconds
 ushort data_size; // bytes
 uchar direction; // 'i' or 'o' (in or out)
 uchar raw_bytes[data_size]; data fragment read
}
```
There is no assumption (and it is not the case) that a record will correspond to complete MSP/LTM/MAV message, and the parsed log shows considerable message fragmentation (which is not unexpected).

When summer's over I should write a real raw log parser based on the actual mwp parser (maybe).

stronnag, 2019-07-06:
development now formats the sleep better (and it's lower case):
```
$ grep sleep /tmp/wx1.txt | sort  -n 
 ...
  1.011938 sleep
  1.012566 sleep
  1.013915 sleep
  1.030885 sleep
  2.000151 sleep
  2.119605 sleep
  9.010797 sleep
 11.780499 sleep

wx4cb, 2019-07-07:
yea i did that and you're right. there's still a little loss, b ut that's not unexpected. my concern was like before... the mavlink saturation. i'm wondering if I drop the baud rate down to 19200 or 38400 that might be a little better for the LTM.

Appreciate the help. we'll see how it goes when i get to actually go out and fly it.

wx4cb, 2019-07-08:
ok, I have more logs for you when i get home.

the RFMD bug is still there, but the force 150hz works great, until mine failsafed at a mile (TGF inav rth lol)

wx4cb, 2019-07-08:
[07.08.2019 - crossfire rfmd taranis and other logs.zip](https://github.com/stronnag/mwptools/files/3369697/07.08.2019.-.crossfire.rfmd.taranis.and.other.logs.zip)

here are the logs from today's session on crossfire 3 if they're any help. the forced 150hz mode works fine with no interruptions however, even at 100mW i only got about a mile or so before it fail-safed. but then again, the bluetooth link worked flawlessly

so all in all, i would say that it's getting there.

stronnag, 2019-07-08:
Absolutely no use to me ... maybe for the Crossfire bods?

wx4cb, 2019-07-08:
@stronnag was just a thought. but yea ... worked great for the most part. now i can pretty much get to use MWP LOL

closed 2019-07-08 by wx4cb

stronnag, 2019-07-08:
Delighted. Pleased the CSRF stuff is somewhat working in transparent serial mode. They probably owe you a few beers for your persistence in getting them to fix it.