stronnag/mwptools #61: Feature Request: Add arming state and waypoint number to dbus

wx4cb, 2019-07-11:
Is it possible to add a method "GetArmState" and "GetWPNumber" to the dbus introspect?

stronnag, 2019-07-11:
The armed state is alreadt reported via the GetState API. https://github.com/stronnag/mwptools/wiki/mwp-Dbus-API#getstate. It's either "DISARMED' or a "flight" mode. If it's not 'DISARMED', then you're armed.

For `GetWPNumber()`, the WP number that the machine to proceeding to, with `-1` if not valid. Is that what you need?

stronnag, 2019-07-11:
Development branch 2f2d676

Documentation
https://github.com/stronnag/mwptools/wiki/mwp-Dbus-API#getwaypointnumber
https://github.com/stronnag/mwptools/wiki/mwp-Dbus-API#waypointchanged

wx4cb, 2019-07-11:
the reason why i asked about the arming state is that even  "disarmed"  it shows as manual - in my case with the wing. Let me go back and take a look at what mwp is showing from these logs (and from others) and ill get back to you.

re WP: yes that would be great just the number that it's heading to.

stronnag, 2019-07-11:
Status seems OK here ... mwp log replay of FW launch ....
```
$ ./mwp-dbus-loc
Intvl 2
State Names: DISARMED MANUAL ACRO HORIZON ANGLE CRUISE RTH LAND WP HEADFREE POSHOLD ALTHOLD LAUNCH AUTOTUNE UNDEFINED
Iniital state: DISARMED
Initial WP: -1
Initial satellites: 21 3
Initial home location: 50.910471 -1.535063 0m
Initial geographic location: 50.910469 -1.535064 0m
Initial polar coordinates: 0m 191° 0°
Initial velocity: 0m/s 90°
State changed: ACRO
State changed: LAUNCH
Home changed: 50.910508 -1.535153 0m
Location changed: 50.910508 -1.535154 0m
Polar coordinates changed: 0m 180° 0°
Velocity changed: 0m/s 308°
Location changed: 50.910508 -1.535154 0m
Velocity changed: 0m/s 297°
Location changed: 50.910509 -1.535155 0m
Velocity changed: 0m/s 304°
Home changed: 50.910507 -1.535153 0m
Location changed: 50.910509 -1.535156 0m
Polar coordinates changed: 0m 320° 0°
Velocity changed: 0m/s 304°
Location changed: 50.910511 -1.535159 0m
Polar coordinates changed: 1m 315° 0°
Location changed: 50.910512 -1.535160 0m
Polar coordinates changed: 1m 316° 0°
Satellites changed: 15s 3
Location changed: 50.910513 -1.535162 0m
Polar coordinates changed: 1m 316° 0°
Velocity changed: 0m/s 287°
Location changed: 50.910511 -1.535167 0m
Polar coordinates changed: 1m 294° 0°
Velocity changed: 0m/s 282°
Location changed: 50.910465 -1.535119 1m
Polar coordinates changed: 4m 157° 14°
Velocity changed: 8m/s 145°
Location changed: 50.910437 -1.535088 1m
Polar coordinates changed: 7m 151° 8°
Velocity changed: 9m/s 142°
Satellites changed: 16s 3
Location changed: 50.910409 -1.535049 2m
Polar coordinates changed: 11m 148° 10°
Velocity changed: 10m/s 139°
State changed: ACRO
```

stronnag, 2019-07-11:
A log will be helpful

wx4cb, 2019-07-11:
so it appears that, it may be a log playback. as if I play a log and then stop it half way through (in this case while it's in POSHOLD), if i then play a log where the aircraft is disarmed, it's still sending POSHOLD out of dbus. - I know this isnt technically a "real world" issue, Of course, when the a/c is armed it changes correctly.

I think a fix would be to just make sure that everything is reset out of DBUS. as it appears it doesn't send a state change signal on log load - MWP shows "unassigned" in the nav status window

Stop playing first Log:

```
Satellites Changed: 8 Sats - Fix Type: 3
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 2.24 Course: 227.00
Sending Command: !UAV: 29.117578 -81.0608626 R:0 V:2.237 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 2.24 Course: 236.00
Sending Command: !UAV: 29.117578 -81.0608626 R:0 V:2.237 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 2.24 Course: 236.00
Sending Command: !UAV: 29.1175747 -81.0608669 R:0 V:2.237 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 0.00 Course: 218.00
```
Start new log:

```
Sending Command: !UAV: 29.117561 -81.0608986 R:0 V:0.0 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 0.00 Course: 279.00
Sending Command: !UAV: 29.1175603 -81.0608991 R:0 V:0.0 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 0.00 Course: 294.00
Sending Command: !UAV: 29.1175603 -81.0608991 R:0 V:0.0 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 0.00 Course: 294.00
Sending Command: !UAV: 29.1175601 -81.0609001 R:0 V:0.0 A:0 B:0 Z:0
UAV (DGPS POSHOLD): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 0.00 Course: 257.00
Sending Command: !UAV: 29.1175601 -81.0609001 R:0 V:0.0 A:0 B:0 Z:0

UAV Armed in new log:
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 0 ft - Elevation: 0 deg - Bearing: 0 - Velocity: 0.00 Course: 257.00
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation: 0 deg - Bearing: 29 - Velocity: 0.00 Course: 205.00
Sending Command: !UAV: 29.1175568 -81.0609039 R:3 V:0.0 A:0 B:29 Z:0
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation: 0 deg - Bearing: 29 - Velocity: 0.00 Course: 245.00
Sending Command: !UAV: 29.1175568 -81.0609039 R:3 V:0.0 A:0 B:29 Z:0
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation: 0 deg - Bearing: 29 - Velocity: 0.00 Course: 245.00
Sending Command: !UAV: 29.1175558 -81.0609044 R:3 V:0.0 A:0 B:29 Z:0
Polar Changed: Range: 1 Bearing: 212 Azimuth: 0
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation: 0 deg - Bearing: 32 - Velocity: 0.00 Course: 245.00
Sending Command: !UAV: 29.1175558 -81.0609044 R:3 V:0.0 A:0 B:32 Z:0
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation: 0 deg - Bearing: 32 - Velocity: 0.00 Course: 203.00
Sending Command: !UAV: 29.1175558 -81.0609044 R:3 V:0.0 A:0 B:32 Z:0
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation: 0 deg - Bearing: 32 - Velocity: 0.00 Course: 203.00
Sending Command: !UAV: 29.1175553 -81.0609046 R:3 V:0.0 A:0 B:32 Z:0
Polar Changed: Range: 1 Bearing: 211 Azimuth: 0
UAV (DGPS MANUAL): +29.1176 -81.0609 Alt: 0 sats: 8 - Distance: 3 ft - Elevation

```

wx4cb, 2019-07-11:
[image]
[image]

stronnag, 2019-07-11:
The issue is in the bit-rotten nav status view, which has never been updated from MW/MSP.

stronnag, 2019-07-12:
And there's no obvious solution that works for the normal case as well as replay, so I'll ignoring it for now. The iNav emulation of this MW API is less than perfect in any case; maybe I'll rename the option to "MW Nav Status". However, the state returned by the DBUS API seems OK (for the stop / restart replay case).

wx4cb, 2019-07-12:
The one thing that came to mind last night was why not  reset on log file load to be whatever the starting state is of the log file or even just a regular disarmed State

wx4cb, 2019-07-12:
I'll take a look when I get home if I can figure it out

stronnag, 2019-07-12:
> The one thing that came to mind last night was why not reset on log file load to be whatever the starting state is of the log file or even just a regular disarmed State

That's what it does for my logs....

The following is the output of samples/mwp-loc-dbus

```
### ... In replay
Velocity changed: 5m/s 55°
Location changed: 50.910205 -1.534874 7m
Polar coordinates changed: 34m 156° 11°
Velocity changed: 5m/s 48°
### Replay stopped in mid-flight. Stopping the replay sets 'disarmed'
State changed: DISARMED
### Start replay again, picks up start state.
State changed: HORIZON
....
```
Corresponding `mwp_stderr_yyyy-mm-dd.log`
```
2019-07-12T13:42:45+0100 Replay "mwp" log /t/inav/2019-05-05_tri/mwp_2019-05-05_160248.log model 0
 .... # stuff happens
### Stop replay (need to spell scenario correctly :)
019-07-12T13:45:02+0100 sceanrio cancelled
2019-07-12T13:45:02+0100 end of scenario
2019-07-12T13:45:02+0100 Distance = 517.1, max speed = 10.0 time = 88
2019-07-12T13:45:02+0100 Disarmed msp
2019-07-12T13:45:03+0100 ============== Replay complete ====================
#### Start it again
2019-07-12T13:45:14+0100 Replay "mwp" log /t/inav/2019-05-05_tri/mwp_2019-05-05_160248.log model 0
...
2019-07-12T13:45:14+0100 Armed 0 # Arming event
2019-07-12T13:45:14+0100 New LTM Mode Horizon (3) 1 0s 0.000000 0.000000 0 0 # Inital state (mode).
```
If you have logs where this doesn't happen I'd be pleased to see them

closed 2019-08-25 by stale[bot]