stronnag/mwptools #62: bearing 180 degrees out?

wx4cb, 2019-07-11:
not sure if it's me or if python is doing something weird, but in the PolarChanged signal, it seems the bearing is 180 degrees out

[image]

stronnag, 2019-07-11:
I recall we already discussed this https://github.com/stronnag/mwptools/issues/56#issuecomment-494688791
A legacy artefact with a (g)setting to reverse it in the UI (see link).
The DBus message is from home to the aircraft regardless https://github.com/stronnag/mwptools/wiki/mwp-Dbus-API#getpolarcoordinates

wx4cb, 2019-07-11:
i couldn't remember whether we did or not. it's been a crazy week

closed 2019-07-11 by wx4cb