stronnag/mwptools #63: can't compile - glib.h

wx4cb, 2019-08-15:
weird issue, fresh SD card in my rPI3 using ubuntu mate as the os.

it complains that glib.h can't be found yet iknow obviously gtk and glib are installed. I tried just a blank test program and same thing.

Any thoughts?

```
charlesb@GroundStation:~$ cat > tst.c
#include <glib.h>

void main(void) {
}
charlesb@GroundStation:~$ cc tst.c
tst.c:1:10: fatal error: glib.h: No such file or directory
 #include <glib.h>
          ^~~~~~~~
compilation terminated.
charlesb@GroundStation:~$ locate glib.h
/snap/core/7275/usr/include/reglib/reglib.h
/snap/core/7399/usr/include/reglib/reglib.h
/usr/include/dbus-1.0/dbus/dbus-glib.h
/usr/include/glib-2.0/glib.h
/usr/include/harfbuzz/hb-glib.h
/usr/include/json-glib-1.0/json-glib/json-glib.h
/usr/include/reglib/reglib.h

```

stronnag, 2019-08-16:
> yet if i compile it like this, it works
> 
> `gcc -o demo $(pkg-config --cflags glib-2.0 dbus-glib-1) demo.c $(pkg-config --libs glib-2.0 dbus-glib-1)`

It's because `glib.h` is in a sub-directory of `/usr/include` and not on the default include search path.
`pkg-config` supplies the correct include search path, and should always be used to build glib and gtk programs.

```
$ pkg-config --cflags glib-2.0
-I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include 

```

wx4cb, 2019-08-16:
so how do i put this in the make file for mwptools ?

stronnag, 2019-08-16:
Try `make clean all` first

stronnag, 2019-09-04:
Closing this on the basis that it's resolved.

closed 2019-09-04 by stronnag