stronnag/mwptools #65: Smartport telemetry packet size

ramoonas, 2019-09-20:
Hi,

Is anyone using smartport telemetry and is it still working for you? I have set up serial bridge over wifi using arduino to my R9 module, but nwp was not showing anything on the screen. Weirdly so, i have changed sport telemetry packet size to 11 (P_SIZE = 11 in serial-device.vala) and it started working for me.

I was wondering if this is due to serial bridge over wifi or something got changed in smart port protocol. If so, maybe i should commit change?

stronnag, 2019-09-20:
I don't have any Taranis hardware, so I don't actually use this. The function was developed from a comprehensive set of logs kindly provided by @wilco1967. These logs replay perfectly with current mwp.

It is possible the the your wifi bridge has extended the packet and the easiest way to confirm this would be if you could provide some logs. The wiki [smartport article](https://github.com/stronnag/mwptools/wiki/Using-the-smartport-telemetry-with-MWP#reporting-bugs) describes the logs and how to obtain them. Just a mwp raw log (`mwp -r`) would probably be enough, with the plane having sufficient satellites and armed (just on the ground, doesn't need to be flying).

ramoonas, 2019-09-20:
I already made raw log. That's what i was testing with. It is not armed, but it does transmit battery voltage at 12.4volts.  So this file is not understood by mwp unless i change P_SIZE to 11 and then i can see voltage, acc data etc.

[mwp_2019-09-18_210239.raw.tar.gz](https://github.com/stronnag/mwptools/files/3636561/mwp_2019-09-18_210239.raw.tar.gz)

stronnag, 2019-09-20:
Whatever you are using appears not to be compliant with the s-port specification, in that it is appending a 0 byte the majority of the packets.  FYI, an s-port packet looks like:
```
START_FRAME = 0, SENSOR_ID = 1, DATA_FRAME = 2, APP_ID_BYTE_1 = 3,
APP_ID_BYTE_2 = 4, DATA_BYTE_1 = 5, DATA_BYTE_2 = 6, DATA_BYTE_3 = 7,
DATA_BYTE_4 = 8, CRC = 9
```
i.e. 10 bytes, start_frame being `0x7e`.

I have approximately 24MB of logs from @wilco1967, accompanied by blackbox logs from real flights, many of which are > 30 minutes, 30+km. This s-port data compiles with the specification. Here's an example of the packet size distribution from one of those logs (32km flight, 1.7Mb of s-port data, matching bbox log:
```
Size   Instances
----------------
 0:        1
 1:        0
 2:   183985
 3:        2
 4:        2
 5:        0
 6:        0
 7:        0
 8:        0
 9:        0
10:   132263
11:        1
```

For comparison, you're provided 16kb of static, non-flight data, distributed as follows:
```
Size   Instances
----------------
 0:        1
 1:        1
 2:        5
 3:      413
 4:        5
 5:       55
 6:        4
 7:        9
 8:        2
 9:        8
10:       16
11:      537
```
i.e. the vast majority of the data does not comply with the specification.

If we look at the actual data (packet hex dump):

From wilco1967's logs:
```
7e 98 10 04 f1 00 00 00 00 f9 
7e 98 10 01 f1 00 00 00 00 fc 
7e 1b 10 30 08 00 00 00 00 b7 
7e 98 10 04 f1 58 00 00 00 a1 
```
As specification (10 bytes packets)

From your log (11 byte packets):
```
7e 98 10 04 f1 00 00 00 00 f9 00 
7e 00 10 20 04 00 00 00 00 cb 00 
7e 00 10 20 08 5c f9 ff ff 71 00 
7e 00 10 20 07 64 00 00 00 64 00 
```
There is an unwanted trailing 0 byte in each record.

Please  ask your vensor why they are appending a NUL byte to each packet. 

So I'm not going the change `P_SIZE 11` and break valid, specification compliant implementations. I'll think about making the test `  if (nb >= FrProto.P_SIZE)`, but it would be preferable if your hardware can be upgraded.

stronnag, 2019-09-20:
And just to add to the analysis of how corrupt your data example is, I updated the packet size distribution analysis to cover 1 - 255 byte packets.

wilco1967's 1.7MB log:
```
Size	Instances
~~~~	~~~~~~~~~
 0:	       1
 2:	  102016
 4:	       1
10:	  118247
12:	       1
18:	       1
```
Which is pretty impressive as it's real, over the rate at extreme range.

Compare that to your bench test, 16kb example.
```
Size	Instances
~~~~	~~~~~~~~~
 2:	       4
 3:	     161
 4:	       7
 5:	      15
 6:	       5
 7:	       7
 8:	       9
 9:	       7
10:	      13
11:	     200
12:	       8
13:	      53
14:	     246
15:	      14
16:	      23
17:	       5
18:	       2
19:	       2
20:	       6
21:	      37
22:	     334
23:	      17
24:	      61
25:	       6
26:	       3
27:	      18
28:	       3
29:	       3
30:	       2
31:	       5
32:	      21
33:	       1
34:	       7
35:	      69
37:	       8
38:	       1
39:	       2
40:	       5
42:	       3
43:	      48
44:	       1
45:	      20
46:	       2
47:	       6
48:	       7
50:	       3
52:	       1
53:	       8
56:	      20
57:	       1
58:	       6
59:	       1
63:	       1
64:	       9
66:	       3
68:	       1
69:	       1
73:	       1
74:	       1
75:	       1
77:	       1
79:	       4
81:	       2
85:	       1
87:	       2
89:	       1
121:	       1
147:	       1
```
Unless you can come up with some better evidence, I'll close this.

ramoonas, 2019-09-21:
Hmm... Very weird indeed. My knowledge is limited in this field, but i will try to find what is causing this. Definitely no need for any code changes. Thanks

closed 2019-09-21 by ramoonas