stronnag/mwptools #66: Unable to compile and install in Linux Mint 18.3 (32-bit)

MRC3742, 2019-10-05:
I am unable to compile and install using the latest current github pull.
Building in Linux Mint 18.3 (32 bit version).

Responds with the following error message.
I have also tried to use "make clean all" command first but it also returns the error.

[image]

I have been able to successfully build mwp until just recently.
The most recent build was on 09/12/2019.

[image]

stronnag, 2019-10-05:
Mint is not a supported OS. The version you're using (18.2) It is too old to support the latest compiler functions required by newer versions of the run-time used by the supported operating systems.

You might try upgrading to a later version of Mint (assuming it mirrors one of the supported Ubuntu releases), or stick with commit 8974a5c. 

I don't have the time / resources / energy to support unsupported / obsolete OS, but I won't refuse patches as long as they don't break modern / supported OS.

MRC3742, 2019-10-05:
OK

I'll update to the newest Mint version and post back to let others know if this is a working option.

Thanks for all your time and effort to keep this alive - Rich

stronnag, 2019-10-05:
If you end up staying with https://github.com/stronnag/mwptools/commit/8974a5c3a34bd649fa4b0835c3e9a423ef545d4c, you're not missing much; iNav doesn't look like it's in a hurry to add more navigation functions, and mwp is pretty much in maintenance mode at the moment.

MRC3742, 2019-10-06:
Updating to Linux Mint 19 (Tara) based on Ubuntu 18.04 (Bionic) fixes the problem and now will compile with the latest git pull (dated Sep 22, 2019)

Also, ever since you added the Terrain Analysis function I was never able to use it.
It would error out and not build the graph even though I had installed all the required dependencies.
I just thought this was a limit of my 32bit system. 
After the OS upgrade, clean and rebuild of mwp, this now works as expected. :+1: 

Thanks for pointing me in the right direction - Rich

Now Closing this issue as Resolved - Requires current OS to compile!

closed 2019-10-06 by MRC3742