stronnag/mwptools #67: running on raspberry or similar?

whoim2, 2019-12-04:
Hello, anybody have instructions for run mwp at pi?

stronnag, 2019-12-04:
See my answer to your similar question in RC Groups; https://www.rcgroups.com/forums/showpost.php?p=43304471&postcount=477

mwp runs fine on ARM devices (I run it on a 2012 Chromebook), however it has a dependency on reasonable graphics which may not be available on very low end ARM devices.

whoim2, 2020-08-05:
> See my answer to your similar question in RC Groups; https://www.rcgroups.com/forums/showpost.php?p=43304471&postcount=477
> 
> mwp runs fine on ARM devices (I run it on a 2012 Chromebook), however it has a dependency on reasonable graphics which may not be available on very low end ARM devices.

Hello!
>If you have a newer RPI3 / RPI4, I'd be interested to know if mwp is usable on these platforms.

Now im build GS based on RPi 3B+, may buy PI4.
Need help for install mwp to buster.

wx4cb, 2020-09-03:
@stronnag  @whoim2 it works perfectly fine on a pi3b, 3b+ and a pi4 which is what i'm currently running it on. as well as an odyssey x86 board. in fact i'm writing this on the pi 4 right now with it running int he background

stronnag, 2020-09-03:
Excellent, the rpi GPUs have improved somewhat in the last 10 years, as one might expect.

closed 2020-09-03 by stronnag