stronnag/mwptools #68: mwp crashing during startup on ubuntu18.4 (virtual_box/win10)

helicocrasher, 2019-12-26:
Had great fun with mwp. 
Now it is crashing on startup.
Initially reboot of v-box ubuntu helped but now no more.
Did an upgrade  by "git pull && make && sudo make install" no improvement.
Is there an easy fix.
Thanks indeed
 

[mwp_stderr_2019-12-26.txt](https://github.com/stronnag/mwptools/files/4002988/mwp_stderr_2019-12-26.txt)

stronnag, 2019-12-26:
I've just tested the following configurations:

* mwp on Ubuntu 18.04, Ubuntu is a KVM VM hosted on Linux
* mwp on Ubuntu 18.04, using WSL in WIndows 10, Windows 10 is a KVM VM hosted on Linux
* mwp on cygwin / WIndows 10, Windows 10 is a KVM VM hosted on Linux

In all cases it works just fine. The difference is most likely buggy VirtualBox display emulation, in particular of the Wayland display server.

You could try
```
mwp --noway # force Xlib (vice Wayland)
```
If that works, create (or update if it exists) `~/.config/mwp/cmdopts`, so that it contains the following:
```
# force Xlib for VirtualBox
--noway
```
Then you don't need to explicitly start mwp with the `--noway` option.

helicocrasher, 2019-12-26:
Funny thing, was just retrying without any changes  before you send your message - and it worked again, now flaw. 
Thanks for testing and the hints on workaround, will use them hen problem reapears.

helicocrasher, 2019-12-26:
Crash reappeared after replaying log.

here the shell output with output --noway:

felix@felix-VirtualBox:~/Projects/mwptools$ mwp --noway # force Xlib (vice Wayland)
2019-12-26T18:52:30+0100 mwp startup version: mwp 0523c9f / 2019-12-26 2.352.603 on Xlib
2019-12-26T18:52:30+0100 "Ubuntu 18.04.2 LTS" on felix-VirtualBox running Linux 4.15.0-48-generic x86_64
2019-12-26T18:52:30+0100 hypervisor: oracle
2019-12-26T18:52:30+0100 Using speech api 1 [espeak]
2019-12-26T18:52:30+0100 libchamplain 0.12.16
2019-12-26T18:52:31+0100 Starting Bing proxy 
2019-12-26T18:52:31+0100 BB load async map zoom : true
Settings blank id osm-mapnik
2019-12-26T18:52:31+0100 Delta speed 0.000000
2019-12-26T18:52:31+0100 Host power: 38%, state: Charging (1), warn: None (1)

(mwp:3857): Gdk-ERROR **: 18:52:33.551: The program 'mwp' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1393 error_code 2 request_code 131 (MIT-SHM) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
felix@felix-VirtualBox:~/Projects/mwptools$

stronnag, 2019-12-26:
I really can't fix VirtualBox issues ....

helicocrasher, 2019-12-26:
More observations:

mwp starts sucessfully when Ubuntu is not in fullscreen mode..
in fullscreen mode it crashes.

At least I know now how to have it working. mwp is cool :-)

stronnag, 2019-12-26:
mwp will work the video layer pretty hard (or rather the mapping library will). Unfortunately the VBox video layer has proved a number of times to be quite fragile. You'll probably find it ways again (F/S) in a future VBox update.

closed 2019-12-26 by stronnag