stronnag/mwptools #69: Blackbox replay not working with newer iNav logfiles

wilco1967, 2019-12-27:
For some reason, 'newer' iNav BB logfiles don't want to play anymore in MWP
If I get an older file, it works fine, but recent files show nothing.

They seem to open fine, but other than the filename in the top, nothing shows. 
After a couple of seconds, 'playback' seems to complete, without ever showing anything.

Anything I could try ?

here's what is logged:

wilco@wilco-VirtualBox:~/Projects/mwptools$ mwp
2019-12-27T17:38:09+0100 mwp startup version: mwp 7e7f116 / 2019-12-27 2.361.576 on Xlib
2019-12-27T17:38:09+0100 "Ubuntu 18.04.3 LTS" on wilco-VirtualBox running Linux 4.15.0-66-generic x86_64
2019-12-27T17:38:09+0100 hypervisor: oracle
2019-12-27T17:38:09+0100 Using speech api 1 [espeak]
2019-12-27T17:38:10+0100 libchamplain 0.12.16
2019-12-27T17:38:11+0100 Starting Bing proxy 
2019-12-27T17:38:11+0100 BB load async map zoom : true
2019-12-27T17:38:11+0100 Delta speed 0,000000
2019-12-27T17:38:11+0100 Host power: 79%, state: Discharging (2), warn: None (1)
2019-12-27T17:38:53+0100 Getting base location from index 1 51,963880 6,734822 true
2019-12-27T17:38:54+0100 Geonames TZ: {"status":{"message":"error parsing parameter","value":14}}2019-12-27T17:39:34+0100 Replay "bbox" log /home/wilco/LOG00036.TXT model 24
2019-12-27T17:39:34+0100 replay_bbox_ltm.rb --fd 25 -i 1 -t 24 --decoder blackbox_decode -g -G /home/wilco/LOG00036.TXT 
2019-12-27T17:40:23+0100 Close child pid 3142, 1
2019-12-27T17:40:23+0100 ============== Replay complete ====================
2019-12-27T17:40:23+0100 Close fd #24 :  G_IO_NVAL (20)
2019-12-27T17:40:23+0100 Serial doom replay 2

stronnag, 2019-12-27:
I use this maybe twice a week without issue. Can you let be have failing log? by private mail if there are any privacy / international incident issues.

wilco1967, 2019-12-27:
sent you a file using private mail.

stronnag, 2019-12-27:
Thanks for the log; as expected it plays just fine on my main (Arch) box, but not in my Ubuntu VM. 

I doubt it's an Ubuntu issue, rather the BB replayer requires a setting that only exists on my main machine.  

Working on it.

stronnag, 2019-12-27:
Fixed by 5a1c53c

Recently the blackbox replayer was updated to understand the concept of `home offset`. This broke the common case (how to define `home offset` for BB replay is undocumented, soon to be rectifed).

As an aside, if BB replay fails, the easiest way to see what's happening is to run mwp and the replayer in separate terminals:

```
# Terminal 'A'
# BB log is stored as /tmp/LOG00036.TXT 
$ replay_bbox_ltm.rb -u :3000 /tmp/LOG00036.TXT 
Waiting for GS to start : 
```
and 
```
# Terminal B
$ mwp -s udp://localhost:3000 
```
When `mwp` started, connect to the serial device (`udp://localhost:3000`).

In this case, you'd see in **Terminal A** the bug:
```
 $ replay_bbox_ltm.rb -u :3000 /tmp/LOG00036.TXT 
Waiting for GS to start :  ... OK
Traceback (most recent call last):
/usr/bin/replay_bbox_ltm.rb:972:in `<main>': undefined method `each' for nil:NilClass (NoMethodError)
```
Which is pretty unequivocal.

wilco1967, 2019-12-27:
Yes, works again !
Logs play fine again.
Thanks a lot for such a quick solution !

closed 2019-12-27 by wilco1967

stronnag, 2019-12-27:
mea culpa for breaking it :(

stronnag, 2019-12-27:
and if anyone else cares, here's the wiki documentation for the offending feature https://github.com/stronnag/mwptools/wiki/replay_bbox_ltm.rb-additional-settings