stronnag/mwptools #71: add option to publish BBL data at a lower rate

stronnag, 2020-02-14:
See  also https://github.com/iNavFlight/inav/issues/5408

If BBL has been generated at a high rate for a long flight, replay in mwp becomes impossible due to the number of recorded points being displayed.

While the options:
```
-S, --n-points=N                    Number of points shown in GPS trail
-M, --mod-points=N                  Modulo points to show in GPS trail
```
exist, with lamentably poor documentation,  it would be helpful to take data from large BBL less frequently.

stronnag, 2020-02-14:
See `-S, -M` examples in earlier issue https://github.com/stronnag/mwptools/issues/19#issuecomment-464392333

stronnag, 2020-02-14:
Should be addressed by 7e375e9 (development branch for now)

stronnag, 2020-02-14:
Please reopen if it doesn't work as intended. Here, replaying a 50m log uses 3.3% of memory on a 16GB machine, which seems "mainly harmless".

closed 2020-02-14 by stronnag

b14ckyy, 2020-02-14:
Works like a charm! 
Update rate with the previous 46min log is now 1s in replay mode and performance is wayyy better in a 8GB Virtual Machine. Thanks!