stronnag/mwptools #72: build fail: ubuntu eoan libbluetooth

wx4cb, 2020-04-12:
Any thoughts on how to resolve? bluetooth isnt a bit deal right now

```
root@hamtop:/usr/src/RC/stronnag/mwptools/docs# apt-get install libbluetooth3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libbluetooth-dev' instead of 'libbluetooth3-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libbluetooth-dev : Depends: libbluetooth3 (= 5.50-0ubuntu4) but 5.50-0ubuntu5 is to be installed
E: Unable to correct problems, you have held broken packages.
root@hamtop:/usr/src/RC/stronnag/mwptools/docs# apt-cache search |grep libbluetooth
E: You must give at least one search pattern
root@hamtop:/usr/src/RC/stronnag/mwptools/docs# apt-cache search bluetooth |grep libbluetooth
libbluetooth-dev - Development files for using the BlueZ Linux Bluetooth library
libbluetooth3 - Library to use the BlueZ Linux Bluetooth stack
libbluetooth3-dbg - Library to use the BlueZ Linux Bluetooth stack with debugging symbols

```

stronnag, 2020-04-12:
Works for me ...

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.10
DISTRIB_CODENAME=eoan
DISTRIB_DESCRIPTION="Ubuntu 19.10"
```

```
 $ apt search .\*libbluetooth.\*dev
 Sorting... Done
 Full Text Search... Done
 libbluetooth-dev/eoan-updates,now 5.50-0ubuntu5 amd64 [installed]
   Development files for using the BlueZ Linux Bluetooth library
```
```
 $ grep libblue  docs/ubuntu-deps.txt 
    libbluetooth-dev \
```
The mwp documentation suggests one installs a valid 19.10 package  (` libbluetooth-dev `) . That package installs on 19.10. 

I'm struggling to see how this is an mwp issue?

closed 2020-04-12 by stronnag

wx4cb, 2020-04-13:
i never said it was an mwp issue, i was just confused, but it was a messed up dependency in apt, i ended up reinstalling (which i needed to do anyway)

stronnag, 2020-04-13:
This is the mwp issue tracker, not Ubuntu support ....