stronnag/mwptools #73: Feature Request: Map Pins

wx4cb, 2020-04-19:
is it possible to add a "pin" on the map for a "Static" site?

IE, i would like to put a pin on the map for the local flying sites so that if i'm just looking at the map offline, it would be nice to reference the sites as to where they are in relation to the zoomed out map. If that makes sense.

Easy to store, just have a "file" with name and lat/long. 

Just a thought

stronnag, 2020-04-19:
You can already do this with with the KML/KMZ "File -> Static Overlays" function. Lines, points, filled areas. Nothing complex, but here's the entrance to the local park, point (label, "Entrance"), car park (light purple fill) and SE boundary (red line).
[image]

[image]

You can load / selectivey remove multiple KML / KMZ files. 

here's the KMZ [jh-local-park.zip](https://github.com/stronnag/mwptools/files/4499508/jh-local-park.zip)

stronnag, 2020-04-19:
btw I created the KMZ (KML) file in Google Earth, which should be easy enough.

wx4cb, 2020-04-19:
ok that's what I thought, wasn't sure

stronnag, 2020-04-19:
And, if you haven't discovered it, you can have a set of place names, locations that the "View -> Centre of Position" menu offers as a drop-down list of favourite places. The file is a simple comma delimited file, `~/.config/mwp/places`, thusly:

```
$ cat ~/.config/mwp/places
# mwp places name,lat,lon[,zoom]
Beaulieu,50.8047104,-1.4942621,18
Lordshill,50.9414699,-1.4612946
Gansey,54.083160,-4.728112
Ballakillowey,54.094777,-4.7323538
Netley Marsh,50.9103894,-1.53434157
Jurby,54.353974,-4.5236
Beaulieu MR Ghetto,50.804697,-1.499553,18
```

closed 2020-04-19 by stronnag