stronnag/mwptools #74: Feature Request: PingRX support on a separate uart

wx4cb, 2020-05-07:
How difficult would it be to be able to ingest the mavlink messages from a pingRX receiver on a separate UART and display on the map?

their integration document is here:  https://uavionix.com/downloads/integration/uAvionix%20Ping%20Integration%20Guide.pdf

If needed i can loan you my receiver for testing.

stronnag, 2020-05-08:
I'll have a look at it over the weekend and let you know.

stronnag, 2020-05-08:
So if I understand this correctly, this is basically an ADS-B receiver that will provide information on multiple targets via a MAVLINK interface.

mwp already does something similar in that it supports "inav Radar" such that you can display the presence of other models using inav radar (e.g.  https://www.rcgroups.com/forums/showpost.php?p=41814633&postcount=9); the only difference here (other than the protocol) would be the effectively unlimited number of targets that might be reported. I have some (professional) experience with integrating ADS-B, so I appreciate the target size potential, though I expect that the size of the device and its antenna will limit the amount of data that it can pick up and the time that an individual target will be in view / range is probably quiet short. Particularly if it's on the ground.

So, yes, it's an interesting project, can be implemented in same way as inav radar is (probably just another protocol within the same scheme of things), and with the I/O device defined via the command line (`mwp --radar-device=device-name`) or `~/.config/mwp/cmdopts`. 

I don't think access to the hardware would be necessary (unless you want to persuade the vendor to provide me with one for support purposes), however having a recording of the data stream would be really useful. I'll put together a bespoke capture tool for this purpose and publish it here later on today.  What's the easiest OS for you to do the capture from (Linux, Windows, MacOS etc.)?

Consider me interested.

stronnag, 2020-05-08:
* Separate devices so you get ADS-B targets and your aircraft on the same display
* Tabular display of ADS-B targets / mouse over popup
* Timeout to remove stale ADS-B targets from display

wx4cb, 2020-05-08:
yea it's similar to inav radar except for "real" aircraft :D. it's obviously actually designed to be put on the aircraft and plugged into the telem2 port on the pixhawk. obviously i'm not using a pixhawk. That way then it would be picking up aircraft that are "around" the model you're flying. but down here, i'm so close to a number of GA and international airports there's a lot of traffic, plus riddle and other schools as well. plus with it just being a serial link. i could theoretically put it higher off the ground.

Another thought that just come to mind, i have a "full size" adsb receiver at my house with an antenna at 50 feet, I could pull  raw data from there over the interwebs and then inject it via mavlink into mwp the same way. that would also increase the target audience :D

If there was a way inav could support an mavlink input and merge it with the telemetry coming down that would be great, but obviously that's not the scope of this request.

As far as the capture tool.. linux would probably be the easiest, but it doesn't matter which OS.

stronnag, 2020-05-08:
Actually very few full sized aircraft flying over here at the moment. Not that it;s busy during normal times.

Anyway, if you could use the attached `mwp-serial-cap` to capture the data. There's Linux 64bit binary in the zip, but it's simple to build on any platform (see README.md). A significant recording (maybe 30 minutes+) would be useful, so I can see targets going stale. 

Once the mavlink decoding is done, the extant "radar"  functionality should display targets on the map; then I can "tart it up" a bit.

With your full sized ADS-B recorder, a small python script to convert it mavlink -> mwp. 
[mwp-serial-cap.zip](https://github.com/stronnag/mwptools/files/4598841/mwp-serial-cap.zip)

wx4cb, 2020-05-08:
ok will do, im the middle of doing some electrical work right now but ill look at it either tonight or tomorrow

wx4cb, 2020-05-08:
[pingrx.cap.gz](https://github.com/stronnag/mwptools/files/4600322/pingrx.cap.gz)

Here you go. almost an hours worth of data.

stronnag, 2020-05-08:
Thanks.
I've refactored (aka hacked up) the radar stuff to be more generic. 
So, it **just** needs the decoder.

stronnag, 2020-05-08:
Thanks for the data, parses OK, such is the benefit of already having a working mavlink parser.
So, for your 36.5 minute recording, where there is no callsign, it's because I give up parsing if the location is invalid; no point if it can't be plotted:
```
Targets 67
 id: 10784396, reports 184 callsign N21PJ
 id: 10743384, reports 244 callsign HPJ352
 id: 11295008, reports 178 callsign AAL9647
 id: 10696208, reports 400 callsign ERU306
 id: 10815288, reports 59 callsign 10815288
 id: 10808981, reports 576 callsign 10808981
 id: 10733133, reports 499 callsign N342SA
 id: 11427646, reports 444 callsign 11427646
 id: 11166989, reports 379 callsign XOJ769
 id: 10629235, reports 614 callsign N240ER
 id: 10947084, reports 596 callsign N5526J
 id: 11096587, reports 151 callsign N7YJ
 id: 11079336, reports 148 callsign DAL1486
 id: 10956790, reports 152 callsign N562SP
 id: 11219165, reports 444 callsign 11219165
 id: 10832541, reports 337 callsign N440A
 id: 10574919, reports 398 callsign N187SK
 id: 11171262, reports 959 callsign 11171262
 id: 11343124, reports 204 callsign N942FA
 id: 10855732, reports 174 callsign ERU463
 id: 11380492, reports 393 callsign N979WD
 id: 10902979, reports 727 callsign N5090M
 id: 11090533, reports 367 callsign N694JL
 id: 11021339, reports 505 callsign N62568
 id: 10916214, reports 416 callsign N522MG
 id: 10864892, reports 177 callsign 
 id: 10702515, reports 398 callsign ERU312
 id: 10959233, reports 81 callsign N565AD
 id: 11209254, reports 487 callsign DPJ81
 id: 10871549, reports 539 callsign 10871549
 id: 10493323, reports 389 callsign N1061T
 id: 11403807, reports 348 callsign HOIST86
 id: 11403826, reports 316 callsign HOIST85
 id: 10959306, reports 1193 callsign N565DB
 id: 11146700, reports 448 callsign AAL1078
 id: 10923482, reports 238 callsign N5298N
 id: 10922254, reports 307 callsign N5280R
 id: 10806128, reports 387 callsign ERU414
 id: 11232554, reports 545 callsign N833TB
 id: 10659505, reports 766 callsign N270CC
 id: 10545230, reports 201 callsign N158EC
 id: 10837364, reports 482 callsign N445CT
 id: 10836461, reports 183 callsign ERU444
 id: 10699061, reports 387 callsign ERU309
 id: 10887967, reports 514 callsign ERU495
 id: 10567301, reports 154 callsign N18C
 id: 10843719, reports 221 callsign ERU451
 id: 11344085, reports 356 callsign N943FL
 id: 11375319, reports 390 callsign DAL928
 id: 10768246, reports 458 callsign N377TA
 id: 11030286, reports 399 callsign N63405
 id: 10553403, reports 86 callsign 
 id: 10694306, reports 382 callsign ERU304
 id: 11179960, reports 380 callsign SWA268
 id: 10530314, reports 305 callsign 10530314
 id: 11341354, reports 232 callsign N940LH
 id: 10925983, reports 322 callsign N5318L
 id: 10822546, reports 317 callsign ERU430
 id: 10768843, reports 292 callsign AJT881
 id: 10996800, reports 258 callsign N601PE
 id: 10776455, reports 93 callsign N385TA
 id: 10922893, reports 92 callsign N529ND
 id: 10820994, reports 81 callsign ERU429
 id: 10544684, reports 62 callsign N157WH
 id: 10810883, reports 58 callsign 10810883
 id: 11214872, reports 26 callsign N816EP
 id: 11260245, reports 10 callsign 
```
The attached file is the more completely decoded data:
* still got to do the velocity

Should have symbols on the map soon, the radar plot stuff is now sufficiently abstracted. 

[mavradar.txt](https://github.com/stronnag/mwptools/files/4600894/mavradar.txt)

This is officially a fun project.

wx4cb, 2020-05-08:
dang.. that's not a bad decode rate considering it's on the ground on my shelf

stronnag, 2020-05-08:
Here's a first view of the view ... lots of work to do (better symbols, better stale resolution). I think the device keeps reporting aircraft even when they are out of range, however I'm not looking at the reported timestamp atm, so that may sort the staleness issue. 
[image]
Still. busy morning in W mid Florida.

stronnag, 2020-05-09:
OK, this is getting better. There's now an experimental `enhanced_radar_view` branch.

* Display name is callsign when available, otherwise ICAO number
* Plots are marked as stale after 120s inactivity (icon colour fade, "Stale" in the Radar View list)
* Plots are removed from the map after 300s inactivity, remain in the list as "Hidden" **for now**
* Units in the list according to user preferences

Caveats:

* The list view is ugly and can hold 80 entries; exceeding this number will undoubtedly crash mwp. This is the next thing to fix, but will require redesigning the radar list view. I'll probably add "Last reported" to the list time as well. 
* The icon may change from the white label to a symbolic aeroplane, however this would lose the name/id display on the map ???? TBD.

The `enhanced_radar_view` branch can be used with either a live feed from any device supplying MAVLINK_MSG_ID_TRAFFIC_REPORT messages, or by replaying the donated data dump above.

```
# For live use, use your actual device name, the udp option is for testing
$ mwp  --centre 29.27,-81.08  --radar-device udp://:30001
# Run the simulator in another terminal, redirect to >/dev/null if you like, it's ugly ...
# Ensure the same UDP port (30001) in these examples is used 
$ samples/parseraw.rb  -u 30001 pingrx.cap
```

[image]

wx4cb, 2020-05-09:
"The list view is ugly and can hold 80 entries; exceeding this number will undoubtedly crash mwp. This is the next thing to fix, but will require redesigning the radar list view. I'll probably add "Last reported" to the list time as well."

If it's stale and hidden, just remove it from the list view. you can always just remove the oldest entries every time you update it and just keep it at 80.

BTW running it live with the pingrx caused a bunch of CRC errors (until i remembered you have to put in "@57600" after the device name :D but right now it seems to be running fine

stronnag, 2020-05-09:
here's the end of recording view. From the positions of the stale tracks, it looks like the device has a maximum range of approaching 30km when at ground level, probably more when raised / in the open. Quite impressive.

Next steps:

* Convert table (fixed size grid) view to proper list view
* Sortable on name, status and last report
* Not limited 80 rows
* Remove 'hidden' items from list view
* Note: The present view is not scaleable  / maintainable

[image]

wx4cb, 2020-05-09:
[image]

Not a bad range for a little RX like that.... you're looking at around a 40 mile radius considering it's just on the shelf above my laptop

stronnag, 2020-05-09:
> If it's stale and hidden, just remove it from the list view. you can always just remove the oldest entries every time you update it and just keep it at 80.

As above, not maintainable, harder than doing it properly, which also gives 'free' stuff like sorting.

> BTW running it live with the pingrx caused a bunch of CRC errors (until i remembered you have to put in "@57600" after the device name :D but right now it seems to be running fine.

:-)

> I'm going to let it run all day with live data, i've also got "logging" checkbox at the bottom of the screen checked too so hopefully that will give you some decent data as well

That won't record the traffic stream; it just records the telemetry.
To record the raw stream, it would be necessary to:
* invoke mwp with --raw-log
* use the commit I've just pushed now a135230

wx4cb, 2020-05-09:
ok running with a raw log

wx4cb, 2020-05-09:
ok ran for a couple hours till it core dumped.

attached is the raw log and the crash-dump
[mwp_2020-05-09_074351.raw.txt](https://github.com/stronnag/mwptools/files/4603940/mwp_2020-05-09_074351.raw.txt)

[_usr_src_RC_stronnag_branches_mwptools-enhanced_radar_view_mwp_mwp.1000.crash.txt](https://github.com/stronnag/mwptools/files/4603945/_usr_src_RC_stronnag_branches_mwptools-enhanced_radar_view_mwp_mwp.1000.crash.txt)

stronnag, 2020-05-09:
Thanks for the log; the dump was not necessary, as crashing was a predictable feature on that commit. I'd hove been disappointed it it hadn't crashed.

I've now replaced the list view with a proper list and delete of old entries, new commit soon.

wx4cb, 2020-05-09:
it lasted longer than i thought it would, that said, today is a pretty quiet day

stronnag, 2020-05-09:
OK, now we have 8ff0655 with the new viewer. 

- No limit on the number of 'radar' targets. LAX or LHR six months ago might be interesting
- Stale (hidden) targets are eventually removed, work in progress as hidden has different semantics for MAV traffic reports and inav radar
- The number of items label should be outside of the scrolling region
- logging is excessive

github seems to have borken their markdown parser, wtf.

stronnag, 2020-05-09:
> * The number of items label should be outside of the scrolling region (fixed by 177d134)

wx4cb, 2020-05-09:
i'm running commit 0655 right now. but how about having the number of items in the title bar? eg

"Radar Data (xxx)"   ?

stronnag, 2020-05-09:
I though of that, OTOH, I've fixed the scrolling region now, and the status bar may acquire total / stale items. We'll see.

wx4cb, 2020-05-09:
now question... if i have my telemetry coming from my crossfire via an hc05 and another hc05 to send it to the mfd tracker using "--forward-to" this data won't get sent out right? it's just ingested in?

just checking cus obviously the tracker would go nuts otherwise :P

stronnag, 2020-05-09:
Whoa. Power user!

Should be fine, the forward only sends only telemetry received on the 'main' channel.

wx4cb, 2020-05-09:
> Whoa. Power user!

no i just likes tuff to "just work" and this does (except when i'm a doofus)

> Should be fine, the forward only sends only telemetry received on the 'main' channel.

That's what I thought, just checking

stronnag, 2020-05-09:
Seriously, it's highly gratifying to have people pushing this to the limit.

wx4cb, 2020-05-09:
like i say... outside of PEBKAC, it works great

wx4cb, 2020-05-09:
now one thought... we're not limited to white and grey for the markers on the map right?

Just going back to what you was saying earlier about the actual icon. maybe have a diff coloured plane like what FR24 does? they have a large plane which is current and as it gets stale it changes to  a smaller plane instead. so you COULD just have the plane icon and when hovered over it it gives the call/icao and alt

stronnag, 2020-05-09:
It's all possible. Once I'm happy that the underlying infrastructure is stable, UI changes are all up for grabs. Let's see how it works in cygwin on the Win10 VM.

wx4cb, 2020-05-09:
here's like 2 hours of data from the previous commit and no crash with the radar window open as well

[mwp_2020-05-09_131215.raw.txt](https://github.com/stronnag/mwptools/files/4604339/mwp_2020-05-09_131215.raw.txt)

stronnag, 2020-05-09:
Thanks. 
Slowly getting to the bottom of the logic necessary to handle the various permutations of targets that oscillate in and out of reporting a position whilst also going from having / not having a callsign.

wx4cb, 2020-05-09:
i don't know how much extra processing,  but maybe only show things within a 10 mile radius of the home position?

stronnag, 2020-05-09:
Don't think it makes much difference, in any case:
* I don't know where the home position is (actually I may have a reasonable guess from analysing the distribution of data reports). 
* If the model is armed, a (moving) radius from model might be more useful
* Unless of course the operator is trying to avoid the FAA black helicopters; however that's futile as the FAA helos will have turned off their transponders.

wx4cb, 2020-05-09:
you should know the home position when the aircraft arms do you not?

2. might be a better option

3. what? the FAA not abiding by their own rules? say it ain't so lmao. but actually. the faa here have active transponders and the only one i'm worried about is "AIR-1" LMAO

stronnag, 2020-05-09:
2. is also the computationally most expensive option. For now, I'll leave it as it is. If you're zoomed in then off-screen stuff is probably quite cheap. And I'll take the assumption that the location of the pingRX and home are in reasonable proximity.

stronnag, 2020-05-10:
So, cutesy little icons with mouse-over and fade for stale?
 I think I like it. Not yet pushed.

[image]

wx4cb, 2020-05-10:
Nice. Ill look when i get home.  In fact. I could run it on my laptop when i go to work tomorrow

wx4cb, 2020-05-10:
new commit appears to be working great.
[image]

stronnag, 2020-05-10:
Yes, not only is it pretty nice, I also remembered to add the extra icon files to the repo  **before** you had a chance to pull. 

Also learned a few new tricks to tidy up some of crufty old stuff ... 

I'll probably merge it into at least development tomorrow (and announce that here), so this branch will then disappear.

wx4cb, 2020-05-11:
so far last commit has been running for 3 hours without an issue. but little air traffic today

stronnag, 2020-05-11:
Merged into development, the `enhanced_radar_view` is defunct and will be deleted soon.

As a side-effect there have been a load of nice enhancements into WP tooltips, so it'll go into master tomorrow. Interesting piece of work.
[image]

wx4cb, 2020-05-11:
nice...

is this the correct version number? I just checked out the development branch and all i get is crc errors.

what am i missing?

mwp -l default --centre 29.27,-81.08 --radar-device /dev/ttyUSB4@57600 --raw-log

```
git branch development
git branch --set-upstream-to=origin/development development
git pull

```

stronnag, 2020-05-11:
The commit is correct. It's working fine here, albeit UDP replay; however nothing affecting serial I/O has changed. Strange, very strange.

wx4cb, 2020-05-11:
it's all good... i just did it on my GCS and it's working fine. so don't worry. i think i might have screwed something up on this one

wx4cb, 2020-05-11:
put the dev version on the GCS and fired up the bluetooth modules.

I'm going Crossfire -> MWP -> MFD Tracker. seems to work ok, i'll try it out in the morning. one thing i did notice, the onboard serial port on this board doesn't seem to be up to snuff for the pingRX. but then again, i'm in the middle of a concrete building too

[image]
[image]

stronnag, 2020-05-13:
Stress test (for mwp, not me) ....
[image]
SImulator, I doubt the device would see this many tracks.

wx4cb, 2020-05-13:
you been flying your "drone" around gatwick again? lmao.

Looks good

stronnag, 2020-05-13:
I'm going to merge this into master (as a side effect has been to improve the display of 'icons' in general). 

Time to close off this one.

closed 2020-05-13 by stronnag

wx4cb, 2020-05-13:
glad I thought of something useful this time :D