stronnag/mwptools #75: (Question) CLI Map Preseed

wx4cb, 2020-05-10:
Is there any way to pre-seed (say) an OSM mapnik map for a given lat/long bounding box? or maybe x radius around a point from the CLI ?

couldn't find anything in the wiki

stronnag, 2020-05-10:
Edit => Seed Current Map

closed 2020-05-10 by stronnag

wx4cb, 2020-05-10:
i know that in the actual window. but im talking directly from the CLI

stronnag, 2020-05-10:
`mwptools/samples/map_seeder.rb`

wx4cb, 2020-05-10:
great

stronnag, 2020-05-10:
next you'll ask me how that works, so I'll try and preempt that.
```
$ Projects/mwptools/samples/map_seeder.rb  --help
map_seeder.rb [options]
    -l, --upper-left=LATLON
    -r, --lower-right=LATLON
    -m, --mission-file=FILE
    -u, --uri=URI
    -i, --id=ID
    -z, --zoom=ZOOMS
    -?, --help                       Show this message
```
define the area with -l, -r or by a mission file. `id` is the map cache name (see `~/.cache/champlain/`
URI is the base URI of the online provider, `-z` is a list of the zoom levels.

Even I think it's easily to use the UI in this case.

wx4cb, 2020-05-10:
no i actually figured that out on my own heh

stronnag, 2020-05-10:
Hoisted by my own petard, I even documented it 
`samples/map-cache-readme`

for osm-mapnik `https://tile.openstreetmap.org/#Z#/#X#/#Y#.png`

It might even work after all these years.