stronnag/mwptools #77: Map flashes continually as GPS position updates

howels, 2020-07-31:
Built mwp-tools under ChromeOS (debian buster).

Connected to UAV, preseeded map data but the map view flashes continually as it tries to re-center and the GPS location changes minutely. The map refresh is so slow that the map UI spends a lot of time with a blank view and the map is unreadable flashing between blank/map/blank/map/blank/map all the time.
Can this blank when re-drawing the map be avoided? The UAV is not moving but the map flashes continually as it tries to account for small <1m map movements of the UAV due to GPS jitter..

stronnag, 2020-07-31:
The map performance is entirely depended on the GPU.  Works OK for me on a 2012 Samsung (ARM) Chromebook using Arch Linux.
Chromos is not a supported environment; there is nothing I can do about hardware / OS limitations.

closed 2020-07-31 by stronnag

howels, 2020-07-31:
> The map performance is entirely depended on the GPU. Works OK for me on a 2012 Samsung (ARM) Chromebook using Arch Linux, and on a 2009 Asus eeepc (ia32, pathetic old GPU).
> 
> Chromos is not a supported environment; there is nothing I can do about hardware / OS limitations.

Pentium J4200 on latest chromeos. Understand it's not amazing hardware but surely it should draw a lightweight vector map?