stronnag/mwptools #78: Connection via SmartPort by wire

julioghigi, 2020-09-03:
Hi!

I'm trying to connect using smartaudio but not by a BT module, i'm using an USB TTL converter with the inverter circuit and a wire connected in the back of the R9M, but when I try to connect using the smartaudio option I get the "No data for 5s" message and nothing happens.

I was able to capture the serial data using "Serial Port Monitor" Windows software and by what I studied about SmartPort protocol eveything is all right, the data can be seen at "teste.html" file.

I also generated the log files that you ask for smartport issues, genereated two and both are included.

I also have the log for the terminal in a file, it is included.

I am not able to make a flight right now, but I can do ground tests, all logs was generated with the airplane's GPS locked.

If you need something more to help to discovery whats is wrong, just ask! I will try to help at what I can.

I am using linux with a virtual machine by VirtualBox, all usb works, I was able to conect with the INAV FC by usb cable and mwptools normal, issues just with smartport

All files inside the zip folder due to they format

[mwp_smartport_issue.zip](https://github.com/stronnag/mwptools/files/5171576/mwp_smartport_issue.zip)

julioghigi, 2020-09-04:
Got it working! I just add the baud rate to the inialization code and now it is working by wire! so to begin it is just necessary to start with:
"mwp --smartport /dev/ttyUSB0@57600 --force-type 8"

If you wanna, I can send photos and you add at the wiki the possibility to run smart port only by wire.

stronnag, 2020-09-04:
Thanks, wiki updated.

closed 2020-09-04 by stronnag