stronnag/mwptools #79: TTGO Lora 915Mhz Module for Telemetry

alexvazquez, 2020-11-27:
Hi,

I was reading your experiment using Lora for INAV telemetry. I have a few TTGO Lora 915Mhz modules that I would like to use for Quad or Plane Telemetry using INAV.

Where can I start? What code do I have to upload to the device so it can read INAV Telemetry? Appreciate so much

stronnag, 2020-11-27:
I'm not familiar with that particular device, so I can't give you specific advice.
In general, any transparent serial link can be used for inav telemetry (e.g. MSP or LTM). By transparent serial I mean that you have two (radio) devices, each one has a serial (TTL) side and a over the air radio side. So any serial telemetry data from the FC does over the radios and emerges as serial at the other side.
```
FC telemetry -> [serial ... LoRA modem .. radio] <----------------> [radio ... LoRA modem .. serial] -> Ground station
```
inav and mwp really don't care what the physical connection happens to be, as long as it looks like a serial device at each end.

closed 2020-11-27 by stronnag

alexvazquez, 2020-11-27:
Thanks @stronnag very well explained. So FC telementry is serial actually, so the telemetry pin will send the serial data and I will get the serial data back. Going to investigate on this particular device how can this be done.

stronnag, 2020-11-27:
Pretty much, it doesn't matter if it's Bluetooth, 3DR, LoRA etc. inav puts out a serial stream. 

The ground station side can be more flexible, for example mwp (and ezgui) support the serial to WiFi ESP8266 devices, serial on the FC side, IP on the ground.

alexvazquez, 2020-11-27:
Ok, good what I think is that I will have to upload an Arduino code that reads serial data from a specific pin that will be connected to the FC telemetry TX pin and the receiver Lora device same code but output the data received wireless. This will be fun for sure