stronnag/mwptools #86: mwp should support the BulletGCSS mqtt protocol

stronnag, 2021-01-24:
Yeah, why not. Might be useful for extreme long range flying ...

stronnag, 2021-01-26:
There is now an experimental  `mqtt_telemetry` branch, which seems to work the BulletGCSS mqtt protocol, tested with both a `fl2mqtt` simulation and a recorded live session (thanks Wilco!).

The MQTT component is not built by default, as you need to install an additional dependency.
```
## Debian / Ubuntu ##
sudo apt install libmosquitto-dev

## Arch ##
sudo pacman -S mosquitto
``` 

Then
```
cd mwptools
git pull
git checkout mqtt_telemetry
make MQTT=1
sudo make install
```

If you want MQTT to be set permanently, add it to `local.mk`
```
MQTT=1
```

Now you can use an MQTT URL as a device name, for example for the demo that runs every other hour (00:00. 02:00 ...) UTC on `broker.emqx.io` with topic `org/mwptools/mqtt/otxplayer`, the mqtt URI for mwp would be:
```
mqtt://broker.emqx.io/org/mwptools/mqtt/otxplayer
```
Or in general:
```
mqtt://[user[:pass]@]broker[:port]/topic[?cafile=file]
```
Note:
* port is the mqtt port (typically and by default 1883), not the websocket port.
* if you want to use TLS, then the port will be different, often 8883, and you'll need to provide the broker's CA file. 

I think MQTT is an incredibly elegant solution to long range telemetry, and I hope Daniel perseveres with it, despite the current IOS client issues.

[image]

danarrib, 2021-01-27:
> I think MQTT is an incredibly elegant solution to long range telemetry, and I hope Daniel perseveres with it, despite the current IOS client issues.

I like MQTT too. Don't worry, I'll keep it working over MQTT. The iOS client problems are iOS problems and I don't plan to give it special treatment. I'll just set iPhone to never sleep and stop worrying about that. :-)

stronnag, 2021-01-28:
merged into master
mosquiito is auomagically detected, dependencies updated. 
Closing.

closed 2021-01-28 by stronnag